bdring/FluidNC #1344: Problem: Midtbod/HBot axes calibration.

Kenabi31, 2024-10-01:
### Wiki Search Terms

http://wiki.fluidnc.com/en/config/kinematics
http://wiki.fluidnc.com/en/hardware/3rd-party/MKS_DLC32
http://wiki.fluidnc.com/en/config/axes

### Controller Board

MKS DLC32 v2.1

### Machine Description

DiY H-Bot

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: Plottergeist
meta: (21.11.2022) by thedevleon

kinematics:
  midtbot:

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

start:
  must_home: false
  deactivate_parking: true
  check_limits: false
  

axes:
  shared_stepper_disable_pin: I2SO.0
  
  x:
    steps_per_mm: 20
    max_rate_mm_per_min: 30000.000
    acceleration_mm_per_sec2: 5000.000
    max_travel_mm: 100.000
    soft_limits: false
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 30
      seek_mm_per_min: 7000.000
      feed_mm_per_min: 1000.000
      settle_ms: 20
      seek_scaler: 2.100
      feed_scaler: 2.100
    motor0:
      limit_neg_pin: gpio.36:high
      hard_limits: true
      pulloff_mm: 10
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:high

  y:
    steps_per_mm: 20
    max_rate_mm_per_min: 30000.000
    acceleration_mm_per_sec2: 5000.000
    max_travel_mm: 100.000
    soft_limits: false
    homing:
      cycle: 3
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 30
      seek_mm_per_min: 7000.000
      feed_mm_per_min: 1000.000
      settle_ms: 20
      seek_scaler: 2.1
      feed_scaler: 2.1
    motor0:
      limit_neg_pin: gpio.35:high
      hard_limits: true
      pulloff_mm: 10
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:high

  z:
    steps_per_mm: 10
    max_rate_mm_per_min: 6000
    acceleration_mm_per_sec2: 2000
    max_travel_mm: 10
    homing:
      cycle: 1
      mpos_mm: 1.3
      positive_direction: false
    motor0:
      rc_servo:
        pwm_hz: 50
        output_pin: gpio.22
        min_pulse_us: 1000
        max_pulse_us: 2000

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.8.3 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine Plottergeist]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:DBG: See http://wiki.fluidnc.com/en/config/sd_card#sdfallbackcs-access-sd-without-a-config-file]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (30.000,130.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO: Axis Y (30.000,130.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO: Axis Z (1.300,11.300)]
[MSG:INFO:   Motor0]
[MSG:INFO:     rc_servo Pin:gpio.22 Pulse Len(1000,2000 period:1048575)]
[MSG:INFO:     Update timer for rc_servo at 20 ms]
[MSG:INFO: Kinematic system: midtbot]
[MSG:INFO: Connecting to STA SSID:Leokero 2.4GHz]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.178.76]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 8080]
```

### User Interface Software

WebUI

### What happened?

Im using the Midtbod/Hbot kinematics.
The Y axis travel distance is about 1/2 of the X axis. 
If i move the x and y axis 10mm, the x axis moves 10mm but the y axis moves about 4.75mm. 
Both steps per mm is 20mm. Setting the y axis steps per mm to 40 (double it) does not work because of the kinematics, thats why the head then moves diagonal; the same happens if  double the max rate mm per min on the y axis.

bdring, 2024-12-12:
Show a picture of your machine. You may be using the wrong kinematics

bdring, 2024-12-27:
Closing, OP is unresponsive.

jvsalatino, 2025-06-06:
I have de same problem. Evidently it's a kinematics problem. Can you fix it?

closed 2025-08-17 by bdring