bdring/FluidNC #1347: Problem: steppers not moving

LogrythmAtGmailDotCom, 2024-10-03:
### Wiki Search Terms

steppers not moving
fluidnc config

### Controller Board

MKS DLC32

### Machine Description

tmc2209 drivers, only X stepper connected, 24v connected, USB to Win10

### Configuration file

```yaml
based on 3axis_v4.yaml

stepping:
  engine: RMT
  idle_ms: 255
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0
axes:
  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    motor0:
      limit_all_pin: gpio.17:pu:low
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.12
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    motor0:
      limit_all_pin: gpio.4:pu:low
      stepstick:
        direction_pin: gpio.15
        step_pin: gpio.26
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    motor0:
      limit_all_pin: gpio.16:pu:low
      stepstick:
        direction_pin: gpio.33
        step_pin: gpio.27
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
start:
  must_home: false
```

### Startup Messages

```text
rbl 3.8 [FluidNC v3.8.3 (wifi) '$' for help]
[MSG:INFO: websocket auto report interval set to 300 ms]
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.8.3 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine Little Stiffy]
[MSG:INFO: Board MKS DLC32]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.14 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.17:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.15 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.4:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.27 Dir:gpio.33 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.16:low:pu]
[MSG:INFO:   Motor1]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:Hello Friends 2G]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.20.2]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

WebUI

### What happened?

Motor is not locked or jittering when connected. I tried both the WebUI jog and the fluidterm jog (X10, X-10 etc), motor is not responding.

### Other Information

Thanks for taking the time to read and apologies for any omissions, please let me know if so.
I used the web installer to put fluid on. 
Then went through various config issues until no errors on start up. 
No jogging happens. 
I've studied similar issues 'stepper not moving' on here, with no result. 
I'm useless at electronics btw.
[image]

LogrythmAtGmailDotCom, 2024-10-03:
Motors test okay on existing grbl 1.1 board

LogrythmAtGmailDotCom, 2024-10-03:
Again, apologies if I've missed some important info as I've tried so many things over the past week or so that I'm quite confused. Any guiding tips greatly appreciated. Kinda wishing I'd waited until I could have afforded one of Bart's genuine boards...

LogrythmAtGmailDotCom, 2024-10-03:
$ME nor $MD has any effect, motor not locked, nor moves with fluidterm: X10

Belott, 2024-10-03:
What do you mean with "only X stepper connected"? Do you have actually only the x motor connected, but several stepper drivers on the board?

LogrythmAtGmailDotCom, 2024-10-03:
> What do you mean with "only X stepper connected"? Do you have actually only the x motor connected, but several stepper drivers on the board?

Yes, correct. Please advise if you think this is an issue..
Although I've tried it with only the x stepper driver on the board.

LogrythmAtGmailDotCom, 2024-10-03:
After re-reading https://github.com/bdring/FluidNC/issues/1192 I found the motor IS IN FACT LOCKING! I did say I'm confused as hell lol.
Motor locks when power is connected! 
Very sorry for the previous misleading info.

LogrythmAtGmailDotCom, 2024-10-03:
Update on motor locking circumstances:

1. DLC32 powered off, connect stepper to X axis, power on DLC32: locked stepper motor, no jogging.
2. DLC32 powered on, connect stepper to X axis: motor *not* locked, but still no jogging.

Yikes, this was not helping my confusion.
Learning slowly lol.

Experiment: in desperation, reversed two centre stepper wires (as per random YT suggestion). 
For both 1 & 2 from above, the motor was not locked but nor would it jog. Backing out.

LogrythmAtGmailDotCom, 2024-10-03:
[image]
Nema17 connected with original wiring layout

MitchBradley, 2024-10-03:
What version TMC2209 do you have?  Are they the ones that are configured by UART, or the ones that are jumpered for standalone use?  UART-configured TMC2209s cannot be used with DLC32 without complicated extra wiring.

LogrythmAtGmailDotCom, 2024-10-03:
> What version TMC2209 do you have? Are they the ones that are configured by UART, or the ones that are jumpered for standalone use? UART-configured TMC2209s cannot be used with DLC32 without complicated extra wiring.

Thanks so much for the prompt reply, much appreciated :)
Info from Aliexpress purchase, not sure if it answers your question though:

Product name: TMC2209
Maximum output current: 2A
Recommended operating current: maximum 1.7A
Micro step: 256 micro step subdivision
Configuration: STEP/DIR OR UART
Load driving electric range: 5.5~38V
IC logic operating voltage: 3/5V

MitchBradley, 2024-10-03:
> Configuration: STEP/DIR OR UART

That either means that they sell both versions and you have to be careful to order the one that you want, or that you can modify it for what you want.  The modification for these things involves removing a tiny solder blob on the bottom and adding another one nearby.

DLC32 only supports the "standalone" configuration method, which is probably what they are calling "STEP/DIR".  UART configuration requires special wiring that DLC32 does not have.  If you plug a UART-configured TMC2209 into a board that doesn't support it, it won't turn on properly.

LogrythmAtGmailDotCom, 2024-10-03:
[image]
Unsure if relevant but here is the same Nema17 but using the wiring layout from presently working CNC. This also locks the stepper motor and does not jog, alas.

LogrythmAtGmailDotCom, 2024-10-03:
> That either means that they sell both versions and you have to be careful to order the one that you want, or that you can modify it for what you want. The modification for these things involves [...]

Great info! It lead me to realise that I ordered the wrong unit, DOH! 
One page had TMC2208, TMC2209, and ATD5833.
I've ordered the ATD5833... Like a putz.

MitchBradley, 2024-10-03:
I have never used ATD5833 , but based on some documentation on an AliExpress product page, it appears to be configured without UART, so it should work in DLC32.  You would need to set the potentiometer for the desired drive current.  Presumably there is some documentation on the web describing the correspondence between the voltage setting and the current.

LogrythmAtGmailDotCom, 2024-10-03:
> I have never used ATD5833 , but based on some documentation on an AliExpress product page, it appears to be configured without UART, so it should work in DLC32. You would need to set the [...]

Being a FluidNC newb and given they all seem affordable, I'd be very grateful if you could suggest a reliable stepper driver I should use :)

LogrythmAtGmailDotCom, 2024-10-03:
[image]
Current hardware layout, feeling better already 😉

MitchBradley, 2024-10-03:
It's not really about FluidNC per se.  If you buy a board with onboard drivers or matched drivers that are sold as a kit with the board, it has a much higher chance of working than if you try to mix and match without knowing all the ins and outs.  That said, the easiest drivers to get working are typically A4988 or 8825.  They are really cheap and not particularly great but they are easy to get working because they are simple and there is a lot of howto documentation out there.  ATD5833 looks like it is pretty simple, but it is pretty new so there is much less documentation out there.

LogrythmAtGmailDotCom, 2024-10-03:
The TMC2209 seemed to be the best option at the time, but hearing that there could be iffyness regarding uart etc., I'd love to hear what is considered bullet or newb-proof :)

MitchBradley, 2024-10-03:
Re the wiring - it is hard to say without being able to ohm out your motors, but I would be tempted to try red blue black green on the 6-pin connector.

MitchBradley, 2024-10-03:
TMC2209 is the worst option for a newbie.  WAY too much that can go wrong.  Once you get it wired and configured correctly it is great, but getting it to work at all can require a lot of knowledge and care.

Skorpi08, 2024-10-03:
You have wrong yaml file.
You need I2SO and not RMT.

MitchBradley, 2024-10-03:
Skorpi is right.  Get one of the DLC32 config files.

LogrythmAtGmailDotCom, 2024-10-03:
> https://github.com/bdring/fluidnc-config-files/tree/main/official

Currently applying "MKS_DLC32_21_XYZ.yaml"

LogrythmAtGmailDotCom, 2024-10-03:
> The talk of using a TMC2209 with DLC32 sent me down the rabbit hole. If you really have ATD5833, there is a good chance they will work after you set the current. They might work to some extent with whatever current setting came from the factory but it is a crap shoot. If the factory setting is too low, the motors will be weak; if too high, the motors and drivers might overheat.

Apologies for dragging you down here with me🤓
I'm keen to try. Which way would you turn this little so-and-so?
[image]

LogrythmAtGmailDotCom, 2024-10-03:
My bet would be counter clockwise 🤞

MitchBradley, 2024-10-03:
Here is a datasheet in Chinese.  http://www.szczkjgs.com/UploadFiles/fujian/4881/ATD5833_datasheet_CN.pdf

MitchBradley, 2024-10-03:
[image]
Now we just need to know what value of sense resistor (Rs) they used on your module.  Does the product page say?

MitchBradley, 2024-10-03:
Actually the formula is 
[image]

The formula above is for the protection limit current.

MitchBradley, 2024-10-03:
You will probably be okay by setting the potentiometer in the middle.

LogrythmAtGmailDotCom, 2024-10-03:
> You will probably be okay by setting the potentiometer in the middle.

Thanks.
That is where they 'currently' appear to be..

MitchBradley, 2024-10-03:
Black/green  and blue/red is common standard.

LogrythmAtGmailDotCom, 2024-10-03:
> Black/green and blue/red is common standard.

For clarity, when you suggested "try red blue black green" do you mean leave the cable as is (Red-Blue/Green-Black), or change it to Red-Blue/Black-Green?

TIA :)

LogrythmAtGmailDotCom, 2024-10-03:
And I take it you meant that was only for the motor-end? Leaving the controller end as Red-Blue/Green-Black..

LogrythmAtGmailDotCom, 2024-10-10:
Not sure what the exact fix was. 
In the end I think it was merely a yaml/user issue. 
No need to change any motor wiring, no need to play with the ATD5833 stepper motor drivers, I just needed my yaml ducks lined up.

Lots of good info from Mitch and a good tip from Skorpi on here; "You need I2SO and not RMT", helped a lot to be sure. 

That, together with much reading of other DLC32 related posts, especially the ability to browse through yaml files and copy-paste various bits in via the excellent web installer, all got me there in the end.

Same with also getting the laser working. 
I tried various yaml snippets until I got it happening. 
Although, as the laser has it's own PSU I had to modify the connector to only use "TTL" (or GND) and "S" pins, leaving "Vin" (Voltage in) unused. 
This mod was also necessary as I'm powering the board with 24V, meaning 24V is what would then be coming through "Vin" and not the 12V that the laser actually wants. 
If I was powering the board with only 12V it would be ok to use "Vin" for the laser but still best to use a separate supply imho.

Thanks

closed 2024-10-10 by LogrythmAtGmailDotCom

LogrythmAtGmailDotCom, 2024-10-10:
name: Little Stiffy
board: MKS DLC32
meta: '2024-10-02'
stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: i2so.0
  x:
    steps_per_mm: 800
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 270
    motor0:
      limit_neg_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2
      stepstick:
        step_pin: i2so.1
        direction_pin: i2so.2
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
  y:
    steps_per_mm: 800
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 170
    motor0:
      limit_neg_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 2
      stepstick:
        step_pin: i2so.5
        direction_pin: i2so.6:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 80
    motor0:
      limit_neg_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      stepstick:
        step_pin: i2so.3
        direction_pin: i2so.4
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

start:
  must_home: false

Laser:
  output_pin: gpio.32
  enable_pin: I2SO.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000% 0=12.500% 1700=100.000%

LogrythmAtGmailDotCom, 2024-11-29:
Important update regarding ATD5833 stepper drivers on MKS DLC32 board.
As usual, I was *wrong* that my issue was fixed.
I wrote earlier that it was fixed, but only after a simple movement test... 
When I ran a basic nc file there was *very* buggy movement issues.
I looked back over many things and all I could say was still not perfect was the ATD5833 stepper motor drivers I mistakenly bought.
AFAICT there was no one using them, and almost no useful info out there. 
Was my earlier purchasing mistake the problem?
So I bit the bullet (I'm kidding, they aren't expensive) and carefully purchased a set of (Makerbase brand) TMC2209 stepper motor drivers. And so far so good, hooray!
This is much more like what I expected from FluidNC!
I hate/love this kind of learning experiences where I'm the only one ever with the exact issue. 
I learned so much (many thanks to those here esp.) and I just hope that I can make use of all that hard-earned knowledge, phew.