bdring/FluidNC #1348: Problem: LY620-A RS485

qulet, 2024-10-03:
### Wiki Search Terms

wrzeciona

### Controller Board

6-pack
[image]

### Machine Description

Router bramowy sterowniki TB6600 wrzeciono 2,2Kw chłodzone wodą VFD YL620-A 

### Configuration file

```yaml
name: "ESP32 V4"
board: "ESP32"
meta: "N_ver"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN

  x:
    steps_per_mm: 214
    max_rate_mm_per_min: 200.000
    acceleration_mm_per_sec2: 20.000
    max_travel_mm: 190.000
    homing:
      cycle: 2
      mpos_mm: -3.000
      feed_mm_per_min: 100.000
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        direction_pin: gpio.32
        step_pin: gpio.33

  y:
    steps_per_mm: 214
    max_rate_mm_per_min: 200
    acceleration_mm_per_sec2: 20
    max_travel_mm: 120
    homing:
      cycle: 2
      mpos_mm: -3
      positive_direction: false
    
    motor0:
      limit_all_pin: gpio.39:low
      hard_limits: false
      pulloff_mm: 3
      stepstick:
        direction_pin: gpio.15
        step_pin: gpio.13

  z:
    steps_per_mm: 319
    max_rate_mm_per_min: 200
    acceleration_mm_per_sec2: 25
    max_travel_mm: 75
    homing:
      cycle: 1
      mpos_mm: 3
      positive_direction: true
    
    motor0:
      limit_all_pin: gpio.36:low
      hard_limits: false
      pulloff_mm: 3
      stepstick:
        direction_pin: gpio.14
        step_pin: gpio.12
  a:
    steps_per_mm: 89
    max_rate_mm_per_min: 200
    acceleration_mm_per_sec2: 25
    max_travel_mm: 365
    homing:
      cycle: 3
      mpos_mm: 3
      positive_direction: true
    motor0:
      limit_all_pin: gpio.35:low
      stepstick:
        direction_pin: gpio.4
        step_pin: gpio.22
         
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 8000000
  
coolant:
  flood_pin: gpio.17:low:pu

probe:
  pin: gpio.16:low:pu

user_outputs:
  digital0_pin: NO_PIN

uart1:
  txd_pin: gpio.25
  rxd_pin: gpio.27
  rts_pin: gpio.26
  baud: 9600
  mode: 8N1

YL620:
  uart_num: 1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
  off_on_alarm: false
```

### Startup Messages

```text
Polecenia
$SS
[MSG:INFO: FluidNC v3.8.3 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:4axis_v4_Y.yaml]
[MSG:INFO: Machine ESP32 V4]
[MSG:INFO: Board ESP32]
[MSG:INFO: UART1 Tx:gpio.14 Rx:gpio.15 RTS:gpio.13 Baud:9600]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (-3.000,187.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.33 Dir:gpio.32 Disable:NO_PIN]
[MSG:INFO:  X All Limit gpio.34:low]
[MSG:INFO: Axis Y (-3.000,117.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.27 Disable:NO_PIN]
[MSG:INFO:  Y All Limit gpio.39:low]
[MSG:INFO: Axis Z (-72.000,3.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.25 Disable:NO_PIN]
[MSG:INFO:  Z All Limit gpio.36:low]
[MSG:INFO: Axis A (-362.000,3.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.22 Dir:gpio.4 Disable:NO_PIN]
[MSG:INFO:  A All Limit gpio.35:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:Orange_Swiatlowod_D290]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.22]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: YL620 Spindle  Tx:gpio.14 Rx:gpio.15 RTS:gpio.13 Baud:9600]
[MSG:INFO: Flood coolant gpio.17:low:pu]
[MSG:INFO: Probe gpio.16:low:pu]
ok
M3 S10000
ok
```

### User Interface Software

Webui

### What happened?

[image]

VFD YL620-A
P00.01 3 RS485 control
P03.00 3 transmission 9600
P03.01 1 identifier
P03.02 2 8 1 None
P03.03 3 Communication error No stop
P07.08 5 frequency controlled by RS485
in the send commad window I entered M3 S10000
the spindle does not respond despite confirming the execution of the "ok" command

My RS485 converter is the same as in the post: Isuues #600
I tried swapping the Tx and Rx pins like #600 but without the expected effect.
Is there anyone who managed to start communication via RS485 on the Fluid VFD YL620-A equipment?
Finally, I will add that of course the spindle control from the VFD panel works.

MitchBradley, 2024-12-10:
I recently tested a YL620 with two different RS485 interfaces.  I found that neither worked with FluidNC because both of them had too little idle-state bias voltage between the RS485 A and B lines.  An RS485 master is supposed to bias the lines so that A idles at 250 mV above B.  If that bias is not present, FluidNC cannot read the data that the YL620 returns during the Rx phase.  On both of these interfaces, I was able to fix the problem by adding a 392 ohm resistor from A to 3.3V, and another 392 ohm resistor from B to GND.

That might work for your system, but it is hard to say for sure, especially since the issue you cite for your converter does not mention an RS485 interface.

bdring, 2024-12-13:
Please tell us exactly what RS485 converter you are using. Send a link to where you bought it and try to find a schematic for it.

qulet, 2024-12-17:
Użyłem tych trzech konwerterów  żaden nie zadziałał. Poddałem się i mam zrobione sterowanie 0-10 V. 
[image]

closed 2024-12-18 by MitchBradley