bdring/FluidNC #1350: Problem: M6 Macro is reporting the same value for _current_tool, _selected_tool and #5400

rvalotta, 2024-10-05:
### Wiki Search Terms

_current_tool _selected_tool

### Controller Board

6-pack

### Machine Description

Queenbee (C-Beam based) with Huyang vfd 

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYYZA Huany Laser
meta: 2023-02-22 B. Dring for Joseph Lehman 375912
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 200
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 785
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: -5
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5
      standard_stepper:
        step_pin: i2so.2
        direction_pin: i2so.1
        disable_pin: i2so.0
  y:
    steps_per_mm: 200
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 1170
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: -5
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5
      standard_stepper:
        step_pin: i2so.5
        direction_pin: i2so.4
        disable_pin: i2so.7
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.33:pu
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5
      standard_stepper:
        step_pin: i2so.10
        direction_pin: i2so.9
        disable_pin: i2so.8
  z:
    steps_per_mm: 200
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 119
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: -5
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.32:pu
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5
      standard_stepper:
        step_pin: i2so.13
        direction_pin: i2so.12:low
        disable_pin: i2so.15
  a:
    steps_per_mm: 71.112
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: -1
      positive_direction: true
      mpos_mm: 0
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.18
        direction_pin: i2so.17:low
        disable_pin: i2so.16
probe:
#  pin: gpio.16:high
  pin: gpio.39:low
  check_mode_start: true
  toolsetter_pin: gpio.36:low
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
control:
  reset_pin: NO_PIN
  safety_door_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
start:
  must_home: true
coolant:
  flood_pin: i2so.24
  mist_pin: NO_PIN
  delay_ms: 0
Huanyang:
  uart:
    txd_pin: gpio.14
    rxd_pin: gpio.15
    rts_pin: gpio.13
    baud: 19200
    mode: 8N1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 6000=25% 24000=100%
  atc:
  m6_macro:$SD/Run=atc/tc.nc
uart2:
  txd_pin: gpio.26
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1
uart_channel2:
  report_interval_ms: 75
  uart_num: 2
Laser:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 30
  speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: true
user_outputs:
  digital0_pin:  gpio.25:LOW
user_inputs:
  digital0_pin:  GPIO.16:LOW
macros:
  startup_line0:$SD/Run=atc/init.nc
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.8.3 (main-c04d38b0-dirty) https://github.com/bdring/FluidNC.git]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:6_Pext_XYYZA_HY_Lzr_ATC.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine 6 Pack External XYYZA Huany Laser]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: UART2 Tx:gpio.26 Rx:gpio.4 RTS:NO_PIN Baud:1000000]
[MSG:INFO: uart_channel2 created at report interval: 75]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: User Digital Output: 0 on Pin:gpio.25:low]
[MSG:INFO: User Digital Input: digital0_pin on Pin gpio.16:low]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (-790.000,-5.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X Pos Limit gpio.34]
[MSG:INFO: Axis Y (-1175.000,-5.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Y Pos Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Y2 Pos Limit gpio.33:pu]
[MSG:INFO: Axis Z (-124.000,-5.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12:low Disable:I2SO.15]
[MSG:INFO:  Z Pos Limit gpio.32:pu]
[MSG:INFO: Axis A (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.18 Dir:I2SO.17:low Disable:I2SO.16]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:TiAmo]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.129]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: UART1 Tx:gpio.14 Rx:gpio.15 RTS:gpio.13 Baud:19200]
[MSG:INFO: Huanyang Spindle  Tx:gpio.14 Rx:gpio.15 RTS:gpio.13 Baud:19200]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.2 Freq:5000Hz Period:8191]
[MSG:DBG: VFD setState:5 SpindleSpeed:0]
[MSG:DBG: RPM:0 mapped to device units:0]
[MSG:INFO: Flood coolant I2SO.24]
[MSG:INFO: Probe gpio.39:low]
[MSG:INFO: Toolsetter gpio.36:low]
```

### User Interface Software

Webui

### What happened?

When attempting to use M6_Macro for tool change _current_tool, _selected_tool, #5400 (from wiki) all report the same tool.  current_tool should be what is in the spindle, _selected_tool should be what is passed in M6 T<X>.   _current tool would be used to put the tool back and selected_tool would be used to indicate what tool needs picked up 

### GCode File

(debug, Current tool #<_current_tool> Selected Tool #<_selected_tool> 5400 #5400)

### Other Information

M6 T1
[MSG:DBG: VFD setState:5 SpindleSpeed:0]
[MSG:DBG: RPM:0 mapped to device units:0]
[MSG:DBG: VFD setState:5 SpindleSpeed:0]
[MSG:DBG: RPM:0 mapped to device units:0]
[MSG:INFO: Huanyang spindle changed to tool:1 using m6_macro]
[MSG:INFO: Huanyang spindle run macro: $SD/Run=atc/tc.nc]
ok
[MSG:DBG: Macro line: $SD/Run=atc/tc.nc]
<Idle|MPos:-5.000,-5.000,-5.000,0.000|FS:0,0|Ov:100,100,100|SD:100.00,>
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T1 F0 S0]
[MSG:DBG: DEBUG, Current tool 1.000000Selected Tool 1.0000005400 1.000000]
[MSG:DBG: DEBUG, Current tool selected. Tool change bypassed.]
[MSG:DBG: /atc/tc.nc job sent]
[MSG:DBG:  job sent]
<Idle|MPos:-5.000,-5.000,-5.000,0.000|FS:0,0>
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T1 F0 S0]
ok
M6 T2
[MSG:DBG: VFD setState:5 SpindleSpeed:0]
[MSG:DBG: RPM:0 mapped to device units:0]
[MSG:DBG: VFD setState:5 SpindleSpeed:0]
[MSG:DBG: RPM:0 mapped to device units:0]
[MSG:INFO: Huanyang spindle changed to tool:2 using m6_macro]
[MSG:INFO: Huanyang spindle run macro: $SD/Run=atc/tc.nc]
ok
[MSG:DBG: Macro line: $SD/Run=atc/tc.nc]
<Idle|MPos:-5.000,-5.000,-5.000,0.000|FS:0,0|WCO:-751.093,-1152.339,-97.600,671.223|SD:100.00,>
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T2 F0 S0]
[MSG:DBG: DEBUG, Current tool 2.000000Selected Tool 2.0000005400 2.000000]
[MSG:DBG: DEBUG, Current tool selected. Tool change bypassed.]
[MSG:DBG: /atc/tc.nc job sent]
[MSG:DBG:  job sent]
<Idle|MPos:-5.000,-5.000,-5.000,0.000|FS:0,0>

bdring, 2024-11-14:
Sorry for the delay. I can look into this problem now. 

Do you know if there is a standard parameter number for current tool. I did not find anything at LinuxCNC. I saw some Fanuc references to #4120

bdring, 2024-11-15:
There is a new branch that should fix the problem.

https://github.com/bdring/FluidNC/tree/Fix_selected_tool_param

This will go into the next release. Testing is appreciated.

rvalotta, 2024-11-15:
I will adjust my M6 Script and try it tomorrow...

bdring, 2024-11-19:
Did you get a chance to test this?

rvalotta, 2024-11-19:
No as i accidently blew up my esp32. Replaced it lay night going to test tonight.

rvalotta, 2024-11-20:
Wasn't able to give it a extensive test but it appears to be working

closed 2024-12-10 by bdring