bdring/FluidNC #1351: Problem: Jerky / Pulsing / Uneven Movement.

tomgale, 2024-10-06:
### Wiki Search Terms

pulse, pulsing, uneven, jerky

### Controller Board

MKS Tinybee

### Machine Description

CORE XY Laser cutter, working area 940x620mm MKS tinybee, generic/unknown nema 23 steppers and YAKO YKD2305M external drivers 20mm wide HTD-5M belts

### Configuration file

```yaml
board: MKS TinyBee V1.0 CoreXY
name: GaleLaserDesign01

kinematics:
  corexy:

i2so:
  bck_pin: gpio.25
  data_pin: gpio.27
  ws_pin: gpio.26

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  # uses TH2 IO34 active low - MAKE SURE jumper J2 is set to SDDET!!!
  card_detect_pin: gpio.34:low

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 6
  dir_delay_us: 1
  disable_delay_us: 2

axes:
  x:
    steps_per_mm: 26.6675
    max_rate_mm_per_min: 30000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 945.000
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low:pu
      hard_limits: true
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low
        disable_pin: I2SO.0:high

   

  y:
    steps_per_mm: 26.6675
    max_rate_mm_per_min: 30000.000
    acceleration_mm_per_sec2: 1500.000
    max_travel_mm: 625.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low:pu
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.4
        direction_pin: I2SO.5:low
        disable_pin: I2SO.3:high

   

  z:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 8000.000
    acceleration_mm_per_sec2: 60.000
    max_travel_mm: 80.000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 300.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.22:low
      hard_limits: true
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.7
        direction_pin: I2SO.8
        disable_pin: I2SO.6
             
    motor1:
      limit_pos_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.11
        disable_pin: I2SO.9
   

macros:
  macro0: M3S1&G1F100&G4P2&G0&M5
  macro1: M3S1&G1F500&G4P2&G0&M5
  macro3: M3S1&G1F1000&G4P2&G0&M5

control:
  safety_door_pin: NO_PIN
  # on MT_DET connector
  reset_pin: NO_PIN
  # on TH1 connector
  feed_hold_pin: gpio.36:low
  # on TB connector
  cycle_start_pin: gpio.39:low
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  # Heated Bed Terminal Block
  flood_pin: i2so.16
  # HE0 Terminal Block
  mist_pin: i2so.17
  delay_ms: 0

Laser:
  pwm_hz: 2000
  output_pin: gpio.15
  enable_pin: i2so.18
  disable_with_s0: true
  s0_with_disable: true
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: true

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 2000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000

uart1:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1

uart_channel1:
  report_interval_ms: 75
  uart_num: 1
  
start:
  must_home: false
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.8.3 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine GaleLaserDesign01]
[MSG:INFO: Board MKS TinyBee V1.0 CoreXY]
[MSG:INFO: UART1 Tx:gpio.17 Rx:gpio.16 RTS:NO_PIN Baud:1000000]
[MSG:INFO: uart_channel1 created at report interval: 75]
[MSG:INFO: I2SO BCK:gpio.25 WS:gpio.26 DATA:gpio.27]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:gpio.34:low freq:8000000]
[MSG:INFO: Stepping:I2S_static Pulse:6us Dsbl Delay:2us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,945.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.1 Dir:I2SO.2:low Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.33:low:pu]
[MSG:INFO: Axis Y (0.000,625.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.4 Dir:I2SO.5:low Disable:I2SO.3]
[MSG:INFO:  Y Neg Limit gpio.32:low:pu]
[MSG:INFO: Axis Z (-80.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.7 Dir:I2SO.8 Disable:I2SO.6]
[MSG:INFO:  Z Pos Limit gpio.22:low]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.11 Disable:I2SO.9]
[MSG:INFO:  Z2 Pos Limit gpio.35:low]
[MSG:INFO: feed_hold_pin gpio.36:low]
[MSG:INFO: cycle_start_pin gpio.39:low]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Connecting to STA SSID:VM8669094]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.59]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: Laser Ena:I2SO.18 Out:gpio.15 Freq:2000Hz Period:32767]
[MSG:INFO: Flood coolant I2SO.16]
[MSG:INFO: Mist coolant I2SO.17]
ok
```

### User Interface Software

Lightbun, but also happens through webui

### What happened?

Hi, Trying to get my large CoreXY Laser cutter operating properly, but having some real issues with movement. Basically I think the motors are moving in little pulses, initially I assumed they were the steps of the motors, but changing microstepping in either direction doesn't seem to change the behaviour. It cuts a wiggly line when it should be straight, its less noticeable on the cut when moving slowly, but you can see the motor 'pulsing' as it turns. I've also noticed that the back of material when it hasn't quite cut through is a series of dots, which suggests to me the laser is dwelling in a position ever so slightly then moving.  
Has anyone seen this behaviour before? I've been trying to figure it out for a while, and wanted a second opinion  before I start making drastic changes to hardware. My thoughts on possible causes are:

-Some kind of comms bottleneck where the board cant send pulses to the drives fast enough (the issue isnt between the board and the g-code sender, the behaviour happens even on single g-code moves)

-an issue with the motors themselves, they were salvaged from equipment at work, I suspect they are high quality, but old.

-some weird belt stretch/resonance thing, my belts are really long (about 4m), 20mm wide steel cored 5mm pitch belts. I was thinking it might be something like the belt stretching and relaxing very slightly and creating the odd movement.

I've attached a picture of a fast and slow test cut, with the same part on a K40 for comparison, as well as a video of the pulsing movement. Any advice or thoughts would be greatly appreciated. 

https://github.com/user-attachments/assets/da2e5768-aecb-4c34-bf79-9052545a90ba

[image]

thanks,
Tom

Trolden112, 2024-10-06:
maybe it is belts there  forced  ? too muthc

MitchBradley, 2024-10-06:
Try setting disable_with_s0 to false

tomgale, 2024-10-06:
> Try setting disable_with_s0 to false

No change sadly, also just spotted how messed up that image is for some reason, hopefully this is better:
[image]

MitchBradley, 2024-10-06:
Probably won't help, but also try setting s0_with_disable to false

And also try setting the laser enable pin to NO_PIN and using a switch or something to enable it.  Or move it to a GPIO pin.  The possible problem is that setting I2S pins is very slow compared to GPIOs, except for the (usual) case where they are used for stepping.  The laser power is modulated dynamically during motion, and that can involve the enable pin.  If enable is on an I2S pin, it can mess with the stepping.

tomgale, 2024-10-06:
Thanks for the help, tried both S0 with disable and removing the laser enable pin, part looks identical I think. Might be worth mentioning that I'm running both the cut and engrave on constant power mode, so it shouldn't be switching those outputs much, or even modulating the output pin (GPIO.15).

MitchBradley, 2024-10-06:
Can you observe the jerky motion with simple moves that do not turn on the laser?

tomgale, 2024-10-06:
Yes, that video was from a G1 move with a low feedrate

tomgale, 2024-10-06:
https://github.com/user-attachments/assets/44c03f00-ac34-44da-9630-391d422cd8a0

this video might be clearer, You can see the pulley better on the other side, same thing, g1 move

MitchBradley, 2024-10-06:
Have you tried disabling uart_channel1 ?

tomgale, 2024-10-07:
[image]
Thanks again for the assist
ok, so top to bottom we have:

-laser_enable set to NO_PIN (this change was kept through the other tests too)
-Belts as tight as they go
-belts as loose as they go before they're at risk of falling off. (I've left it like this as it didnt seem to change much and it's a bit less resistance)
-UART disabled (again, left like this)
-Acceleration dropped from 1500 to 500

I'm surprised the belt tension didn't make more difference. I was thinking it might have been the mirror wobbling, but I'd have expected a big difference from dropping the acceleration down in that case.

bdring, 2024-10-07:
Is the gcode using m3 or m4 to control the laser?

Try removing the corexy as a test. The output will be rotated 45 degrees.

MitchBradley, 2024-10-07:
Try Bart's corexy thing first, and if that doesn't help, here are a couple of other things to try in hopes of tracking this down:
1. Change steps/mm to an integer.  Obviously that will change the size but this test could reveal rounding issues.
2. Try adding :low to the step pin definitions.  Many stepper drivers will more or less work with either pulse polarity but can have subtle problems with the wrong polarity.
3. Try bumping up the microstepping a lot.  At the speeds you are going, the short segments are pulsing quite slowly compared to the "segment time", especially considering the need to accelerate and decelerate.  Due to timing limits that are baked into the FluidNC code in consideration of ESP32 hardware characteristics, motion is carved up into segments of a few milliseconds in length.  When going really slowly, each segment can have only one step in it, and it is hard to construct a smooth pulsing profile when you have to distribute a very small number of pulses over a few segments.  So maybe there is some sort of "aliasing jitter" related to that.

I don't know how much moving mass is in your toolhead, but it is possible that the mechanical structure is just not going to work well at certain speeds, considering the discrete nature of the steps, the length and compliance of the belts, and the inertia of the toolhead.  For every discrete step, the impulse at the driving gear has to propagate down the belt and impinge on the toolhead, and then a reflection will propagate back.  I.e. the resonance that you mentioned.

On one of my lasers, I once tried to mount an extra fan on the laser assembly.  It unbalanced the assembly and caused a lot of wobble that caused wiggly lines like the ones in your picture.

tomgale, 2024-10-07:
Thanks guys, tried the cartesian change, integer steps/mm, adding low to the step pins, and 10x the microstepping. All look the same to me, I'm pretty sure we've ruled out all the firmware side things now. I have a plan to rig it up as a traditional cartesian machine to see if that rules out weird long belt effects. I'll mark this as closed since its likely a hardware thing. Really appreciate the help in ruling things out

closed 2024-10-07 by tomgale