bdring/FluidNC #1353: Problem: How to restart wifi after settings

zengfanfan, 2024-10-10:
### Wiki Search Terms

restart radio

### Controller Board

irrelevant

### Machine Description

irrelevant

### Input Circuits

```yaml
irrelevant
```

### Configuration file

```yaml
irrelevant
```

### Startup Messages

```text
irrelevant
```

### What happened?

1. I setup ap ssid: `$AP/SSID=ESP32.XYZ`
2. restart wifi: `$Radio/State=on`
It says: error:3 
[MSG:ERR: Invalid $ statement]

Is this command removed? How can I restart wifi ?

MitchBradley, 2024-10-10:
Just reset the firmware with $bye

zengfanfan, 2024-10-10:
Is it the only way? what if i dont want to reset it?

MitchBradley, 2024-10-10:
It is the only way.

closed 2024-10-10 by zengfanfan