bdring/FluidNC #1354: Problem: Steppers not moving

SpikZASA, 2024-10-12:
### Wiki Search Terms

as per discord 

### Controller Board

XproV5

### Machine Description

Bulkman ultimate bee 1000x1000, 23HD86001Y-21B steppers, 2.2kw spindle (VFD)

### Configuration file

```yaml
# For more information please see: http://wiki.fluidnc.com/en/config/overview

board: XPro V5
name: xPro V5 ASH 

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010

stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 161.212
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 760.00
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 2000.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.500
      tmc_5160:
        step_pin: gpio.12
        direction_pin: gpio.14:low
        use_enable: true
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.050
        run_amps: 1.950
        hold_amps: 0.250
        microsteps: 8
        toff_disable: 0
        toff_stealthchop: 5
        run_mode: CoolStep
        homing_mode: CoolStep
        stallguard: 16
        stallguard_debug: false
[... 284 lines cut ...]
[MSG:DBG: RPM:0 mapped to device units:0]
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG: safety_door_pin 0]
[MSG:DBG: macro0_pin 0]
[MSG:DBG: macro1_pin 0]

Grbl 3.8 [FluidNC 3.8.4-pre2 (wifi) '$' for help]
[MSG:INFO: Huanyang PD0011, PD005 Freq range (100,400) Hz (6000,24000) RPM]
[MSG:INFO: Huanyang PD144 Rated RPM @ 50Hz:3000]
[MSG:INFO: Huanyang PD143 Poles:4]
[MSG:INFO: Huanyang PD014 Accel:6.000]
[MSG:INFO: Huanyang PD015 Decel:6.000]
[MSG:DBG: vfd_cmd_task mode:1]
[MSG:DBG: WStype_Connected]
[MSG:DBG: WebSocket 0 from 192.168.0.7 uri /]
```

### User Interface Software

Webui

### What happened?

Testing the 3.8.4-pre2 release from the 3.8.3....the stepper motors do not respond at all either with jogging or homing.

3404gerber, 2024-10-12:
How did you set the E-Stop Bypass jumper inside the XPro?

SpikZASA, 2024-10-12:
its all set as default i haven't opened the xpro. everything works fine on v3.8.3.

SpikZASA, 2024-10-13:
Mitch resolved the issue [here](https://github.com/bdring/FluidNC/commit/b71fea873f56c89ac250f5ffe2de2a36c5519c7a)

closed 2024-10-13 by SpikZASA