bdring/FluidNC #1355: Problem: Z axis inverted with 3.8.3 pre3 and :high / :low makes no difference

rvalotta, 2024-10-13:
### Wiki Search Terms

Reviewed wiki on how to change direction pin 

### Controller Board

6-pack External

### Machine Description

Gantry Router with external steppers, dual y. Mechanical Switches for endstops.  1.5 VFD Spindle controlled by rs485

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYYZA Huany Laser
meta: 2023-02-22 B. Dring for Joseph Lehman 375912
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 200
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 785
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: -5
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5
      standard_stepper:
        step_pin: i2so.2
        direction_pin: i2so.1
        disable_pin: i2so.0
  y:
    steps_per_mm: 200
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 1170
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: -5
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5
      standard_stepper:
        step_pin: i2so.5
        direction_pin: i2so.4
        disable_pin: i2so.7
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.33:pu
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5
      standard_stepper:
        step_pin: i2so.10
        direction_pin: i2so.9
        disable_pin: i2so.8
  z:
    steps_per_mm: 200
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 119
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: -5
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.32:pu
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5
      standard_stepper:
        step_pin: i2so.13
        direction_pin: i2so.12:high
        disable_pin: i2so.15
  a:
    steps_per_mm: 71.112
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: -1
      positive_direction: true
      mpos_mm: 0
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.18
        direction_pin: i2so.17:low
        disable_pin: i2so.16
probe:
#  pin: gpio.16:high
  pin: gpio.39:low
  check_mode_start: true
  toolsetter_pin: gpio.36:low
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
control:
  reset_pin: NO_PIN
  safety_door_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
start:
  must_home: true
coolant:
  flood_pin: i2so.24
  mist_pin: NO_PIN
  delay_ms: 0
Huanyang:
  uart:
    txd_pin: gpio.14
    rxd_pin: gpio.15
    rts_pin: gpio.13
    baud: 19200
    mode: 8N1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 6000=25% 24000=100%
  atc: atc_manual
uart2:
  txd_pin: gpio.26
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1
uart_channel2:
  report_interval_ms: 75
  uart_num: 2
Laser:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 30
  speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: true
atc_manual:
  safe_z_mpos_mm: -1.000000
  probe_seek_rate_mm_per_min: 800.000000
  probe_feed_rate_mm_per_min: 80.000000
  change_mpos_mm: -400.000 -1000.000 -1.000
  ets_mpos_mm: -94.000 -105.000 -77.000
  ets_rapid_z_mpos_mm: -1.000000
```

### Startup Messages

```text
[MSG:INFO: FluidNC 3.8.4-pre3 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:6_Pext_XYYZA_HY_Lzr_Manual.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine 6 Pack External XYYZA Huany Laser]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: UART2 Tx:gpio.26 Rx:gpio.4 RTS:NO_PIN Baud:1000000]
[MSG:INFO: uart_channel2 created at report interval: 75]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (-790.000,-5.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X Pos Limit gpio.34]
[MSG:INFO: Axis Y (-1175.000,-5.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Y Pos Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Y2 Pos Limit gpio.33:pu]
[MSG:INFO: Axis Z (-124.000,-5.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12:low Disable:I2SO.15]
[MSG:INFO:  Z Pos Limit gpio.32:pu]
[MSG:INFO: Axis A (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.18 Dir:I2SO.17:low Disable:I2SO.16]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:TiAmo]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.129]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: ATC:atc_manual]
[MSG:INFO: UART1 Tx:gpio.14 Rx:gpio.15 RTS:gpio.13 Baud:19200]
[MSG:INFO: Huanyang Spindle  Tx:gpio.14 Rx:gpio.15 RTS:gpio.13 Baud:19200]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.2 Freq:5000Hz Period:8191]
[MSG:DBG: VFD setState:5 SpindleSpeed:0]
[MSG:DBG: RPM:0 mapped to device units:0]
[MSG:INFO: Flood coolant I2SO.24]
[MSG:INFO: Probe gpio.39:low]
[MSG:INFO: Toolsetter gpio.36:low]
```

### User Interface Software

Web UI

### What happened?

Upgraded firmware to 3.8.3.  Tried to home and z started to go into the bed instead of up.
Tried switching dir using :high and same behaviour

bdring, 2024-10-13:
You tried high and low?

rvalotta, 2024-10-14:
as in I do :high or :low on the direction pin and it made no difference on how it behaved

bdring, 2024-10-14:
try the newer pre4 release

MitchBradley, 2024-10-14:
pre5 now

regin92, 2024-10-14:
pre4 works well, I've been running it for two days.

rvalotta, 2024-10-19:
Working as expected again. Closing issue

commit f073e00f (2024-10-13): Fix #1355 - I2SO pin inversion ignored
closed 2024-10-19 by rvalotta