bdring/FluidNC #1357: Feature: Setting step_per_mm paramater for each Y motor individually.

Simon5162, 2024-10-19:
### Machine Context

Hi everyone,

I have two Y motors on my CNC machine, each driving its own belt. I'm trying to set the steps_per_mm parameter for each Y motor individually. At the moment, I can change the step_per_mm parameter, but it applies to the entire axis rather than to each motor separately. Is there a way to modify the config.yaml file to achieve this?

Thank you in advance for your help!

### Feature Description

Setting steps_per_mm paramater for each Y motor individually.

### Other Approaches

Can't solve this problem in other ways currently.

### How I Can Help

Changing config.yaml file to get one steps_per_mm param for each motors on each axis.

bdring, 2024-10-19:
Internal to FluidNC, axes track and calculate everything in steps. It would not be possible to have 2 different values of steps for an axis without a massive overhaul. It is just not worth the effort and risk the stability of the regular code.

Please describe the problem you are having with your machine in greater detail

MitchBradley, 2024-10-19:
It would probably be possible to write a custom Kinematics module that ganged two motors with different steps_per_mm onto one axis.

closed 2024-12-10 by MitchBradley

summerfind, 2025-01-09:
Hi,
My calibration not working, the machine's real movement doesn't change whatever the steps_per_mm set.
how to fix?

MitchBradley, 2025-01-09:
Please do not hijack existing issues.  Open a new one with your information.  Especially do not use a feature request issue for a problem report.

MitchBradley, 2025-06-17:
It could be done by writing a custom kinematics systems that gangs the motors at a higher level than the existing mechanism which gangs them below the step-computation layer.  That said, I am not volunteering to do the coding and support for such a feature.