bdring/FluidNC #1358: Problem: Works perfectly in open-loop, misses steps in closed loop

Nicknico69, 2024-10-21:
### Wiki Search Terms

Closed loop mis-steps

### Controller Board

6-pack

### Machine Description

Home made Gantry XYYZ

### Configuration file

```yaml
name: Hercules_v0
board: 6 Pack
start:
  deactivate_parking: true
  check_limits: false
  must_home: false
axes:
  x:
    steps_per_mm: 640
    max_rate_mm_per_min: 3000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 1020
    soft_limits: true
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 1500
      feed_mm_per_min: 100
      seek_scaler: 1.5
      feed_scaler: 1.5
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3
      standard_stepper:
        step_pin: i2so.2
        direction_pin: i2so.1:low
        disable_pin: i2so.0
  y:
    steps_per_mm: 640
    max_rate_mm_per_min: 3000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 1030
    soft_limits: true
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 1500
      feed_mm_per_min: 100
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3
      standard_stepper:
        step_pin: i2so.5
        direction_pin: i2so.4:low
        disable_pin: i2so.7
    motor1:
      limit_all_pin: NO_PIN
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3
      standard_stepper:
        step_pin: i2so.10
        direction_pin: i2so.9:low
        disable_pin: i2so.8
  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 3000
    acceleration_mm_per_sec2: 250
    max_travel_mm: 165
    soft_limits: false
    homing:
      cycle: 1
      mpos_mm: 0
      seek_mm_per_min: 1500
      feed_mm_per_min: 200
      seek_scaler: 1.5
      feed_scaler: 1.5
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.36
      hard_limits: false
      pulloff_mm: 3
      standard_stepper:
        step_pin: i2so.13
        direction_pin: i2so.12:low
        disable_pin: i2so.15
stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 3
  disable_delay_us: 3
probe:
  pin: gpio.32:low
  toolsetter_pin: NO_PIN
  check_mode_start: false
  hard_stop: false
10V:
  forward_pin: gpio.15:low
  reverse_pin: gpio.14:low
  off_on_alarm: true
  output_pin: gpio.13
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  spinup_ms: 2000
  spindown_ms: 2000
  speed_map: 0=0.000% 24000=100.000%
Laser:
  enable_pin: NO_PIN
  output_pin: gpio.27
  s0_with_disable: true
  off_on_alarm: true
  tool_num: 1
  disable_with_s0: false
  speed_map: 0=0.000% 1000=100.000%
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
coolant:
  flood_pin: I2SO.17
  mist_pin: I2SO.16
  delay_ms: 0
RapidChange:
  alignment: X_AXIS
  direction: POSITIVE
  number_of_pockets: 4
  pocket_offset: 38
  x_pocket_1: 794.5
  y_pocket_1: 987.8
  z_engage: -140
  z_traverse: -10
  z_safe_clearance: 0
  engage_feed_rate: 1200
  load_rpm: 2100
  unload_rpm: 2100
  dust_cover_enabled: false
  tool_recognition_enabled: false
  tool_setter_enabled: true
  x_tool_setter: 947
  y_tool_setter: 987.7
  z_seek_start: -50
  z_safe_tool_setter: 0
  set_tool_offset: 70
  seek_feed_rate: 300
  seek_retreat: 5
  set_feed_rate: 150
  set_tool_max_travel: 150
  dust_cover_axis: A_AXIS
  dust_cover_open_position: 0
  dust_cover_closed_position: 0
  dust_cover_use_output: false
  z_tool_recognition_zone_1: 0
  z_tool_recognition_zone_2: 0
  dust_cover_pin: NO_PIN
  tool_recognition_pin: NO_PIN
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.7.8 (main-b17ab726-dirty)]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.4]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config18-10-24-B.yaml]
[MSG:WARN: Ignored key hard_stop]
[MSG:INFO: Machine Hercules_v0]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: Stepping:I2S_static Pulse:4us Dsbl Delay:3us Dir Delay:3us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,1020.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1:low Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.35]
[MSG:INFO: Axis Y (0.000,1030.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4:low Disable:I2SO.7]
[MSG:INFO:  Y Neg Limit gpio.34]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9:low Disable:I2SO.8]
[MSG:INFO:  Y2 Neg Limit gpio.39]
[MSG:INFO: Axis Z (-165.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12:low Disable:I2SO.15]
[MSG:INFO:  Z Pos Limit gpio.36]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.13 Dir:NO_PIN Fwd:gpio.15:low Rev:gpio.14:low Freq:5000Hz Period:8191]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.27 Freq:5000Hz Period:8191]
[MSG:INFO: Using spindle Laser]
[MSG:INFO: Flood coolant I2SO.17]
[MSG:INFO: Mist coolant I2SO.16]
[MSG:INFO: Probe Pin: gpio.32:low]
[MSG:INFO: WiFi is disabled]
```

### User Interface Software

Lightburn and CNCJS

### What happened?

When working in open loop --> perfect.
When working in closed loop --> position errors.

I use the CL57T v4.1 drivers from stepper online. When set them up in open-loop mode, every thing works perfectly with Lightburn for the laser, or CNCJS for the router. The only thing is I am limited to maximum 3000 mm/min. To solve this, I chose closed-loop drivers and stepper. 
When I switch the drivers to closed loop, in appearance it works fine, but in reality, the steps are not precise, and I get several millimeters errors over a few moves (on both X and Y). It seems to be happening in between the jobs (eg. Two different job, even if there are in the same files directly from fusion 360 /  In lightburn between framing and cutting). The error is reproducible perfectly when executing the same gcode. And the same gcode works perfectly in open-loop mode.

I spend hours on this and looking for answers... Thank you so much in advance for your help.

MitchBradley, 2024-10-21:
This doesn't seem like a problem with FluidNC.

bdring, 2024-10-21:
FluidNC does not know if it is closed loop or not.

Closed loop typically uses a PID algorithm to try to match the encoder position to the steps. You often need to tune it. The PID uses positional error to determine what to do. There is always a little error as it works. 

YouTube has hundreds of videos on PID

Nicknico69, 2024-10-22:
Ok, thank you. The CL57T is advertised as no tuning necessary, so I am surprised that the error margin can be so big (several millimeters) in a few moves. I'll dig on the parameters this week end. 

Thank you for your reply, at least I know I can stop looking on the fluid nc side.

MitchBradley, 2024-10-22:
Try testing with simple linear moves at different speeds and directions, instead of running a complex gcode file.

Nicknico69, 2024-10-23:
Yes I tried, and it happens in just a few manual moves in lightburn : moving the head from one point to another with the pointer, and then launching a simple circle, I get several millimeters aside. 
Even at slow speed such a 500 mm/min. Max acceleration remain unchanged at 200mm/sec².

It loose all coordinates, If I go to home machine then it can hit the limits if the error is in the minus direction.

MitchBradley, 2024-10-23:
Using Lightburn for the moves just adds a layer of uncertainty about what is actually happening, especially since it is not open source and thus is a "black box".  We prefer to test by sending direct GCode commands in FluidTerm, where we know exactly what is happening.

The only idea I have from the FluidNC side is maybe it would help to invert the step pulse by adding :low to step_pin.  On the motor side, I wonder if your encoders are hooked up wrong or faulty. 

Closed loop step-driven motors are supposed to work exactly like open loop as far as FluidNC motion is concerned.  With closed loop, you have the option of hooking up the ALARM output to a fault_pin input, but that does not affect motion in the normal case.  I assume that you have checked the alarm light on the drivers to see if they are unhappy.

I am closing this issue because there is nothing more that we can do to help.  Perhaps you could get help from the hardware vendor.  If you find a solution, please to add a comment to the closed ticket for the benefit of others.

closed 2024-10-23 by MitchBradley

bdring, 2024-10-23:
What does the servo supplier say?