bdring/FluidNC #1359: Problem: Unable to move any axis - Motor leds stay off

gbooysen, 2024-10-23:
### Wiki Search Terms

Axis setup, Testing movements, Testing enable motors.

### Controller Board

6x CNC Controller v1.2

### Machine Description

Gantry router XYZY, hardwired to DRV8825 drivers on Nema17s. Proximity end stops and a Makita r700 router.
Just temp testing before I commit to larger motors and drivers. The cnc is currently working on a Mega5x GRBL in the same config.

### Configuration file

```yaml
'''yaml
board: 6x
name: GB CNC XYZY
stepping:
  engine: I2S_STREAM
  idle_ms: 254
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: I2SO.0
  shared_stepper_reset_pin: NO_PIN
  homing_runs: 2  
  x:
    steps_per_mm: 750.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 314.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.2:high:pu
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: NO_PIN

  y:
    steps_per_mm: 750.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 309.000
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 500.000
      settle_ms: 500
[... 261 lines cut ...]
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$Sta/SSDP/Enable=ON
$AP/Country=01
$AP/SSID=FluidNC
$AP/Password=********
$AP/IP=192.168.0.1
$AP/Channel=1
$Hostname=GBcnc-fluidnc
$HTTP/BlockDuringMotion=ON
$HTTP/Enable=ON
$HTTP/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
ok

[VER:3.7 FluidNC v3.7.17:]
[OPT:MPHS]
[MSG:Machine: 6x Default]
[MSG:Mode=STA:SSID=MyWIFI:Status=Connected:IP=192.168.xx.132:MAC=08-B6-1F-BA-77-7C]
ok

bdring, 2024-10-23:
Are you sure you are using the correct config file.

It says config.yaml, but you mentioned a different one.

Be careful with those stepstick  drivers. The controller was not designed for them and any miswire can break the I2SO chips.

bdring, 2024-10-23:
The disable pins are not correct. See this and send a photo of your setup

http://wiki.fluidnc.com/en/hardware/official/6x_CNC_Controller#motors

gbooysen, 2024-10-24:
I'm sure about the config file - renamed it from original example files on git.
Wiring: I just connected ena, stp, dir and neg to the stepstics - the rest I wire directly from powersupply and then to the steppers.

As for the disable pin, my initial config file was using a shared pin: I2SO.0 - which is motor1's in the example config files.
I tried the 6x_default.yaml file which I renamed to config.yaml which still uses the individual disable pins, still no joy.
I'll take a picture of the setup when I can.