bdring/FluidNC #1360: Problem: job running from SD stalls if wifi conection is lost

Ams-Laser, 2024-10-26:
### Wiki Search Terms

job stall wifi disconect

### Controller Board

MKS tinybee

### Machine Description

laser dual  motors

### Input Circuits

```yaml
not applicable
```

### Configuration file

```yaml
any
```

### Startup Messages

```text
any
```

### User Interface Software

webui

### What happened?

started a job from webui, router accidentally disconected and job stalled

### GCode File

any file

### Other Information

nothing more

MitchBradley, 2024-10-26:
Don't use auto-reporting if you do not absolutely trust wifi.

MitchBradley, 2024-10-26:
You should probably mention the version you are using too.

Ams-Laser, 2024-10-26:
> Don't use auto-reporting if you do not absolutely trust wifi.

ok, turning it off, thanks

MitchBradley, 2025-09-06:
Did you try turning off autoreporting?  Use poll mode.

Gerporgl, 2025-09-06:
My bad, I did not understand at first the comment about auto reporting... I thought it was a github feature we should turn off.

After searching and finding the autoreporting and turning it off, yes it seems to be reliable now, I no longer can reproduce this issue.
Perhaps it should be the default option, or there should be a warning about using autoreporting (maybe there was and I didn't see).

MitchBradley, 2025-09-06:
Yes, I am going to do something about that eventually.

The issue is that, with autoreporting on, the program tries to send reports over the websocket.  Websockets often do not die cleanly, and the autoreports jam up the network stack as it tries to retry the message that does not get through.

Gerporgl, 2025-09-07:
I see, makes sense, this is kind of what I thought.

Some kind of tcp send buffer that gets full and then blocks and takes time to timeout.
When any computer on a network "dies" silently (and this could also happen by a firewall dropping the connection state), it does not have time to gracefully close the tcp connection. The server has to time out to detect that the client disconnected. This has nothing to do with wifi by itself, but is more a general tcp network consideration.

On most OSes, you can do a non blocking send that relies on network send buffer stack, and if the buffer is full, the send call will immediately return with an error.

Although I have done a good amount of ESP32 programming myself, I do not know if it is possible to do non blocking send that only rely on network stack send buffer (I am not sure how much of an independent network stack esp32 has). I use mostly mqtt on my side with async libraries, but I have to admit I did not tests all cases because of the little impact it has on my non real time applications.

MitchBradley, 2025-09-07:
ESP32's network stack is LWIP.  It is pretty good, but depending on how it is configured, it doesn't do everything that a stack running on a big machine with "infinite" RAM can do.  I tried to add code to check for stack congestion but did not quite succeed.  There might be a way to leverage the websocket keepalive pings to solve the problem.  Unfortunately there are too many things to do and not enough time - and certainly not enough people supporting the project with donations.

Gerporgl, 2025-09-07:
Maybe ESPAsyncTCP could be used instead in the arduinoWebSockets library,if it is not already the case, and https://github.com/ESP32Async/ESPAsyncWebServer too. 

Yes, I know these projects take a lot of time. Thanks a lot for all your time. I'll see if how I can help, perhaps with a donation.

Gerporgl, 2025-09-07:
I am sure it would be a good amount of work, I took a look at the code briefly...

Personally for all my ESP32 projects I have switched to async libraries because of similar issues with blocking libraries when the wifi is unavailable, or when my mqtt server may be unreachable once in a while when I do maintenance. My IoT projects are small compared to FluidNC though.

MitchBradley, 2025-09-07:
If you want to have a go at it, all of the network dependencies are isolated in the WebUI subdirectory.