bdring/FluidNC #1361: When using expressions to create a macro, expressions using "and" or 'or" logic are returning an error

rvalotta, 2024-10-26:
### Wiki Search Terms

gcode expressions

### Controller Board

6-pack external

### Machine Description

gantry router with 1,5kw vfd

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYYZA Huany Laser
meta: 2023-02-22 B. Dring for Joseph Lehman 375912
stepping:
  engine: I2S_STATIC
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 200
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 785
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: -5
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5
      standard_stepper:
        step_pin: i2so.2
        direction_pin: i2so.1
        disable_pin: i2so.0
  y:
    steps_per_mm: 200
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 1170
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: -5
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5
      standard_stepper:
        step_pin: i2so.5
        direction_pin: i2so.4
        disable_pin: i2so.7
    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.33:pu
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5
      standard_stepper:
        step_pin: i2so.10
        direction_pin: i2so.9
        disable_pin: i2so.8
  z:
    steps_per_mm: 200
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 200
    max_travel_mm: 119
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: -5
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.32:pu
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 5
      standard_stepper:
        step_pin: i2so.13
        direction_pin: i2so.12:low
        disable_pin: i2so.15
  a:
    steps_per_mm: 71.112
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 300
    soft_limits: false
    homing:
      cycle: -1
      positive_direction: true
      mpos_mm: 0
      feed_mm_per_min: 100
      seek_mm_per_min: 800
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.18
        direction_pin: i2so.17:low
        disable_pin: i2so.16
probe:
#  pin: gpio.16:high
  pin: gpio.39:low
  check_mode_start: true
  toolsetter_pin: gpio.36:low
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
control:
  reset_pin: NO_PIN
  safety_door_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
start:
  must_home: true
coolant:
  flood_pin: i2so.24
  mist_pin: NO_PIN
  delay_ms: 0
Huanyang:
  uart:
    txd_pin: gpio.14
    rxd_pin: gpio.15
    rts_pin: gpio.13
    baud: 19200
    mode: 8N1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 6000=25% 24000=100%
  atc:atc_manual
uart2:
  txd_pin: gpio.26
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1
uart_channel2:
  report_interval_ms: 75
  uart_num: 2
Laser:
  pwm_hz: 5000
  output_pin: gpio.2
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  tool_num: 30
  speed_map: 0=0.000% 1000=100.000%
  off_on_alarm: true
user_outputs:
  digital0_pin:  gpio.25:LOW
user_inputs:
  digital0_pin:  GPIO.16:LOW
macros:
  startup_line0:$SD/Run=atc/init.nc
```

### Startup Messages

```text
[MSG:INFO: FluidNC 3.8.4-pre7 https://github.com/bdring/FluidNC.git]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:6_Pext_XYYZA_HY_Lzr_RCATC.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine 6 Pack External XYYZA Huany Laser]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: UART2 Tx:gpio.26 Rx:gpio.4 RTS:NO_PIN Baud:1000000]
[MSG:INFO: uart_channel2 created at report interval: 75]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STATIC Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: User Digital Output: 0 on Pin:gpio.25:low]
[MSG:INFO: User Digital Input: digital0_pin on Pin gpio.16:low]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (-790.000,-5.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X Pos Limit gpio.34]
[MSG:INFO: Axis Y (-1175.000,-5.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Y Pos Limit gpio.35]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Y2 Pos Limit gpio.33:pu]
[MSG:INFO: Axis Z (-124.000,-5.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12:low Disable:I2SO.15]
[MSG:INFO:  Z Pos Limit gpio.32:pu]
[MSG:INFO: Axis A (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.18 Dir:I2SO.17:low Disable:I2SO.16]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:TiAmo]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.129]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: UART1 Tx:gpio.14 Rx:gpio.15 RTS:gpio.13 Baud:19200]
[MSG:INFO: Huanyang Spindle  Tx:gpio.14 Rx:gpio.15 RTS:gpio.13 Baud:19200]
[MSG:INFO: Laser Ena:NO_PIN Out:gpio.2 Freq:5000Hz Period:8191]
[MSG:DBG: VFD setState:5 SpindleSpeed:0]
[MSG:DBG: RPM:0 mapped to device units:0]
[MSG:INFO: Flood coolant I2SO.24]
[MSG:INFO: Probe gpio.39:low]
[MSG:INFO: Toolsetter gpio.36:low]
ok
```

### User Interface Software

.nc sript run from webui console window

### What happened?

Created a simple script to demo the issue

o100 if[1 EQ 1 OR 0 EQ 1]
(print, hello)
o100 endif

[MSG:DBG: O100IF[1EQ1OR0EQ1]]
[MSG:DBG: Bad GCode: O100IF[1EQ1OR0EQ1]]
[MSG:ERR: 173 (Expression Unknown Operator) in /sd/test.nc at line 1]
[MSG:INFO: PRINT, hello]
[MSG:DBG: O100ENDIF]
[MSG:DBG: Bad GCode: O100ENDIF]
[MSG:ERR: 176 (Flow Control Syntax Error) in /sd/test.nc at line 2]
[MSG:DBG: /test.nc job sent]

but has no errors when run with everything from the or forward removed.

jeyeager65, 2024-10-26:
It works as expected with AND and XOR, but not OR.  It looks like there's a bug in Expression.cpp.  If you use "RR" instead of "OR" it works, but obviously that's not how it's intended to work.

In read_operation, case R should be case O:
`case 'R':
    if (line[pos] == 'R') {
        operation = Binary_NotExclusiveOR;
        pos++;
    } else
        status = Error::ExpressionUnknownOp;  // Unknown operation name starting with R
    break;`

michmela44, 2024-10-26:
In Expression.cpp, there's also a ```binary_ops``` map that is created with "RR"

```
std::map<std::string, ngc_cmd_t, std::less<>> binary_ops = {
    { "+", Binary_Plus },
    { "-", Binary_Minus },
    { "/", Binary_DividedBy },
    { "*", Binary_Times },
    { "**", Binary_Power },
    { "]", Binary_RightBracket },
    { "AND", Binary_And2 },
    { "MOD", Binary_Mod },
    { "RR", Binary_NotExclusiveOr },
    { "XOR", Binary_ExclusiveOr },
    { "EQ", Binary_EQ },
    { "NE", Binary_NE },
    { "GE", Binary_GE },
    { "GT", Binary_GT },
};
```

rvalotta, 2024-10-27:
Thank you i appreciate you finding that...

I have another odd one 

trying to do some simple math and am experiencing some oddness.  

1) print isn't evaluating the math
2) When trying to assign it to another variable its giving me a gcode error

`#<one> = 1
ok
#<two> = 2
ok
(print, #<one>)
[MSG:INFO: PRINT, 1.000000]
ok
(print, #<two>)
[MSG:INFO: PRINT, 2.000000]
ok
(print, [#<two>-#])
[MSG:INFO: PRINT, [2.000000]
ok
#<three>=#-#]
[MSG:DBG: Missing value]
[MSG:DBG: Bad GCode: #<THREE>=]
ERROR:2
Missing the expected G-code word value or numeric value format is not valid.
[MSG:ERR: Bad GCode number format]
#<three>=[#-#]
[MSG:DBG: Bad GCode number format]
[MSG:DBG: Missing value]
[MSG:DBG: Bad GCode: #<THREE>=[]
ERROR:2
Missing the expected G-code word value or numeric value format is not valid.
[MSG:ERR: Bad GCode number format]

Any help is appreciated.
`

rvalotta, 2024-10-27:
seems things are being cut out with a paste here is a pastebin

https://pastebin.com/cgYsqtM3

Also I had typed #<three>=[#<two>-#<one>]  but the console shows it as  #<three>=#-#]

jeyeager65, 2024-10-27:
It's a little hard to follow what you pasted even on pastebin, so I created this:
```
#<one> = 1
#<two> = 2
(print, #<one>)
(print, #<two>)
#<three> = #<one> + #<two>
(print, #<three>)
#<three> = [#<one> + #<two>]
(print, #<three>)
(print, #<one> + #<two>)
(print, [#<one> + #<two>])
```

Running it looks like this:
```
#<one> = 1
ok
#<two> = 2
ok
(print, #<one>)
[MSG:INFO: PRINT, 1.000000]
ok
(print, #<two>)
[MSG:INFO: PRINT, 2.000000]
ok
#<three> = #<one> + #<two>
error:1
(print, #<three>)
[MSG:INFO: PRINT, #]
ok
#<three> = [#<one> + #<two>]
ok
(print, #<three>)
[MSG:INFO: PRINT, 3.000000]
ok
(print, #<one> + #<two>)
[MSG:INFO: PRINT, 1.000000+ 2.000000]
ok
(print, [#<one> + #<two>])
[MSG:INFO: PRINT, [1.000000+ 2.000000]
ok
```

To sum this up:
1. You can't do math inside a print statement.  I think that's reasonable.
2. You need [ ]s when doing math but that still doesn't work in a print.
3. If a variable is undefined, it displays as # in a print.

I think all of that is working as expected.

rvalotta, 2024-10-27:
Figured out the core of my second issue... it was pebcak and frustration related.    I am leaving this open incase the developers want to fix the OR in expressions.cpp

Thankyou for your help

commit 396fb41f (2024-10-27): Fix #1361 - changed RR to OR in expressions
closed 2024-10-27 by rvalotta