bdring/FluidNC #1363: Problem: 3.9.0 Start a file causes reboot.

V1EngineeringInc, 2024-10-29:
### Wiki Search Terms

None

### Controller Board

JAckpot

### Machine Description

Known good Lowrider, board, config, and file. All previously tested through all the 3.8-PreX

### Configuration file

```yaml
board: Jackpot TMC2209
name: LowRider
meta: 03-15-2024 RyanZ

planner_blocks: 32

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

uart1:
  txd_pin: gpio.0
  rxd_pin: gpio.4
  rts_pin: NO_PIN
  baud: 115200
  mode: 8N1

axes:
  shared_stepper_disable_pin: NO_PIN

  x:
    steps_per_mm: 50.000
    max_rate_mm_per_min: 9000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 1220
    soft_limits: false
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 300.000
      seek_mm_per_min: 1500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    #X
    motor0:
      limit_neg_pin: gpio.25:high
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 4.000
      tmc_2209:
        uart_num: 1
        addr: 0
        cs_pin: NO_PIN
        r_sense_ohms: 0.110
        run_amps: 0.800
        homing_amps: 0.800
        hold_amps: 0.500
        microsteps: 8
        stallguard: 0
[... 343 lines cut ...]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: Flood coolant gpio.2]
[MSG:INFO: Mist coolant gpio.16]
[MSG:INFO: Probe gpio.36:low]
```

### User Interface Software

Fluidterm, webuiV2

### What happened?

start know good file get a panic and reboot.

### GCode File

[strut_plate.txt](https://github.com/user-attachments/files/17564137/strut_plate.txt)

### Other Information

[image]

V1EngineeringInc, 2024-10-29:
Confirmed by another user with a different know good file.

aaronse, 2024-10-29:
started with know**n** good file, got a panic and reboot?

V1EngineeringInc, 2024-10-29:
Edited that sentence.

V1EngineeringInc, 2024-10-30:
The file works as expected with 3.8.4-pre9. I am out of ideas to help you out, I have a few boards to test with, just let me know how I can help.

bdring, 2024-10-30:
I can reproduce the issue. I'll see what the backtrace shows and look at the latest commits.

MitchBradley, 2024-10-30:
@dymk - the differences between 3.8.4-pre9 and main are all in your looping code.  Could you take a shot at debugging?

jeyeager65, 2024-10-30:
Looks like this fixes it.  @michmela44 figured it out, I just validated it.

[image]

Here was my stacktrace:
```
Grbl 3.9 [FluidNC v3.9.0 (wifi) '$' for help]
WebSocket disconnect 0
[MSG:INFO: WebUI: Request from 192.168.0.120]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400e4965  PS      : 0x00060530  A0      : 0x800dcc3c  A1      : 0x3ffb20b0  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x80000020  A5      : 0x00000007  
A6      : 0x00000005  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffb2090
A10     : 0x580ea9b9  A11     : 0x3ffc643c  A12     : 0x3ffc6339  A13     : 0x3ffc57b4  
A14     : 0x00003a97  A15     : 0x3f40d0d8  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4008bbe8  LEND    : 0x4008bbfe  LCOUNT  : 0xffffffff  

Backtrace: 0x400e4962:0x3ffb20b0 0x400dcc39:0x3ffb20d0 0x400dcc6c:0x3ffb2100 0x401fdca1:0x3ffb2120 0x400fc275:0x3ffb2150 0x400fc5f5:0x3ffb21b0 0x400fc665:0x3ffb21d0 0x400fd6b1:0x3ffb2210 0x400ee0e4:0x3ffb2250 0x40128d81:0x3ffb2290
  #0  0x400e4962 in Job::save() at FluidNC/src/Job.cpp:37
  #1  0x400dcc39 in _ZL7runFilePKcS0_19AuthenticationLevelR7Channel$isra$58 at FluidNC/src/FileCommands.cpp:257
      (inlined by) runFile at FluidNC/src/FileCommands.cpp:251
  #2  0x400dcc6c in runSDFile(char const*, AuthenticationLevel, Channel&) at FluidNC/src/FileCommands.cpp:269  
  #3  0x401fdca1 in WebCommand::action(char const*, AuthenticationLevel, Channel&) at FluidNC/src/Settings.cpp:539
  #4  0x400fc275 in do_command_or_setting(char const*, char const*, AuthenticationLevel, Channel&) at FluidNC/src/ProcessSettings.cpp:898
  #5  0x400fc5f5 in settings_execute_line(char*, Channel&, AuthenticationLevel) at FluidNC/src/ProcessSettings.cpp:1024
  #6  0x400fc665 in execute_line(char*, Channel&, AuthenticationLevel) at FluidNC/src/ProcessSettings.cpp:1041  
  #7  0x400fd6b1 in protocol_main_loop() at FluidNC/src/Protocol.cpp:289 (discriminator 4)
  #8  0x400ee0e4 in loop() at FluidNC/src/Main.cpp:152
  #9  0x40128d81 in loopTask(void*) at C:/Users/jeyea/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

ELF file SHA256: e4c469edd0e307f2
```

bdring, 2024-10-30:
It works for me too.

There is some test stuff for the PR. We should see if that still runs.

https://github.com/bdring/FluidNC/tree/main/fixture_tests/fixtures

renhil, 2024-10-30:
[image]
[image]
[image]
the activeChannel become 1, when open a file.  but the stack is empty. so get a panic and reboot.

MitchBradley, 2024-10-30:
Each of the 21 uses of Job::active() needs to be checked carefully to see if the change to its definition is appropriate for that situation.  The change was:
```diff
 bool Job::active() {
-    return !job.empty();
+    return !job.empty() || activeChannel;
 }
```
My suspicion is that the change is good for most of the uses in Flowcontrol.cpp, but not for most of the uses elsewhere.  @dymk should weigh in.

V1EngineeringInc, 2024-10-31:
3.9.1 works as expected. 
Thanks guys!

closed 2024-10-31 by V1EngineeringInc