bdring/FluidNC #1364: Problem: Homing issues

mattydboom, 2024-10-30:
### Wiki Search Terms

homing

### Controller Board

RootCNC ISO v3

### Machine Description

UlitmateBee Gantry Router with external DM542T v4,0 stepper drivers, dual Y motors, optical endstops on all axes. Spindle is a 240v Huanyang 2.2kw   

### Configuration file

```yaml
board: Root Controller ISO
name: DatumCAD UltimateBee
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 3
axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 160
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 740
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 100
      seek_mm_per_min: 3000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.7:low
        direction_pin: i2so.5
        disable_pin: i2so.3
  y:
    steps_per_mm: 160
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 300
    max_travel_mm: 1260
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 100
      seek_mm_per_min: 3000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.32
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.12:low
        direction_pin: i2so.10:low
        disable_pin: i2so.8
    motor1:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.13:low
        direction_pin: i2so.11:low
        disable_pin: i2so.9
  z:
    steps_per_mm: 400
    max_rate_mm_per_min: 1500
    acceleration_mm_per_sec2: 50
    max_travel_mm: 150
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0
      feed_mm_per_min: 100
      seek_mm_per_min: 1500
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.27
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.18:low
        direction_pin: i2so.16
        disable_pin: i2so.14
i2so:
  bck_pin: gpio.22
  data_pin: gpio.12
  ws_pin: gpio.21
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: gpio.15:low
coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0
probe:
  pin: gpio.02:low
  toolsetter_pin: NO_PIN
  check_mode_start: true
macros: {}
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: i2so.20
  digital1_pin: i2so.21
  digital2_pin: i2so.22
  digital3_pin: gpio.13
start:
  must_home: true
uart1:
  txd_pin: gpio.17
  rxd_pin: gpio.16
  rts_pin: gpio.4
  baud: 9600
  mode: 8N1
Huanyang:
  uart_num: 1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
```

### Startup Messages

```text
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$SS
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
[MSG:INFO: FluidNC 3.8.4-pre9 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:DatumCAD_UltimateBee.yaml]
[MSG:INFO: Machine DatumCAD UltimateBee]
[MSG:INFO: Board Root Controller ISO]
[MSG:INFO: UART1 Tx:gpio.17 Rx:gpio.16 RTS:gpio.4 Baud:9600]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.21 DATA:gpio.12]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:3us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: User Digital Output: 0 on Pin:I2SO.20]
[MSG:INFO: User Digital Output: 1 on Pin:I2SO.21]
[MSG:INFO: User Digital Output: 2 on Pin:I2SO.22]
[MSG:INFO: User Digital Output: 3 on Pin:gpio.13]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,740.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.7:low Dir:I2SO.5 Disable:I2SO.3]
[MSG:INFO:  X Neg Limit gpio.34]
[MSG:INFO: Axis Y (0.000,1260.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.12:low Dir:I2SO.10:low Disable:I2SO.8]
[MSG:INFO:  Y Neg Limit gpio.32]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.13:low Dir:I2SO.11:low Disable:I2SO.9]
[MSG:INFO:  Y2 Neg Limit gpio.35]
[MSG:INFO: Axis Z (-150.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.18:low Dir:I2SO.16 Disable:I2SO.14]
[MSG:INFO:  Z Pos Limit gpio.27]
[MSG:INFO: estop_pin gpio.15:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:GematMesh]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 10.35.1.31]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc-UltimateBee.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: Huanyang Spindle  Tx:gpio.17 Rx:gpio.16 RTS:gpio.4 Baud:9600]
[MSG:INFO: Probe gpio.2:low]
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
```

### User Interface Software

WebUI, gSender

### What happened?

As per recent discord discussion with Bart: 

There seems to be a serious homing issue with v3.9 that isn’t present on the pre-releases that I have since rolled back and tested (that is to say 3.8.4-pre7, pre8, pre9 all work fine) but 3.9 sends my Y2 axis off (at pace) in the wrong direction after initially homing in the right direction?! 

Only effects the Y2 motor and as far as timing, it seems to be either; just after it first hits its own limit switch OR just after Y1 hits its one… it all happens rather fast and I was always jumping for the estop at the same moment.

mattydboom, 2024-10-30:
OK see below, doesn't seem to differentiate the two Y axis from one another

```
$message/level=debug
ok
$hy
[MSG:DBG: Homing Cycle Y]
[MSG:DBG: Homing nextPhase FastApproach]
[MSG:DBG: Starting from 0.000,0.000,0.000]
[MSG:DBG: Planned move to 0.000,-1386.000,0.000 @ 3000.000]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Homing limited Y]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Homing limited Y]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG: estop_pin 1]
[MSG:INFO: Stopped by estop_pin]
[MSG:INFO: ALARM: Hard Stop]
ALARM:13
ALARM:13
[MSG:ERR: Reset to continue]
ok
```

bdring, 2024-10-30:
Your startup messages are from pre9. Can you show the messages from 3.9?

I see no evidence of  second switch in the debug. Try manually triggering each switch when you are in debug mode. it should look something like this.

```
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Limit switch tripped for Y motor 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG:  Y2 Neg Limit 1]
[MSG:DBG: Limit switch tripped for Y motor 1]
[MSG:DBG:  Y2 Neg Limit 0]
```

mattydboom, 2024-10-30:
Start-up messages with v3.9.0 reinstalled:
```
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$SS
<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
[MSG:INFO: FluidNC v3.9.0 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:DatumCAD_UltimateBee.yaml]
[MSG:INFO: Machine DatumCAD UltimateBee]
[MSG:INFO: Board Root Controller ISO]
[MSG:INFO: UART1 Tx:gpio.17 Rx:gpio.16 RTS:gpio.4 Baud:9600]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.21 DATA:gpio.12]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:3us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: User Digital Output: 0 on Pin:I2SO.20]
[MSG:INFO: User Digital Output: 1 on Pin:I2SO.21]
[MSG:INFO: User Digital Output: 2 on Pin:I2SO.22]
[MSG:INFO: User Digital Output: 3 on Pin:gpio.13]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (0.000,740.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.7:low Dir:I2SO.5 Disable:I2SO.3]
[MSG:INFO:  X Neg Limit gpio.34]
[MSG:INFO: Axis Y (0.000,1260.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.12:low Dir:I2SO.10:low Disable:I2SO.8]
[MSG:INFO:  Y Neg Limit gpio.32]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.13:low Dir:I2SO.11:low Disable:I2SO.9]
[MSG:INFO:  Y2 Neg Limit gpio.35]
[MSG:INFO: Axis Z (-150.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.18:low Dir:I2SO.16 Disable:I2SO.14]
[MSG:INFO:  Z Pos Limit gpio.27]
[MSG:INFO: estop_pin gpio.15:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:GematMesh]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 10.35.1.31]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc-UltimateBee.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: Huanyang Spindle  Tx:gpio.17 Rx:gpio.16 RTS:gpio.4 Baud:9600]
[MSG:INFO: Probe gpio.2:low]
ok
<Alarm|MPos:0.000,0.000,0.000|FS:0,0>
```

mattydboom, 2024-10-30:
Axis limit switches all triggered in turn as suggested:
```
$message/level=debug
ok
[MSG:DBG:  Z Pos Limit 1]
[MSG:DBG: Limit switch tripped for Z motor 0]
[MSG:DBG:  Z Pos Limit 0]
[MSG:DBG:  X Neg Limit 1]
[MSG:DBG: Limit switch tripped for X motor 0]
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Limit switch tripped for Y motor 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG:  Y2 Neg Limit 1]
[MSG:DBG: Limit switch tripped for Y motor 1]
[MSG:DBG:  Y2 Neg Limit 0]
```

bdring, 2024-10-30:
When you have time to test again, contact me on Discord via DM. I want to work interactively with you to debug this.

mattydboom, 2024-10-31:
Closing out as homing issues are resolved in v3.9.1

closed 2024-10-31 by mattydboom