bdring/FluidNC #1369: Feature: software axis skewness compensation

stez90, 2024-11-02:
### Machine Context

FluidNC is targeted to DIY machines that often may have some weakness from design and build quality point of view. Some of those geometrical weaknesses can be addressed via software, like axes being not perfectly square to each other.

### Feature Description

Some cartesian machines may have axes not perfectly squared and don't have an easy ways to adjust them, for example machines with fixed gantry or without  double motors on Y axis.
A kinematic that can take relative angles between axes as config parameters could solve the problem, something like the [M852](https://marlinfw.org/docs/gcode/M852.html) command in Marlin Firmware for 3D printers. Something like:
```
kinematics:
  SkewedCartesian:
    xy angle:
      ...
    xz angle:
      ...
    yz angle:
      ...
```

### Other Approaches

When my machine was out of square and I needed more precision I carefully skewed the geometry in cad prior to gcode generation to match  the expected output.
It's something that could be done with a GCODE postprocessor, but there are none with that feature that I'm aware of, I see only scale and rotations.

### How I Can Help

I'm doing my best and currently looking at how to make a custom kinematic, but not being a developer makes the thing look painful.

MitchBradley, 2024-11-02:
https://github.com/bdring/FluidNC/pull/849 has a skew compensation implementation that is parameterized different, using distance/length instead of angles.  Small angles are hard to measure accurately, except my measuring distances and doing math to get the angle - in which case you might as well enter the distances.  You are welcome to try the PR to see if it meets your needs.

One reason that we haven't merged the PR is out of concerns about documentation and answering questions from people who do not understand it.

In general, getting your machine mechanically square will give better results than trying to compensate in software.

stez90, 2024-11-02:
Thank you, I will try it, looks like what I was looking for.

About different parametrization, I would have left out of the code the way of measuring the skewness, keeping only the relevant number (angle) to describe the geometry.
Getting that number should be made with external tools (as basic as a spreadsheet or fancy like [this](https://www.desmos.com/calculator/ojsqm1dv4b?lang=it)) as there are more ways to get that angle.
For example one can run a dial indicator against a machinist square, or can make a rectangle with the CNC and then measure the diagonals, use a probe against a reference squared piece with a macro. All this should lead to the same angle.

Having a machine mechanically square is obviously the best option, but I see a lot of machines (other than mine) that rely only in the build accuracy of parts without enough margin to adjust some small errors.

MitchBradley, 2024-11-02:
Right.  Queue up the previous comment about not wanting to handhold people who are trying to figure out how to use it.

closed 2024-12-12 by bdring