bdring/FluidNC #1371: Possibilità di imporre una velocità diversa dalla velocità massima del movimento di traslazione nei movimenti G0

Deltaplc, 2024-11-06:
### Machine Context

Buongiorno
mi scuso per l'inconveniente
uso ligtburn per pilotare un laser dove ho installato il tuo firmware.

### Feature Description

Per spostarmi (ad esempio) dal punto di partenza alla zona di incisione, uso il comando G0,
dato che nel mio caso uso una velocità di spostamento massima molto alta,
volevo sapere se era possibile impostare un valore per i movimenti in modalità G0 diverso dal valore di spostamento massimo degli assi?
Magari anche di default per l'intero firmware.... oppure sarebbe fantastico se potessi aggiungere un'impostazione $ per fare questo!
Grazie per l'aiuto

### Other Approaches

 --

### How I Can Help

 --

breiler, 2024-11-06:
The G0 uses the max speed setting `max_rate_mm_per_min`:
http://wiki.fluidnc.com/en/config/axes#axis-letter

If the current max speed is too fast you should consider lowering that value.

Since you are using FluidNC you might consider closing your GRBL issue at:
https://github.com/gnea/grbl/issues/1259

Deltaplc, 2024-11-06:
Thanks for the reply
But if I lower the max_rate_mm_per_min...then when ligtburn uses the G1 with values ​​higher than max rate it is limited!!
How can I solve this problem?

breiler, 2024-11-06:
I don't understand your problem...

`G0` will move at the maximum rate that the machine allows (the `max_rate_mm_per_min`).

`G1` will use the speed defined in your job (the speed setting in LightBurn). If this is set to high it will be capped by the controller so that it doesn't exceed the max speed.
[image]

Deltaplc, 2024-11-06:
ligtburn uses the G0 command to approach the work area and for returning home etc... if I use the max rate it is too high! I would like these movements to be performed at a lower speed.... but ligtburn uses practice the command G0.....
I need very high G1 speeds when it works! but if I lower the max rate for the G0 movement I consequently limit all movements in G1!!
Non so come risolvere questo problema....?

breiler, 2024-11-06:
I'll leave this up to the developers to decide if a config attribute for a max rapid speed should be introduced. 

But to me this is a strange requirement. In what way is the rapid speed (G0) too high when returning to home or to work zero and not while running an actual job (G1, G2, G3)? I don't understand the use case.

breiler, 2024-11-06:
> I would like the movements approaching the area to be worked and the return home after work to be slower....

Yes, you keep writing that - and I am asking why? 
  * Does it feel unsafe?
  * Does the machine stall?
  * Does this in combination with other settings (such as acceleration) make the machine to jerk/move causing it to become unaligned?

> And right now I don't think it's possible?

No it is currently not possible - the rapid movement commands will use the maximum machine speed.

MitchBradley, 2024-11-06:
Everybody expects G0 to travel at the maximum configured rate because that is how it is supposed to work.  If you need to go at a different rate, use M5 to turn off the laser and program a G1 move at the rate you want.  We aren't going to change the meaning of G0 that has existed since forever to help you feel calmer.

MitchBradley, 2024-11-06:
Maybe you can get the LightBurn people to add an option, in exchange for the money you paid for a license to use their non-free software.

closed 2024-11-06 by MitchBradley

bdring, 2024-11-06:
However this is done, it requires a change to Lightburn.

G1 move is the best way.

Deltaplc, 2024-11-06:
The problem is that it doesn't depend on me.... ligbtburn uses G0 to move to the area to be worked on and it cannot be modified!!!
If the G0 speed and the G1-G2-G3 max speed were two separate parameters, in my opinion it would be better.
And it would solve this problem!

Deltaplc, 2024-11-06:
> In ogni caso, è necessario apportare una modifica a Lightburn.
> 
> La mossa G1 è la soluzione migliore.

Yes of course they should create a parameter in G1 for approach moves etc!!

MitchBradley, 2024-11-06:
We are not going to change that.  It is only a problem for you and nobody else.  We don't have time to code, document, and support an unusual parameterization on behalf of a single user.  Sometimes we will add a feature on behalf of a single organization, but only if they pay us for our time.  The work we do for free is for the benefit of multiple users.

Deltaplc, 2024-11-06:
Ok thanks so much for your answers! Sorry if I wasted your time!
Greetings!

bdring, 2024-11-06:
G0 is defined as rapid motion. The machine is supposed to move as fast as possible. This is limited by the max_rate_mm_per_min value for each axis.

The actual rate of the end effector can exceed the values for the axes. If you have an XY G0 move at 45 degrees with equal max rates. The actual feedrate is 1.4x the max rates.

This the expected and defined behavior.

https://linuxcnc.org/docs/2.8/html/gcode/g-code.html#gcode:g0

Deltaplc, 2024-11-06:
> G0 è definito come movimento rapido. La macchina dovrebbe muoversi il più velocemente possibile. Questo è limitato dal valore max_rate_mm_per_min per ogni asse.
> 
> La velocità effettiva dell'effettore finale può superare i valori per gli assi. Se hai un movimento XY G0 a 45 gradi con velocità massime uguali. La velocità di avanzamento effettiva è 1,4 volte le velocità massime.
> 
> Questo è il comportamento previsto e definito.
> 
> https://linuxcnc.org/docs/2.8/html/gcode/g-code.html#gcode:g0

Thanks for the information....But unfortunately this doesn't help me solve the problem, I will try to make my thoughts known to the lightburn developers.

bdring, 2024-11-06:
At any time (in IDLE) you can change the G0 speed like this `$axes/x/max_rate_mm_per_min=200`

The change is only temporary and will revert to the config value the next time the controller starts.

You can also change the acceleration value the same way. I suspect the real problem is your machine cannot handle that value.

Deltaplc, 2024-11-06:
Thanks for your reply
My machine manages the values ​​very well! In fact it approaches the work area and finally returns home at a very high speed! But in my opinion it is not necessary for it to go at those speeds for those movements! Unfortunately the problem is that lighburn uses G0 as a rule, without the possibility of using a speed established by the user!