bdring/FluidNC #1372: Problem: G93 timing inaccurate

AndreasEpp, 2024-11-06:
### Wiki Search Terms

G93, feedrate, inverse

### Controller Board

ESP32 WROOM-32

### Machine Description

Two naked boards without any periphery connected except two probes to my oscilloscope to measure the timing.
One machine is configured with 4 axis and the other one is configured with 1 axis (plus two dummy ones).
Both machines have user output pins defined. I toggle these pins at the start and end of the program to measure timing with my oscilloscope.

I have pasted both my config files in the config file section

[image]

### Input Circuits

```yaml
Two naked boards without any periphery connected except two probes to my oscilloscope to measure the timing.
```

### Configuration file

```yaml
#########################################
# first config file
board: ESPDUINO-32
name: PANTS

user_outputs:
  digital0_pin: gpio.2
  digital1_pin: gpio.4

axes:
  shared_stepper_disable_pin: gpio.12:high
  x:
    steps_per_mm: 160
    max_rate_mm_per_min: 10000
    acceleration_mm_per_sec2: 500
    max_travel_mm: 1000
    motor0:
      stepstick:
        direction_pin: gpio.16:low
        step_pin: gpio.26

  y:
    steps_per_mm: 257.777
    max_rate_mm_per_min: 10000
    acceleration_mm_per_sec2: 500
    max_travel_mm: 1000
    motor0:
      stepstick:
        direction_pin: gpio.27
        step_pin: gpio.25

  z:
    steps_per_mm: 257.777
    max_rate_mm_per_min: 10000
    acceleration_mm_per_sec2: 500
    max_travel_mm: 1000
    motor0:
      stepstick:
        direction_pin: gpio.14:low
        step_pin: gpio.17

  a:
    steps_per_mm: 44.444
[... 549 lines cut ...]
G1X107.60F1440.00
G1X107.62F1440.00
G1X107.64F1440.00
G1X107.66F1440.00
G1X107.69F1440.00
G1X107.71F1440.00
G1X107.73F1440.00
G1X107.75F1440.00
G1X107.78F1440.00
G1X107.80F1440.00
G1X107.82F1440.00
G1X107.84F1440.00
G1X107.86F1440.00
G1X107.89F1440.00
M62 P1
G4 P0.1
M63 P1

### Other Information

I have tested this with version 3.8.4-pre5 and 3.9.1
If both controllers have the same file, they start and finish at the same time.

The overall delay varies with the number of moves.
When I added axes to the first controller so it also has 4 axes that are included in the moves but don't to anything the delay got even worse. The second controller finishes even earlier.

AndreasEpp, 2024-11-06:
I have tested this now with shorter moves (1/60 s, 1/96s, 1/100s, 1/120s). The overall travel is the same, it is just chopped up into more moves (500, 800, 833, 1000) instead of the original 200 moves at 1/24 s. The result is that the shorter the moves, the smaller the delay between controller 1 and 2. The sweetspot seems to be at around 1/100 s with only 20 ms delay between the two controllers. However the overall duration of the program got more inaccurate with ca 8.6s instead of the 8.3 desired. 
At 1/120 s controller 2 finished even after controller 1 instead of the other way around with all the other tests

AndreasEpp, 2024-11-06:
I think I found an explanation:
1. Some of the lines in the file are identical because the original motion is so slow. => identical lines get skipped because the machine is already at the position.
2. because of the slow movement the maximum segment time is reached without planning a step, so a step is performed anyway https://github.com/bdring/FluidNC/issues/772#issuecomment-1380580297

MitchBradley, 2024-12-13:
Your explanation seems plausible.  Is this a practical problem, i.e. do you have a system requirement for motion that slow?

AndreasEpp, 2024-12-14:
I am building a motion control rig for macro videography you can see here: https://youtube.com/shorts/NohEinu3J5U?si=_9nAc7xaJqgFtQb9

I have tackled the problem by increasing the microsteps of the tmc drivers to 64. This got rid of most of the problem, but it still occurs with ultra slow moves. Right now I am preparing to switch the drivers to UART mode to get them to 256 microsteps.

bdring, 2024-12-14:
The video looks great. Is that running FluidNC?

This might be part of the problem. We use a timer to control the time between steps. Choosing the timer speed and resolution affects the speed range. If you allow it to go slower it reduces how fast it can go.

We might be able to create an experimental branch that skews the dynamic range to allow slower speeds for you to test.

This is very deep in the code, so it might take some time before we try anything.

AndreasEpp, 2025-01-10:
I have now switched my TMC2209s to uart and implemented 256 microsteps which makes the timing problem disappear completely. The only downside is that because of the extra cables I can't close the lid anymore, which I will have to resolve in the next version.

For now, the problem no longer occurs with 64 microsteps or higher.

Is there a way to tell the minimum step time for a single axis?
I imagine for a multi-axis move it would be constrained to the slowest moving axis, but for a single axis it should be straight forward, right?

MitchBradley, 2025-01-10:
I suspect that you are asking about the segment time, which controls how slow you can go, instead of the minimum step time, which is related to going fast.  Each segment needs at least one step. ACCELERATION_TICKS_PER_SECOND, defined as 100 in Config.h, controls that.  To change it you would need to recompile the code.

If you want to look at the segment generation code, it is in Stepper.cpp.  It is gnarly.