bdring/FluidNC #1373: Feature: Updated to M6 Macro behavior

rvalotta, 2024-11-06:
### Machine Context

Currently the M6 macro does not run when issuing a M6 T0 and there is no way to unload the current tool as well a M61 Q4 is actually executing the M6 Macro when all it should be doing is setting the currently loaded tool

### Feature Description

a M6 T0 should be handled by the macro if a macro is defined otherwise existing behaviour should continue
a M61 QX should just set the tool and not run the macro

### Other Approaches

Seeing as this is part of the core functionality of FluidNC I am not sure how else to solve this

### How I Can Help

- Am happy to test any changes on my personal setup with the M6_Macro I have developed

bdring, 2024-11-06:
Can you make the changes, test and create a pull request?

rvalotta, 2024-11-06:
Yes I will do so in the next few evenings.

rvalotta, 2024-11-07:
Submitted pull request

bdring, 2024-11-14:
Changes are in the main branch now.

closed 2024-11-14 by bdring