bdring/FluidNC #1375: Problem: Spindle stopped for tool change, while in motion, before operation completes.

ikhwan-ma, 2024-11-09:
### Wiki Search Terms

spindle stopped unexpectedly before tool change

### Controller Board

6-pack External

### Machine Description

Gantry router with external HBS860H all closed loop, dual X motor, single Y, single Z, Fuling/Fohlinn DB600 VFD 3.7KW, 3.5KW 8pole ATC spindle.

### Configuration file

```yaml
board: 6 Pack External
name: 6 Pack Ext XXYZ Fuling ATC
meta: Last changed 22/10/2024
use_line_numbers: true

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 850.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

    motor1:
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  y:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 5000.000
[... 641 lines cut ...]
ok
> N1230G3X61.000Y-21.000Z-2.000I-3.000J0.000K0.000F950.000
ok
> N1240G1X-11.000Y-21.000Z-2.000F950.000
> N1250G0Z5.000
> N1260G0Z5.000
ok
> N1270
> N1280
> N1290M5
> N1300G17G90
> N1310M2
ok
ok
ok
ok
ok
ok
ok

**** Finished sending file in 00:04:12 ****

ok
[MSG:INFO: Program End]
ok

bdring, 2024-11-09:
This is hard to reproduce on my test fixtures.

There should probably be a `protocol_buffer_sync(...)` here to complete all buffered gcode. 

Probably right before `spindle->stop();` 

```c++
if (gc_state.selected_tool != gc_state.tool) {
            bool stopped_spindle;
            Spindles::Spindle::switchSpindle(gc_state.selected_tool, Spindles::SpindleFactory::objects(), spindle, stopped_spindle);
            if (stopped_spindle) {
                spindle->stop();  // stop the new spindle
                gc_state.spindle_speed = 0.0;
                gc_block.modal.spindle = SpindleState::Disable;
            }
            spindle->tool_change(gc_state.selected_tool, false, false);
            ...
   ```

ikhwan-ma, 2024-11-10:
Hi.. thanks for your reply.

I went into the source code and saw this function call as well. But not
pursuing where this function originates from because I think I'd spend the
time to test on another gcode file.

Found out everything went well for another file (and combined operations),
and did the same thing again on the same 'problem' file, even after I
rebooted the machine and restart UGS on my laptop.

I understand this is hard to replicate on your setup but can you at least
share with me what kind of situation that might contribute to this erratic
problems..? Let me know if you need more info on my setup or whatever it is
to narrow this down.

bdring, 2024-11-10:
Try this pull request.

https://github.com/bdring/FluidNC/pull/1374

ikhwan-ma, 2024-11-11:
thanks for the update.

ikhwan-ma, 2024-11-12:
> Try this pull request.
> 
> #1374

I need some help with this.

bdring, 2024-11-12:
I'll create a pre release with the new code.  Is that the help you need?

ikhwan-ma, 2024-11-12:
Yes, please! I very much appreciate your help on that mstter. Thank you!

bdring, 2024-11-12:
Pre-release is done.

https://github.com/bdring/FluidNC/releases/tag/v3.9.2-pre1

Get is via Github or the Web Installer (be sure to enable pre-releases)

ikhwan-ma, 2024-11-12:
Awesome, thank you so much! I had to shutdown my machine yesterday after I
can't progress with my work, no matter how I changed the milling operations
or restarted the machine.

I'll make the update today and report the outcome accordingly.

Thanks again.

ikhwan-ma, 2024-11-13:
Happy to report that the 'early stop' spindle bug is gone. There's something that i thought was part of the erratic problem from the current 3.9.1 release, but persist in the 3.9.2 pre-release. I'll keep an eye for the current state of update and notify if there any other problem.

I'll open another issue later base on my findings. Perhaps another video too to document the tool change procedure.

Thanks for this! I'll make a donation soon as a token of gratitude for all your work guys..

closed 2024-11-13 by ikhwan-ma