bdring/FluidNC #1376: Feature: Blow chips off the top of TLO sensor during probing offsets, between tool changes.

ikhwan-ma, 2024-11-12:
### Machine Context

Some have a TLO sensor with the blower nozzle to clear of chips whenever it's probing. It's important to clear the top where the bits would touch the sensor.

This is a suggestion for the **atc_manual** function, since it's offered by the developer, not written manually in a macro by any end user.

### Feature Description

I suggest to create a new parameter like 'tlo_blower_pin' and when paired to a digital output to control a solenoid, is capable of turning on and off accordingly during offset probing within the tool change procedure.

### Other Approaches

I don't know if there're any better ways to do this, but adding this feature will add to the autonomous capability off the system, and require less time for the operator to stand around to clear off chips manually during each probing.

### How I Can Help

I'm not much of a programmer to start with, but I can understand the flow. I don't think I can be much of help. I'll make a donation later.

bdring, 2024-11-12:
Were you able to test the PR for your other issue?

ikhwan-ma, 2024-11-12:
> Were you able to test the PR for your other issue?

I don't know how to perform the test.. can't figure out how. Any help is highly appreciated.

Anyway, the spindle stopped event happened again yesterday while it was doing some spot drilling. Lucky it wasn't a heavy operation and I was quick to push the feed hold button and then stopping the program. This was the only time I skipped 'air cutting'..

bdring, 2024-12-11:
I don't think we want to implement this in firmware. The ATC manual feature assumes an operator is there to change the bit. It also assumes a pretty simple system (because, manual). A simple button for a solenoid valve should work.

This is easy to implement in Gcode using M62.

Note: There is no way in firmware to know which probe is being used with a G38 command, so adding it there seems wonky.

closed 2024-12-11 by bdring