bdring/FluidNC #1378: Problem: Pulloff_mm not added to Machine Position after homing

Quintenluyten, 2024-11-13:
### Wiki Search Terms

homing pulloff_mm mpos_mm "homing and limit switches" "limit switch and homing setup flowchart"

### Controller Board

Fysetc E4

### Machine Description

I am building a custom 4 axis hot wire cutter. I am currently testing with 4 linear Cartesian axes, every axis has its own limit switch. So the machine has a XY gantry, and a AZ gatry. Once this homing issue is fixed, I will convert the kinematics to 2 makerbot gantries.
[image]
[image]

### Configuration file

```yaml
name: EleksLaser with Fysetc E4 controller
board: Fysetc E4
stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0
kinematics: {}
uart1:
  txd_pin: gpio.22
  rxd_pin: gpio.21
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 115200
  mode: 8N1
axes:
  shared_stepper_disable_pin: gpio.25
  x:
    steps_per_mm: 100
    max_rate_mm_per_min: 500
    acceleration_mm_per_sec2: 50
    max_travel_mm: 290
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 200
      seek_mm_per_min: 500
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 3
      tmc_2209:
        uart_num: 1
        addr: 1
        r_sense_ohms: 0.11
        run_amps: 1
        homing_amps: 1
        hold_amps: 0.5
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: gpio.27
        direction_pin: gpio.26
        disable_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
  y:
    steps_per_mm: 100
    max_rate_mm_per_min: 500
    acceleration_mm_per_sec2: 50
    max_travel_mm: 150
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 200
      seek_mm_per_min: 500
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.35:low
      hard_limits: false
      pulloff_mm: 3
      tmc_2209:
        uart_num: 1
        addr: 3
        r_sense_ohms: 0.11
        run_amps: 1
        homing_amps: 1
        hold_amps: 0.5
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: gpio.33
        direction_pin: gpio.32
        disable_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
  z:
    steps_per_mm: 100
    max_rate_mm_per_min: 500
    acceleration_mm_per_sec2: 50
    max_travel_mm: 150
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 200
      seek_mm_per_min: 500
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.15:low
      hard_limits: false
      pulloff_mm: 3
      tmc_2209:
        uart_num: 1
        addr: 0
        r_sense_ohms: 0.11
        run_amps: 1
        homing_amps: 1
        hold_amps: 0.5
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: gpio.16
        direction_pin: gpio.17:low
        disable_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
  a:
    steps_per_mm: 100
    max_rate_mm_per_min: 500
    acceleration_mm_per_sec2: 50
    max_travel_mm: 290
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 200
      seek_mm_per_min: 500
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.36:low
      hard_limits: false
      pulloff_mm: 3
      tmc_2209:
        uart_num: 1
        addr: 2
        r_sense_ohms: 0.11
        run_amps: 1
        homing_amps: 1
        hold_amps: 0.5
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: false
        step_pin: gpio.14
        direction_pin: gpio.12:low
        disable_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
start:
  must_home: true
  deactivate_parking: false
  check_limits: false
PWM:
  output_pin: gpio.2
```

### Startup Messages

```text
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$SS
<Alarm|MPos:0.000,0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,70.000,30.000>
[MSG:INFO: FluidNC v3.0.x (noGit)  (noGit)]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yml]
[MSG:INFO: Machine EleksLaser with Fysetc E4 controller]
[MSG:INFO: Board Fysetc E4]
[MSG:INFO: UART1 Tx:gpio.22 Rx:gpio.21 RTS:NO_PIN Baud:115200]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Shared stepper disable gpio.25]
[MSG:INFO: Axis X (0.000,290.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 UART1 Addr:1 CS:NO_PIN Step:gpio.27 Dir:gpio.26 Disable:NO_PIN R:0.110]
[MSG:INFO:  X Neg Limit gpio.34:low]
[MSG:INFO: Axis Y (0.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 UART1 Addr:3 CS:NO_PIN Step:gpio.33 Dir:gpio.32 Disable:NO_PIN R:0.110]
[MSG:INFO:  Y Neg Limit gpio.35:low]
[MSG:INFO: Axis Z (0.000,150.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 UART1 Addr:0 CS:NO_PIN Step:gpio.16 Dir:gpio.17:low Disable:NO_PIN R:0.110]
[MSG:INFO:  Z Neg Limit gpio.15:low]
[MSG:INFO: Axis A (0.000,290.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     tmc_2209 UART1 Addr:2 CS:NO_PIN Step:gpio.14 Dir:gpio.12:low Disable:NO_PIN R:0.110]
[MSG:INFO:  A Neg Limit gpio.36:low]
[MSG:INFO: X Axis driver test passed]
[MSG:INFO: Y Axis driver test passed]
[MSG:INFO: Z Axis driver test passed]
[MSG:INFO: A Axis driver test passed]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:EC-net-Guest]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.0.191]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.2 Dir:NO_PIN Freq:5000Hz Period:8191]
ok
<Alarm|MPos:0.000,0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
$H
[MSG:Homed:XA]
[MSG:Homed:YZ]
ok
<Idle|MPos:0.000,0.000,0.000,0.000|FS:0,0>
```

### User Interface Software

WebUI

### What happened?

Expected behavior:
After homing, I expect the MPos to be set to "MPos:3.000,3.000,3.000,3.000" since the machine homes at (0,0,0,0), and has a pulloff distance of 3 mm. 

Observed behavior:
However, the machine sets the machine position to "MPos:0.000,0.000,0.000,0.000" after the homing routine (including pulloff) is done. So now the limit switches are at (-3,-3,-3,-3) and out of reach of the soft limits of the machine.

Other thing I tried:
If I set the mpos_mm of each axis to 3 mm, then the soft limits limit movement to (3 mm, max_travel_mm + 3 mm). However, I want the travel to be limited to (0 mm, max_travel_mm), with 0 mm being the limit switch location. Therefore, the pulloff distance needs to be used to set the machine position after the homing routine.

Why the expected behavior should make sense:
The following page indicates that my expected behavior is what fluidnc should do, but I do not observe it in practice: http://wiki.fluidnc.com/en/support/machine_space_and_homing

MitchBradley, 2024-11-13:
The wiki was misleading and inconsistent.  I edited it to reflect the intended behavior, which FluidNC implements correctly.  After homing, mpos_mm is set to one end of the travel range.  pulloff_mm is a guard zone that is outside the travel range.  When soft limits are enabled, you can travel all the way to mpos_mm without consideration of pulloff_mm

closed 2024-11-14 by bdring

MitchBradley, 2024-11-14:
Can you put the Z homing switch on the end away from the bed, and use the probe function to reference to the bed?  Or not home Z at all but rather just probe that axis?