bdring/FluidNC #1379: Problem: Incorrect I2S BCK signal on v3.9.1

MrTransistorsChannel, 2024-11-13:
### Wiki Search Terms

Used the wiki to configure I2S

P.S. Also found that in [image] in GPIO Pins section there is a link to *pin guidelines* that points to old github wiki

### Controller Board

ESP32-WROOM-32 dev kit board with ESP32-D0WD-V3 (rev3.1), 4M flash

### Machine Description

In progress

### Configuration file

```yaml
Tried using different official config files with I2S and making my own. Tried using I2S_STREAM and I2S_STATIC, but I know that in the v3.9.1 they should work the same way
```

### Startup Messages

```text
No errors, can "move" the machine and everything in the console works
```

### User Interface Software

ESP3D for FluidNC

### What happened?

This is my first time using FluidNC after GRBL, I am in progress of making a custom PCB and wanted to try using I2S mode to drive stepsticks.

My 74HC595 weren't responding so I checked I2S signals with an oscilloscope and found out that the BCK signal is not 16MHz as it chould be, but instead a strange 27MHz signal that periodically changes to 40MHz and back (photo)
[image]
Changing I2S pins in config does not help - the same signal comes up on all of them.

I downgraded to v3.8.3 and the problem went away: I am now getting correct 16MHz square wave

I tried searching for the same issue but since the v3.9.1 is only 2 weeks old, there are none

michmela44, 2024-11-13:
> P.S. Also found that in [Controller Design Guidlines](http://wiki.fluidnc.com/hardware/controller_design_guidelines) in GPIO Pins section there is a link to pin guidelines that points to old github wiki

This should be fixed now

MitchBradley, 2024-11-13:
Attach a specific config file that you used to do the test above.

MrTransistorsChannel, 2024-11-13:
Right now I am using this config with v3.8.3 and it works.
```yaml
board: ESP32
name: PCB mill/engraver
meta: "12-11-2024 v1"

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010

stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 400.000
    max_rate_mm_per_min: 1600.000
    acceleration_mm_per_sec2: 200.000
    soft_limits: false
    max_travel_mm: 200.000

    motor0:
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: I2SO.3:low

i2so:
  bck_pin: gpio.23
  data_pin: gpio.22
  ws_pin: gpio.21
```

Also I tried using official 6-pack stepstick xyzabc and it works fine with v3.8.3
```yaml
board: 6 Pack
name: 6 Pack StepStick XYZABC
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        ms3_pin: i2so.3
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        ms3_pin: i2so.6
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      stepstick:
        ms3_pin: i2so.11
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
        
  a:
    steps_per_mm: 53.400
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 960.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        ms3_pin: i2so.14
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

  b:
    steps_per_mm: 808.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.2
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        ms3_pin: i2so.19
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16
        
  c:      
    steps_per_mm: 808.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.25
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        ms3_pin: i2so.22
        step_pin: I2SO.21
        direction_pin: I2SO.20
        disable_pin: I2SO.23
        

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.36
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: i2so.24
  mist_pin: i2so.27
  delay_ms: 0

probe:
  pin: gpio.39
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

start:
  must_home: false

PWM:
  pwm_hz: 5000
  output_pin: gpio.14
  enable_pin: gpio.13
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
```
Both of the configs output this signal in v3.9.1, changing GPIOs does not help either

MitchBradley, 2024-11-13:
The clock periods change because the new code uses a fractional divisor to get a faster clock whose average frequency is a multiple of 16 Mhz.  If the signal quality is good, the shift registers should be able to cope with it since the data is synchronized with the clock edges.

Breadboarding I2S is tricky because it is hard to control ringing without a ground plane and good signal routing.  Making it work at 16 MHz is hard enough; I can imagine that 32 MHz would be very difficult to make work.  The new code, with its 32 MHz modulated-period clock, works fine on the various PCBs that we have tested.

MitchBradley, 2024-11-13:
Please note that we don't really have time to provide support for people making custom boards.  Supporting firmware usage on our own boards and a few other high-volume commercial boards takes more time than we have.

closed 2024-11-13 by MrTransistorsChannel

MitchBradley, 2024-11-13:
If you think it is a good idea, please add it.  We allow wiki-editing privileges on request.  <http://wiki.fluidnc.com/en/support/controller_design_guidelines#i2so-chips> provides recommendations and rationale.  I don't have time to write a book on what can happen if one chooses to ignore the recommendations.

<https://www.goodreads.com/quotes/190777-to-do-good-is-noble-to-tell-others-to-do>