bdring/FluidNC #1382: Problem:  The FluidNC stuck at saving after filled WiFi credential

notusenn, 2024-11-18:
### Wiki Search Terms

problem happened during installtion, firstly, the  V3.9.2 can't install into esp32 when selected WiFi - fresh-install.  got warning:
Was not able to flash device: Error: Failed to connect with the device
esptool.js
Serial port webSerial Vendor ID ox1a86 ProductID ox7522

I have trid hold down the boot button, no luck.

### Controller Board

cnc controller 6 axis breakout board 32bit GRBL ESP32 wifi control panel Z probe setter sensor for cnc router milling machine
https://vi.aliexpress.com/item/1005004644140352.html?spm=a2g0o.order_list.order_list_main.116.7c9c1802Ae4eFM&gatewayAdapt=glo2vnm

### Machine Description

no machine yet.
I have 6 axis board + stepper motor + driver;
I just hooked up one motor driver and one motor, with limit switch to do testing.

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack StepStick XYZABC
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 6
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000         
    max_rate_mm_per_min: 5000.000  
    acceleration_mm_per_sec2: 100.000  
    max_travel_mm: 300.000   
    soft_limits: false       
    homing:
      cycle: 1          
      positive_direction: false 
      mpos_mm: 0.000    
      feed_mm_per_min: 100.000   
      seek_mm_per_min: 200.000   
      settle_ms: 500           
      seek_scaler: 1.100 
      feed_scaler: 1.100 

    motor0:
      limit_neg_pin: NO_PIN   
      limit_pos_pin: NO_PIN  
      limit_all_pin: gpio.33:low:pu 
      hard_limits: true 
      pulloff_mm: 1.000 
      stepstick:
        ms3_pin: i2so.3
        step_pin: I2SO.2 
        direction_pin: I2SO.1 
        disable_pin: I2SO.0  

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.32:low
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        ms3_pin: i2so.6
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        ms3_pin: i2so.11
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
        
  a:
    steps_per_mm: 800
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 960.000
    soft_limits: false
    homing:
      cycle: -1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        ms3_pin: i2so.14
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

  b:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: -1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        ms3_pin: i2so.19
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16
        
  c:      
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle:-1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        ms3_pin: i2so.22
        step_pin: I2SO.21
        direction_pin: I2SO.20
        disable_pin: I2SO.23
        

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.36
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.16:low:pu 
  macro1_pin: gpio.14:low:pu 
  macro2_pin: gpio.25:low:pu 
  macro3_pin: gpio.26:low:pu 
coolant:
  flood_pin: i2so.24
  mist_pin: gpio.4
  delay_ms: 0

probe:
  pin: gpio.34:low
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=test.nc
  macro1: $SD/Run=test.nc
  macro2: $SD/Run=test.nc
  macro3: $SD/Run=test.nc

start:
  must_home: false

PWM:
  pwm_hz: 5000
  output_pin: gpio.27
  enable_pin: gpio.12
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
```

### Startup Messages

```text
$ss
[MSG:INFO: FluidNC v3.9.1 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6 Pack StepStick XYZABC]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:6us Idle Delay:250ms]
[MSG:INFO: Axis count 6]
[MSG:INFO: Axis X (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X All Limit gpio.33:low:pu]
[MSG:INFO: Axis Y (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Y All Limit gpio.32:low]
[MSG:INFO: Axis Z (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Z All Limit gpio.35:low]
[MSG:INFO: Axis A (150.000,1110.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO: Axis B (150.000,350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.18 Dir:I2SO.17 Disable:I2SO.16]
[MSG:INFO: Axis C (150.000,350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.21 Dir:I2SO.20 Disable:I2SO.23]
[MSG:INFO: feed_hold_pin gpio.36]
[MSG:INFO: macro0_pin gpio.16:low:pu]
[MSG:INFO: macro1_pin gpio.14:low:pu]
[MSG:INFO: macro2_pin gpio.25:low:pu]
[MSG:INFO: macro3_pin gpio.26:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:SUNMI_59]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: PWM Spindle Ena:gpio.12 Out:gpio.27 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant I2SO.24]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe gpio.34:low]
ok
```

### User Interface Software

WebUI

### What happened?

problem happened during installtion, firstly, the  V3.9.2 can't install into esp32 when selected WiFi - fresh-install.  got warning:
Was not able to flash device: Error: Failed to connect with the device
esptool.js
Serial port webSerial Vendor ID ox1a86 ProductID ox7522

 what I excepted is to run the testing of control axis moving and stop by CODE sequencely.

### GCode File

I also needs to learned how to upload GCODE.

bdring, 2024-11-18:
What does the supplier have to say about this? I have never heard about that controller. It is not worth spending much time on a $7 controller that will probably have a long list of issues. They don't even mention FluidNC, so it must be pretty old and they are just dumping them.

Get a controller [from this page](http://wiki.fluidnc.com/en/hardware/existing_hardware).

bdring, 2024-11-18:
We do not believe this is a FluidNC firmware issue. I am going to close this issue.

Read the wiki on alternate ways to try to install the firmware. If you want to continue the discussion, you can open an issue on the hardware forum of our Discord. The developers of FluidNC are not likely to participate unless we believe there is a FluidNC issue.

https://discord.com/channels/780079161460916227/1064635434417795164

closed 2024-11-18 by bdring

notusenn, 2024-11-18:
Actually, the supplier said it is compatible with FluidNC, and it is in their decription page.

notusenn, 2024-11-18:
my unit I mean 6 axis board + TB6600 + Nema17, did work by click the X HOME button to run motor, and stop it by limit switch. 

I just can't make sure if the WiFi is setup or NOT, how to find it? 

and next step I will try to use GCODE for testing.

notusenn, 2024-11-18:
Thanks.
I'm new to FluidNC, can I think the WiFi is setup successed if I can operate the 6 axis board by Phone through the FluidNC AP?

MitchBradley, 2024-11-18:
The following startup lines show the WiFi setup

[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]

If you can access FluidNC from a phone, the wifi is working.

It is better to use STA mode to connect to an external WiFi network because AP mode on the ESP32 does not work well for long-term use.  The WebUI Setup wizard will guide you through the process of connecting to external wifi

[image]

MitchBradley, 2024-11-18:
STA mode is when FluidNC is connected to an external WiFi network instead of creating its own network.  Like pretty much every other WiFi device in the world.  You enter the SSID of your WiFi router and its password
[image]
Then reset FluidNC and it will try to connect to that WiFi.  The startup messages that you can see on a serial console will show the attempt to connect and if successful, the IP address on that network that the router assigned.  Like this:
```
[MSG:INFO: Connecting to STA SSID:CampOlindaHS]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.4.68]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
```
You can then access the WebUI by connecting your phone to the same WiFi and using either the IP address shown by the messages, or "fluidnc.local".

I'm going to lock this conversation because these questions are very basic and answered in the wiki.  If you need further handholding, join the FluidNC Discord server and ask there.  Maybe other users who are not as busy as us developers can help you understand the documentation that we spent a ridiculous amount of time writing.