bdring/FluidNC #1384: Engine blocking

FMD14, 2024-11-23:
### Wiki Search Terms

Z Motor Lock

### Controller Board

MKS DLC32

### Machine Description

Laser core XY  Z

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: CoreXY Laser NJE A64640
meta: 09/11/2024 FMD14
kinematics: 
  corexy:
stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: i2so.0
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 25000
    acceleration_mm_per_sec2: 3200
    max_travel_mm: 520
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 300
      seek_mm_per_min: 5000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 5
      stepstick:
        step_pin: i2so.1
        direction_pin: i2so.2:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 25000
    acceleration_mm_per_sec2: 3200
    max_travel_mm: 550
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 300
      seek_mm_per_min: 5000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.36:low
      hard_limits: true
      pulloff_mm: 5
      stepstick:
        step_pin: i2so.5
        direction_pin: i2so.6:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
  z:
    steps_per_mm: 53.628
    max_rate_mm_per_min: 1200
    acceleration_mm_per_sec2: 500
    max_travel_mm: 45
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 300
      seek_mm_per_min: 1000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.34:low
      hard_limits: true
      pulloff_mm: 5
      stepstick:
        step_pin: i2so.3
        direction_pin: i2so.4:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14
sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:pu:low
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
macros:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
coolant:
  flood_pin: NO_PIN
  mist_pin: gpio.04:low
  delay_ms: 0
probe:
  pin: gpio.22
  check_mode_start: true
  toolsetter_pin: NO_PIN
Laser:
  pwm_hz: 5000
  output_pin: gpio.32
  enable_pin: i2so.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000%  1000=100.000%
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
start:
  must_home: false
```

### Startup Messages

```text
board: MKS-DLC32 V2.1
name: CoreXY Laser NJE 30W
meta: 09/11/2024 FMD14
kinematics: 
  corexy:
stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: i2so.0
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 25000
    acceleration_mm_per_sec2: 3200
    max_travel_mm: 520
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 300
      seek_mm_per_min: 5000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 5
      stepstick:
        step_pin: i2so.1
        direction_pin: i2so.2:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 25000
    acceleration_mm_per_sec2: 3200
    max_travel_mm: 550
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 300
      seek_mm_per_min: 5000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.36:low
      hard_limits: true
      pulloff_mm: 5
      stepstick:
        step_pin: i2so.5
        direction_pin: i2so.6:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
  z:
    steps_per_mm: 53.628
    max_rate_mm_per_min: 1200
    acceleration_mm_per_sec2: 500
    max_travel_mm: 45
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 300
      seek_mm_per_min: 1000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.34:low
      hard_limits: true
      pulloff_mm: 5
      stepstick:
        step_pin: i2so.3
        direction_pin: i2so.4:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14
sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:pu:low
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
macros:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
coolant:
  flood_pin: NO_PIN
  mist_pin: gpio.04:low
  delay_ms: 0
probe:
  pin: gpio.22
  check_mode_start: true
  toolsetter_pin: NO_PIN
Laser:
  pwm_hz: 5000
  output_pin: gpio.32
  enable_pin: i2so.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000%  1000=100.000%
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
start:
  must_home: false
```

### What happened?

Good morning 
I have just modified my laser by adding the motorized axis. With a rack system but with the weight of the laser it descends after homing. I blocked the motors by setting 255 idle_ms. My question is can we prevent the Z motor from moving. I noticed that by setting 255 the Z motor heats up

bdring, 2024-11-23:
Not in firmware. You must change your machine. You can use motor drivers with an idle current like anything from trinamic. TMC2209, etc. That would likely require a different controller.

Maybe you could add a counterweight or spring.

FMD14, 2024-11-23:
I use TMC2208 So if I understood correctly it TMC2209

MitchBradley, 2024-11-23:
Note that he said a different controller, because MKS DLC32 does not have the wiring to configure the special current settings on a TMC2209.

One possibility would be to use an external driver for the Z axis that can handle the current without getting hot.  Another would be to add a cooling fan to keep the TMC2208 cooler.  Heat sinks are much less effective than fans.

bdring, 2024-11-23:
I do not know if or how the standalone Trinamic drivers handle standby current. Here is a list of alternative controllers.

http://wiki.fluidnc.com/en/hardware/existing_hardware

I am going to close this issue because it is not a FluidNC issue. You might find help from other MKS users on our Discord.

closed 2024-11-23 by bdring

FMD14, 2024-11-23:
So I have to change my rack system with a screw. It's the motor that's heating up; there's a fan above the card.