bdring/FluidNC #1387: $Start/Message doesn't change all displayed firmware version numbers

daxliniere, 2024-11-26:
### Wiki Search Terms

N/A

### Controller Board

N/A

### Machine Description

N/A

### Configuration file

```yaml
N/A
```

### Startup Messages

```text
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.0.x (FixPlus-a32983c) https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine SWOLE-CNC]
[MSG:INFO: Board MKS-DLC32]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:150ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,301.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2:low Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO:  X Pos Limit gpio.34]
[MSG:INFO: Axis Y (0.000,186.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO:  Y Pos Limit gpio.23]
[MSG:INFO: Axis Z (-100.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4:low Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.18]
[MSG:INFO: safety_door_pin gpio.33:low:pu]
[MSG:INFO: macro0_pin gpio.19:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: 10V Spindle Ena:gpio.5:low Out:gpio.32 Dir:NO_PIN Fwd:NO_PIN Rev:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant gpio.0]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe gpio.22:low]

Grbl 3.0 [FluidNC 4.0.0 (wifi) '$' for help]
```

### User Interface Software

FluidTerm

### What happened?

Joacim had the idea to fool UGS when running a test version of FLC. UGS expects 3.3.0 or higher. I used **$Start/Message** to sert FNC v4.0.0 (see last line in startup messages), but the first line still says 3.x.0

MitchBradley, 2024-11-26:
msg:info is not the start message

daxliniere, 2024-11-26:
No worries. Thought we had a workaround.

closed 2024-11-26 by daxliniere

MitchBradley, 2024-11-26:
If there is important knowledge that can only be gained from info messages, we need to have a discussion about a more formal way to present it, as part of the protocol specification.  The INFO tag is for stuff that is explicitly not guaranteed to be present or stable in its format.