bdring/FluidNC #1388: During startup, only one hard limit axis is shown at a time.

daxliniere, 2024-11-26:
### Wiki Search Terms

N/A

### Controller Board

N/A

### Machine Description

N/A

### Configuration file

```yaml
N/A
```

### Startup Messages

```text
[MSG:RST]
[MSG:INFO: FluidNC v3.9.1 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine SWOLE-CNC]
[MSG:INFO: Board MKS-DLC32]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:150ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,301.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2:low Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO:  X Pos Limit gpio.34]
[MSG:INFO: Axis Y (0.000,186.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO:  Y Pos Limit gpio.23]
[MSG:INFO: Axis Z (-100.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4:low Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.18]
[MSG:INFO: safety_door_pin gpio.33:low:pu]
[MSG:INFO: macro0_pin gpio.19:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: 10V Spindle Ena:gpio.5:low Out:gpio.32 Dir:NO_PIN Fwd:NO_PIN Rev:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant gpio.0]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe gpio.22:low]

Grbl 3.9 [FluidNC v3.9.1 (wifi) '$' for help]
[MSG:WARN: Active limit switch on X axis motor 0]
[MSG:INFO: ALARM: Hard Limit]
ALARM:1
[MSG:ERR: Reset to continue]
Resetting MCU
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.9.1 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine SWOLE-CNC]
[MSG:INFO: Board MKS-DLC32]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:150ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,301.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2:low Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO:  X Pos Limit gpio.34]
[MSG:INFO: Axis Y (0.000,186.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO:  Y Pos Limit gpio.23]
[MSG:INFO: Axis Z (-100.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4:low Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.18]
[MSG:INFO: safety_door_pin gpio.33:low:pu]
[MSG:INFO: macro0_pin gpio.19:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: 10V Spindle Ena:gpio.5:low Out:gpio.32 Dir:NO_PIN Fwd:NO_PIN Rev:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant gpio.0]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe gpio.22:low]

Grbl 3.9 [FluidNC v3.9.1 (wifi) '$' for help]
[MSG:WARN: Active limit switch on Y axis motor 0]
[MSG:INFO: ALARM: Hard Limit]
ALARM:1
[MSG:ERR: Reset to continue]
Resetting MCU
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.9.1 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine SWOLE-CNC]
[MSG:INFO: Board MKS-DLC32]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:150ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,301.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2:low Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO:  X Pos Limit gpio.34]
[MSG:INFO: Axis Y (0.000,186.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO:  Y Pos Limit gpio.23]
[MSG:INFO: Axis Z (-100.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4:low Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.18]
[MSG:INFO: safety_door_pin gpio.33:low:pu]
[MSG:INFO: macro0_pin gpio.19:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: 10V Spindle Ena:gpio.5:low Out:gpio.32 Dir:NO_PIN Fwd:NO_PIN Rev:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant gpio.0]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe gpio.22:low]

Grbl 3.9 [FluidNC v3.9.1 (wifi) '$' for help]
```

### User Interface Software

FT

### What happened?

I had a hard limit situation on both X and Y axes during testing. FluidTerm informed me that a hard limit was triggered on X axis. I moved the axis off the switch and CTRL+R as ther terminal suggests. After reboot, I then got a warning that the Y axis limit switch was also triggered.

It would be best if all triggered axes/errors were reported in the startup message, otherwise we have to reboot multiple times if we don't notice an axis has triggered a limit.

daxliniere, 2024-11-26:
Hmm... with 3.8.2 I can see the error for both axes:

```Grbl 3.8 [FluidNC v3.8.2 (noradio) '$' for help]
[MSG:WARN: Active limit switch on X axis motor 0]
[MSG:WARN: Active limit switch on Y axis motor 0]
[MSG:INFO: ALARM: Hard Limit]
ALARM:1
[MSG:ERR: Reset to continue]
```

bdring, 2024-12-10:
It reported fine with 3.9.1 and I just tested it with v3.9.2-pre2.

All switches on an XYYZ machine were made active on startup.

```
Grbl 3.9 [FluidNC v3.9.2-pre2 (wifi) '$' for help]
[MSG:WARN: Active limit switch on X axis motor 0]
[MSG:WARN: Active limit switch on Y axis motor 0]
[MSG:WARN: Active limit switch on Y axis motor 1]
[MSG:WARN: Active limit switch on Z axis motor 0]
[MSG:INFO: ALARM: Unhomed]
ALARM:14
```

bdring, 2024-12-11:
Can you test with the latest pre release?

daxliniere, 2024-12-12:
Will do tomorrow or day after.

⁣Sent from my mobile office.​

bdring, 2025-01-21:
Please test. I want to close this issue.