bdring/FluidNC #1389: 3.9.1 & 3.9.2pre1 - Direction not working

daxliniere, 2024-11-26:
### Wiki Search Terms

N/A

### Controller Board

MKS DLC32

### Machine Description

3 axis mill, 3x MKS Servo57c closed-loop drivers on NEMA23 motors

### Configuration file

```yaml
board: MKS-DLC32
name: SWOLE-CNC
meta: V2.6.9 (2202.11.09) Dax Liniere

kinematics:
  Cartesian:

stepping:
  engine: I2S_STREAM
#  engine: I2S_STATIC
#was I2S_STREAM, changed when FNC 3.8.4 was released
  idle_ms: 150
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
  
axes:
  shared_stepper_disable_pin: I2SO.0
  x:
    steps_per_mm: 803.144
#was 802.341
    max_rate_mm_per_min: 1600.000
#was 1500.000
    acceleration_mm_per_sec2: 80.000
    max_travel_mm: 301.000
#was 302.000    
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 600.000
#was 300
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36
      limit_pos_pin: gpio.34
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.1
        direction_pin: I2SO.2:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

  y:
    steps_per_mm: 802.322
#was 802.322
    max_rate_mm_per_min: 1600.000
    acceleration_mm_per_sec2: 80.000
# was 300.000
    max_travel_mm: 186.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 600.000
#was 300      
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: gpio.23
#this pin doesn't have the same input circuitry as the 3 dedicated on-board limit switch inputs
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.5
        direction_pin: I2SO.6:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN
  z:
    steps_per_mm: 802.497
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 50.000
# was 500.000
    max_travel_mm: 100
#46.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
#was 200      
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_pos_pin: gpio.18
#this pin doesn't have the same input circuitry as the 3 dedicated on-board limit switch inputs
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        step_pin: I2SO.3
        direction_pin: I2SO.4:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
# This could be GPIO.39, but Card Detect has no supported functions in FluidNC

control:
  safety_door_pin: gpio.33:low:pu
  cycle_start_pin: NO_PIN
  feed_hold_pin: NO_PIN
  reset_pin: NO_PIN
  macro0_pin: gpio.19:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

parking:
  enable: true
  axis: Z
  pullout_distance_mm: 5.000
  pullout_rate_mm_per_min: 250.000
  target_mpos_mm: 0.000
  rate_mm_per_min: 800.000
  
macros:
  startup_line0:
  startup_line1:
  macro0: $H
#unlock and home XYZ
  macro1: G38.2 G91 Z-46 F100&G0 Z1&G38.2 G91 Z-2 F50&G10 L20 P1 Z4.985&G10 L20 P2 Z4.985&G10 L20 P3 Z4.985&G10 L20 P4 Z4.985&G10 L20 P5 Z4.985&G10 L20 P6 Z4.985&G53 G0 Z0&G90
#z-probe
  macro2:
  macro3: $SD/Run=lasertest.gcode

coolant:
  flood_pin: gpio.0
# continuous air
  mist_pin:  gpio.4
# pulsed air 
  delay_ms: 0

probe:
  pin: gpio.22:low
  check_mode_start: true

10V:
# Spindle
  direction_pin: NO_PIN
  forward_pin: NO_PIN
  reverse_pin: NO_PIN
  output_pin: gpio.32
  enable_pin: gpio.5:low
  pwm_hz: 5000
  disable_with_s0: false
  s0_with_disable: false
  off_on_alarm: true
#switch off spindle when in alarm state
  spinup_ms: 10000
  spindown_ms: 12000
  tool_num: 0
  speed_map: 0=0% 0=25% 5868=25% 23237=99% 24000=100%
#DLC32 Spindle TTL max output voltage is 4.89v, so it can never reach 24000rpm/400Hz

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false
# Pins that could be used: 0(SDA), 4(SCL), 5, 18, 19, 22, 23, 25, 26, 27, 32 (TTL spindle control), 33, 39, I2SO.7
# Pins not currently in use: 25 (TMC_CS_Y), 26 (TMC_CS_X), 27 (TMC_CS_Z), (39 SD card detect), I2SO.7
```

### Startup Messages

```text
rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.8.2 https://github.com/bdring/FluidNC] ***It's the same for 3.9.1, but the motors start moving and jam against limit switches so I need to switch off the system before it can get past connecting to STA***
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine SWOLE-CNC]
[MSG:INFO: Board MKS-DLC32]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:150ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,301.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2:low Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.36]
[MSG:INFO:  X Pos Limit gpio.34]
[MSG:INFO: Axis Y (0.000,186.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.35]
[MSG:INFO:  Y Pos Limit gpio.23]
[MSG:INFO: Axis Z (-100.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4:low Disable:NO_PIN]
[MSG:INFO:  Z Pos Limit gpio.18]
[MSG:INFO: safety_door_pin gpio.33:low:pu]
[MSG:INFO: macro0_pin gpio.19:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:Puzzle Factory]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.2.81]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: 10V Spindle Ena:gpio.5:low Out:gpio.32 Dir:NO_PIN Fwd:NO_PIN Rev:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant gpio.0]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe gpio.22:low]
```

### User Interface Software

Any

### What happened?

Negative control command = negative movement. Positive control command = negative movement.

I thought I was losing my mind, but once I reverted to 3.8.2, the problem vanished.

daxliniere, 2024-11-27:
Ahh, I think I know the problem. I'll guess that it's something to do with **:low** which is not being read. It would also explain the weird behaviour with door switch and macro button.

bdring, 2024-11-27:
The direction is being read. You can see it in your start messages.

```
[MSG:INFO: safety_door_pin gpio.33:low:pu]
[MSG:INFO: macro0_pin gpio.19:low:pu]
```
I am using the latest firmware and both switches and direction pins read and function correctly when :low :high are swapped.

daxliniere, 2024-11-27:
Hi @bdring,

> The direction is being read. You can see it in your start messages.
Perhaps read, but not being applied?
I can assure you that switching to the latest firmwares resulted in my axes only being able to be driven in one direction. I have seen the crashes during homing with my own eyes. After reverting to an old version and changing nothing else, both directions are working again.
If there is a debug version I can install to help track this down, I'm happy to do so.

Yesterday I fired up my CNC again after several months, updated FNC and UGS but I was experiencing a bunch of problems, including door switch getting stuck in the triggered state even after closing the switch. I think if my suggestion is possible, it would probably explain that as well.

MitchBradley, 2024-11-27:
I am going to look into it, but as I mentioned, I am really busy with paying work at the moment.

bdring, 2024-11-27:
Try swapping the active state in the config file on the door switch. Check the response to the ? command for each version of the config file in both the open and closed state.

Getting "Stuck in" a state may be unrelated to the :high or :low value.

jappiemike, 2024-12-10:
Hello, I appear to be having a similar problem, sorry I'm not near my cnc laptop to give you the yaml or startup files. I upgraded to 3.9.1 so I could use my rapidchange magazine with the .nc files from Rocco on the Rapidchange discord site. When I use homing: false  and do a manual homing my X axis reports 480 when it should be 0, the 480 is the X length in config.yaml. The X axis then only goes in the negative direction. If I set homing to true the problem goes away but I cannot use my toolchanger.

bdring, 2024-12-10:
@jappiemike Create your own issue. We need to see all the info and don't want to get it confused with OP data.

MitchBradley, 2024-12-10:
@jappiemike When you create the issue, please use accurate wording.  "homing: false" is not a possible setting, since "homing:" is a section name, not an item.  We can try to guess what you actually mean, but if we get it wrong, we will waste a lot of time that we don't have.

bdring, 2024-12-13:
@daxliniere Can you test the latest pre release?

daxliniere, 2024-12-13:
Hey Bart,
I loaded 3.9.2pre2 and then ran my homing macro and immediately the Z-axis started moving DOWN instead of up.
I hit the door switch (my e-stop) and stopped motion.

This is the console from UGS:

```
*** Resetting controller
*** Fetching device status
*** Fetching device status (2 of 10)...
*** Fetching device firmware version
*** Fetching device status codes
*** Fetching device state
*** Fetching device settings
*** Connected to FluidNC 3.6.9
> $J=G21G91Z-3F300
ok
> $J=G21G91Z-3F300
ok
> $J=G21G91Z3F300
ok
> $J=G21G91Z-3F300
ok
*** Connection closed
*** Connecting to jserialcomm://COM13:115200
*** Fetching device status
*** Fetching device firmware version
*** Fetching device status codes
*** Fetching device state
*** Fetching device settings
*** Connected to FluidNC 3.9.2
> $H
ALARM:7
```

bdring, 2024-12-13:
Can you move the Z up or does it always go down?

daxliniere, 2024-12-13:
Video incoming, this is pertinent to that.

```
> $H
ALARM:9
> $J=G21G91Z-0.05F300
ok
> $J=G21G91Z-0.05F300
ok
> $J=G21G91Z-0.05F300
ok
> $J=G21G91Z-0.05F300
ok
> $J=G21G91Z-0.05F300
ok
> $J=G21G91Z-0.05F300
ok
> $J=G21G91Z-0.05F300
ok
> $J=G21G91Z-0.05F300
ok
> $J=G21G91Z-0.05F300
ok
> $J=G21G91Z-0.05F300
ok
> $J=G21G91Z-0.05F300
ok
> $J=G21G91Z-0.05F300
ok
> $J=G21G91Z-0.05F300
ok
> $J=G21G91Z-0.05F300
ok
> $J=G21G91Z-0.05F300
ok
> $J=G21G91Z-0.05F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $H=Z
ALARM:8
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $H=Z
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91X99Y-99F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
```

daxliniere, 2024-12-13:
Ahhhh shit. I was filming but had paused without realising. I would have caught it going up with Z+ for a few manual steps from UGS, THEN DOWN, then up again. (There is no chance I accidentally hit the Z- button, it just went on its own.)

bdring, 2024-12-13:
Just jog up and down with one jog each way. Use the console rather than buttons.

daxliniere, 2024-12-13:
I got it on camera! This is the console from UGS. As you can see, it's sending acceptable commands.

```
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z-1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
> $J=G21G91Z1F300
ok
```

daxliniere, 2024-12-13:
https://youtu.be/d14wInmZR0I

daxliniere, 2024-12-13:
Shorter video:
https://youtu.be/VQvJi1ns2P0

MitchBradley, 2024-12-14:
Try increasing dir_delay_us.  You could also, separately, try inverting the step pulse.

daxliniere, 2024-12-14:
Hey Mitch. I'm happy to do either of those things, but any idea why this would be different/necessary between the two firmware versions? 
I don't believe the streaming changes were implemented until some time after 3.8.2 and this version is affected by the bug.

bdring, 2024-12-14:
All data helps us figure out the problem.

MitchBradley, 2024-12-14:
The idea is that the I2S clock speedup may have improved the time resolution of the dir delay, so it is effectively shorter than before, instead of being rounded up to a multiple of the older, slower clock.

The information that 3.8.2 fails is not spelled out clearly in this ticket, whose title refers to 3.9.1.  If you look carefully at the startup messages and at the long comment that is mostly off the screen unless you scroll horizontally, you can maybe see something about 3.8.2, but the meaning doesn't exactly jump out at you.

MitchBradley, 2024-12-14:
I looked at the source code for the MKS Servo57 board.  It is a C program running on an ARM processor.  The DIR pin is handled by an interrupt, so a software-mediated action, while the STEP pin clocks a hardware counter.  The time relationship between those two processes is hard to figure out without very deep system analysis.  I can easily imagine that 1 microsecond is not enough to guarantee that the DIR pin is recognized soon enough.  This is yet another example where the extreme diversity of the barely-supported, barely-documented components that can be used with the FluidNC ecosystem causes us to spend ridiculous amounts of support time.

bdring, 2024-12-14:
Would that only account for 1 step the wrong way?

MitchBradley, 2024-12-14:
I don't know.  It depends on when and how the code ultimately takes into account the pin change.  Here is the code if you want to analyze it https://github.com/makerbase-mks/MKS-SERVO42B/blob/master/firmware

One diagnostic would be to put an LED on the DIR signal to see if it is in fact incorrect during the suspicious movement.  Another would be to hook up a different motor/driver to that axis, to see if it happens without a Servo57 - but you might still need more dir delay even with a different driver.

If the OP had just done what was asked instead of pushing back, we might be farther along.

daxliniere, 2024-12-14:
Hey Mitch, not sure where I was pushing back? I just looked through this issue and couldn't see anything.

The discovery that the problem existed in 3.8.2 was only made after further investigations that came after I made this post.

Adding LEDs at the servo seems like a great step. I think there would also be a lot of value in pinning down the exact version change that introduced this bug.
I will download and install each version between 3.6.9 and 3.8.2 on Monday when I'm back in the workshop.

MitchBradley, 2024-12-14:
Questioning the utility of my suggestion of increasing dir_delay_us, instead of just trying it, was "pushing back".  That caused me to spend a lot of time explaining my thought process and we still don't have the result of the proposed experiment.

Bisecting versions might be helpful ultimately, but the only guess that I have right now is that the servo57 can't handle the tighter timing.  If that guess turns out to be correct, I don't really care which version changed the timing, because it means that the solution is just to fix the config file to be less aggressive.

daxliniere, 2024-12-14:
> Questioning the utility of my suggestion of increasing dir_delay_us, instead of just trying it, was "pushing back".
Mate, it's the weekend, I'm not at my workshop.

> Bisecting versions might be helpful ultimately, but the only guess that I have right now is that the servo57 can't handle the tighter timing. If that guess turns out to be correct, I don't really [...]

Sure, but the problem occurred before you changed improved I2S, so that seems to suggest it is unrelated to timing, no?

MitchBradley, 2024-12-14:
https://youtube.com/shorts/Cz_opgrN8nM?si=TxoAf8yRnem3f4NM

I could not make it fail

bdring, 2024-12-14:
I like the LED idea. 

It feels like a timing or or circuit sensitivity. We recently solved a 485 issue that worked on only some revs with a circuit change.

daxliniere, 2024-12-14:
> I could not make it fail. 

Is there any chance that when I run **install-wifi.bat** something is not being cleared? While I've been troubleshooting, I think I may have run **erase.bat** as well, at some point.

Let's see what the LED tests show when I'm back on Monday.

bdring, 2024-12-28:
Closing this issue. See similar issue #1408

closed 2024-12-28 by bdring