bdring/FluidNC #1391: Problem: GCODE visualizer bad work

CodeNameHawk, 2024-12-02:
### Wiki Search Terms

N/A

### Controller Board

6-pack

### Machine Description

6 Pack StepStick XYZABC

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack StepStick XYZABC
stepping:
  engine: I2S_STREAM
  idle_ms: 250
  pulse_us: 4
  dir_delay_us: 6
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000         
    max_rate_mm_per_min: 5000.000  
    acceleration_mm_per_sec2: 100.000  
    max_travel_mm: 300.000   
    soft_limits: false       
    homing:
      cycle: 1          
      positive_direction: false 
      mpos_mm: 0.000    
      feed_mm_per_min: 100.000   
      seek_mm_per_min: 200.000   
      settle_ms: 500           
      seek_scaler: 1.100 
      feed_scaler: 1.100 

    motor0:
      limit_neg_pin: NO_PIN   
      limit_pos_pin: NO_PIN  
      limit_all_pin: gpio.33:low:pu 
      hard_limits: true 
      pulloff_mm: 1.000 
      stepstick:
        ms3_pin: i2so.3
        step_pin: I2SO.2 
        direction_pin: I2SO.1 
        disable_pin: I2SO.0  

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.32:low
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        ms3_pin: i2so.6
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 1.000
      stepstick:
        ms3_pin: i2so.11
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
        
  a:
    steps_per_mm: 800
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 960.000
    soft_limits: false
    homing:
      cycle: -1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        ms3_pin: i2so.14
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

  b:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle: -1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        ms3_pin: i2so.19
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16
        
  c:      
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 200.000
    soft_limits: false
    homing:
      cycle:-1
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      stepstick:
        ms3_pin: i2so.22
        step_pin: I2SO.21
        direction_pin: I2SO.20
        disable_pin: I2SO.23
        

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.36
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.16:low:pu 
  macro1_pin: gpio.14:low:pu 
  macro2_pin: gpio.25:low:pu 
  macro3_pin: gpio.26:low:pu 
coolant:
  flood_pin: i2so.24
  mist_pin: gpio.4
  delay_ms: 0

probe:
  pin: gpio.34:low
  check_mode_start: true

macros:
  startup_line0:
  startup_line1:
  macro0: $SD/Run=test.nc
  macro1: $SD/Run=test.nc
  macro2: $SD/Run=test.nc
  macro3: $SD/Run=test.nc

start:
  must_home: false

PWM:
  pwm_hz: 5000
  output_pin: gpio.27
  enable_pin: gpio.12
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 1000=100.000%
```

### Startup Messages

```text
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.9.1 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine 6 Pack StepStick XYZABC]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:6us Idle Delay:250ms]
[MSG:INFO: Axis count 6]
[MSG:INFO: Axis X (0.000,300.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X All Limit gpio.33:low:pu]
[MSG:INFO: Axis Y (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  Y All Limit gpio.32:low]
[MSG:INFO: Axis Z (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Z All Limit gpio.35:low]
[MSG:INFO: Axis A (150.000,1110.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO: Axis B (150.000,350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.18 Dir:I2SO.17 Disable:I2SO.16]
[MSG:INFO: Axis C (150.000,350.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.21 Dir:I2SO.20 Disable:I2SO.23]
[MSG:INFO: feed_hold_pin gpio.36]
[MSG:INFO: macro0_pin gpio.16:low:pu]
[MSG:INFO: macro1_pin gpio.14:low:pu]
[MSG:INFO: macro2_pin gpio.25:low:pu]
[MSG:INFO: macro3_pin gpio.26:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:Siedem]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.3]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: PWM Spindle Ena:gpio.12 Out:gpio.27 Dir:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant I2SO.24]
[MSG:INFO: Mist coolant gpio.4]
[MSG:INFO: Probe gpio.34:low]

Grbl 3.9 [FluidNC v3.9.1 (wifi) '$' for help]
```

### User Interface Software

WebUI

### What happened?

Draws the tool path line poorly.
The tool moves along the correct trajectory, but the drawing looks wrong.
Here is an example of how it draws a square.
The movement X100 and Y-100 is shown normally, but X-100 and Y100 are double the length.

### GCode File

%
G21
G90 X0 Y0
G91
G1 F1200 X100.0
G1 Y-100.0
G1 X-100.0
G1 Y100.0
M30
%

### Other Information

[image]

MitchBradley, 2024-12-02:
https://github.com/bdring/FluidNC/blob/242d6ba59eb5b244f205c81db38f9652604e6d97/FluidNC/data/index.html.gz fixes it.

MitchBradley, 2024-12-02:
Fixed

closed 2024-12-02 by MitchBradley

CodeNameHawk, 2024-12-03:
Thx. Work.
[image]