bdring/FluidNC #1395: Problem: Servo direction pins do not seem to work.

MrMagoo-lab, 2024-12-07:
### Wiki Search Terms

spindle direction pins

### Controller Board

6-pack

### Machine Description

Prototyping a system replicating a Langmuir MR-1,

### Input Circuits

```yaml
N/A
```

### Configuration file

```yaml
MR1_V3_x1yz.yaml
client $Localfs/Show=MR1_V3_x1yz.yaml
name: MR1_V3
board: 6 Pack
start:
  must_home: true
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 10
  dir_delay_us: 0
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 320
    max_rate_mm_per_min: 2540.000
    acceleration_mm_per_sec2: 10.000
    max_travel_mm: 580.000
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 762
      feed_mm_per_min: 100
      seek_scaler: 1.5
      feed_scaler: 1.5
    motor0:
      limit_neg_pin: gpio.2:high:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      pulloff_mm: 1.5
      standard_stepper:
        step_pin: i2so.2
        direction_pin: i2so.1
        disable_pin: i2so.0:high
  y:
    steps_per_mm: 320
    max_rate_mm_per_min: 2540.000
    acceleration_mm_per_sec2: 10.000
    max_travel_mm: 580.000
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 762
      feed_mm_per_min: 100
      seek_scaler: 1.5
      feed_scaler: 1.5
    motor0:
      limit_neg_pin: gpio.26:high:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      pulloff_mm: 1.5
      standard_stepper:
        step_pin: i2so.5
        direction_pin: i2so.4
        disable_pin: i2so.7:high
  z:
    steps_per_mm: 320
    max_rate_mm_per_min: 2540.000
    acceleration_mm_per_sec2: 10.000
    max_travel_mm: 580.000
    homing:
      cycle: 2
      mpos_mm: 0
      positive_direction: false
      seek_mm_per_min: 762
      feed_mm_per_min: 100
      seek_scaler: 1.5
      feed_scaler: 1.5
    motor0:
      limit_neg_pin: gpio.32:high:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      pulloff_mm: 1.5
      standard_stepper:
        step_pin: i2so.13
        direction_pin: i2so.12
        disable_pin: i2so.15:high
coolant:
  flood_pin: gpio.12
  mist_pin: NO_PIN
  delay_ms: 0
probe:
  pin: gpio.39:low
  toolsetter_pin: gpio.36:low
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
10V:
  forward_pin: gpio.15
  reverse_pin: gpio.14
  output_pin: gpio.13
  speed_map: 0=0.000% 100=0.000% 8000=100.000%
```

### Startup Messages

```text
No error messages.
```

### User Interface Software

WebUI, GSender

### What happened?

Send M3 S1000 command.
io.13 generates 0.8 volt signal which is fine.
The led in the center of the board lights up as expected.
However, pin io.15 never changes state. I expected this to do something. Then I found this on the wiki.
[image]
My original expectation was that pins 14 and 15 were outputs. I considered that they might be optically coupled outputs.
That said, the way the schematic is drawn, pins 14 and 15 are shown as inputs feeding some OCs that go to TB13 which the six pack does not seem to have.
The servo I am using needs the direction pins to take 24VDC from an optocoupler to ground to turn it on for that direction.
Again, as drawn, the figure shows the direction pin as inputs and that just won't work.
I am guessing a little here that the figure provided is incorrect.
Can you please provide a corrected schematic?

### Other Information

[VER:3.8 FluidNC v3.8.3:]
[OPT:PHS]
[MSG:Machine: MR1_V3]
[MSG:Mode=AP:SSID=FluidNC:IP=192.168.0.1:MAC=E4-65-B8-1E-07-B1]
ok

$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Telnet/Enable=ON
$Telnet/Port=23
$HTTP/BlockDuringMotion=ON
$HTTP/Enable=ON
$HTTP/Port=80
$MDNS/Enable=ON
$WiFi/PsMode=None
$WiFi/Mode=STA>AP
$Sta/Password=********
$Sta/MinSecurity=WPA2-PSK
$WiFi/FastScan=OFF
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$AP/Country=01
$AP/SSID=FluidNC
$AP/Password=********
$AP/IP=192.168.0.1
$AP/Channel=1
$Hostname=fluidnc
$Sta/SSID=
$Grbl/HomingCycleEnable=1
$Grbl/HardLimitsEnable=0
$Grbl/SoftLimitsEnable=0
$Grbl/Resolution/Z=320.000
$Grbl/Resolution/Y=320.000
$Grbl/Resolution/X=320.000
$Grbl/MaxRate/Z=2540.000
$Grbl/MaxRate/Y=2540.000
$Grbl/MaxRate/X=2540.000
$Grbl/Acceleration/Z=10.000
$Grbl/Acceleration/Y=10.000
$Grbl/Acceleration/X=10.000
$Grbl/MaxTravel/Z=580.000
$Grbl/MaxTravel/Y=580.000
$Grbl/MaxTravel/X=580.000
$GCode/Echo=OFF
$Start/Message=Grbl \V [FluidNC \B (\R) \H]
$Firmware/Build=
$SD/FallbackCS=-1
$Report/Status=1
$Config/Filename=MR1_V3_x1yz.yaml
$Message/Level=Info

bdring, 2024-12-11:
Most VFD direction signals work by connecting a signal to a common ground. The optos provide a simple and isolated way to do this.

[image]

MrMagoo-lab, 2024-12-11:
Please see attached.

Thank you.

MitchBradley, 2024-12-11:
The attachment did not make it into the github issue.   You many need to do it directly on github instead of by email.

MrMagoo-lab, 2024-12-12:
Mitch,

Please let me know if this attachment comes through.

Thank you,

Beth Schmitz

bdring, 2024-12-12:
Do not use email. Go to this link on Github
https://github.com/bdring/FluidNC/issues/1395

bdring, 2024-12-12:
When you say "Servo direction pins" do you mean the ***spindle?*** Can you give use the device information and a link to the user manual?

bdring, 2024-12-28:
@MrMagoo-lab   Please respond or this issue will be closed. We do not believe this is a FluidNC issue.

bdring, 2025-01-02:
Unresponsive OP

closed 2025-01-02 by bdring

DVA85, 2025-01-12:
Bonsoir, j'ai un VDF YL620 et je n'arrive pas à faire démarrer/arrêter la broche...io.13 s'allume, io.14 et io.15 s'allument, j'ai connecté depuis la carte vers le VDF comme suit...GND/GND...io.14/Rev...io.15/Fwd...10v/10v....et un pontage sur la carte depuis GND/COM...paramètres VDF...P00.01=1.......P07.08=3..........mais je n'ai JAMAIS le 10v à la sortie de la carte V1.2 (6 x Fluidnc ) même en tournant à droite ou gauche le potentiomètre... Une idée ???
Merci à tous
Phil

bdring, 2025-01-12:
@DVA85 This issue is closed. Please start a new issue.