bdring/FluidNC #1398: Problem: 6X CNC Controller is dead

CMarkussen, 2024-12-11:
### Wiki Search Terms

http://wiki.fluidnc.com/en/installation
http://wiki.fluidnc.com/en/hardware/official/6x_CNC_Controller
http://wiki.fluidnc.com/en/hardware/official/6_Pack_External
http://wiki.fluidnc.com/en/support/requesting_help

### Controller Board

6x CNC Controller V1.1

### Machine Description

3-axis gantry router with Makerbase MKS SERVO57D NEMA23 closed loop stepper motors, inductive end-stops, 1.5kW spindle with Huanyang VFD

### Input Circuits

```yaml
# Pinout for 6X CNC controller: http://wiki.fluidnc.com/hardware/6x_pinout.png

# Function        Motor       End-stop pin
# X-axis        : Motor1      gpio.2
# Y-axis left   : Motor2      gpio.26
# Y-axis right  : Motor3      gpio.33
# Z-axis        : Motor4      gpio.32

# RS485 / UART1 : Pin 15 (TX), 16 (RX), 14(RTS)
# Expansion port: Pin 25 (RX), 27 (TX)

# Input           Pin
# Touch probe   : gpio.34
# Start button  : gpio.39
# Pause button  : gpio.36 

# Outputs         Pin
# Cooling pump  : gpio.12 MOSFET
# Cooling Fan   : gpio.4
# Alarm pin     : gpio.13
```

### Configuration file

```yaml
This is the basis for the configuration file. The actual file has been modified to fix some syntax errors in the file but I am unable to access the current file

board: 6 Pack
name: C-mill CNC
meta: Version 1.0

verbose_errors: true

start:
  must_home: true
  deactivate_parking: true
  check_limits: true

axes:
  x:
    steps_per_mm: 400
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 590
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: gpio.2:pd
      limit_pos_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.2
        direction_pin: i2so.1
        disable_pin: i2so.0

  y:
    steps_per_mm: 400
    max_rate_mm_per_min: 5000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 450
    soft_limits: true
    homing:
      cycle: 2
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.26
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.5
        direction_pin: i2so.4
        disable_pin: i2so.7
    motor1:
      limit_all_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.33:pd
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.10
        direction_pin: i2so.9
        disable_pin: i2so.8

  z:
    steps_per_mm: 800
    max_rate_mm_per_min: 2000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 145
    soft_limits: true
    homing:
      cycle: 2
      mpos_mm: 145
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.32:pd
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.13
        direction_pin: i2so.12
        disable_pin: i2so.15

  
# Definition for a rotary axes (stepper motor 5)
#  a:
#    steps_per_mm: 53.4
#    max_rate_mm_per_min: 5000.0
#    acceleration_mm_per_sec2: 100.0
#    max_travel_mm: 960.0
#    soft_limits: false
#    homing:
#      cycle: 2
#      mpos_mm: 0
#    motor0:
#      limit_neg_pin: gpio.35
#      limit_pos_pin: NO_PIN
#      limit_all_pin: NO_PIN
#      hard_limits: false
#      pulloff_mm: 3.000
#      standard_stepper:
#        step_pin: I2SO.18
#        direction_pin: I2SO.17
#        disable_pin: I2SO.16

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

# Sensors and buttons
probe:
  pin: gpio.34
  toolsetter_pin: NO_PIN
  check_mode_start: true
  hard_stop: false

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 8000000

# Push buttons
control:
  feed_hold_pin: gpio.36
  cycle_start_pin: gpio.39
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

status_outputs:
  report_interval_ms: 500
  alarm_pin: gpio.13:high
  # idle_pin: gpio.26
  # run_pin: gpio.4
  # hold_pin: gpio.16

# Cooling fan
# Use gcode command M67 E0 Qx to control fan, where x is duty cycle
# Match PWM frequency of Arduino
user_outputs:
  analog0_pin: gpio.4
  analog0_hz: 490

# Relay for spindle cooling pump. Using MOSFETs (Check Spindle Pin Usage)
# Use M7, M8 and M9 to turn on and off
coolant:      
  flood_pin: gpio.12:high
  #mist_pin: gpio.4:high
  delay_ms: 0

# UART definition
# UART1 linked to RS485 port for VFD 
uart1:
  txd_pin: gpio.15
  rxd_pin: gpio.16
  #rts_pin: gpio.14
  baud: 9600
  #baud: 115200
  mode: 8N1

# UART2 is linked to exansion slot for the pendant
uart2:
  txd_pin: gpio.27
  rxd_pin: gpio.25
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1

# Used by the pendant
uart_channel1:
  uart_num: 2
  report_interval_ms: 75
  message_level: Verbose
  

# Huanyang spindle VFD configuration
Huanyang:
  uart_num: 1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 1600=1% 2000=10% 2000=24% 6000=25% 24000=100%
  off_on_alarm: false
  atc: atc_manual
  m6_macro: 

  
# 10V spindle defintion
# 10V:
  # forward_pin: gpio.15
  # reverse_pin: gpio.14
  # pwm_hz: 5000
  # output_pin: gpio.13
  # enable_pin: NO_PIN
  # direction_pin: NO_PIN
  # disable_with_s0: false
  # s0_with_disable: true
  # spinup_ms: 0
  # spindown_ms: 0
  # tool_num: 0
  # speed_map: 0=0.0% 1000=0.0% 24000=100.0%
  # off_on_alarm: false

# Laser tool definition
# Laser:
  # pwm_hz: 5000
  # output_pin: gpio.4
  # enable_pin: gpio.12
  # disable_with_s0: false
  # s0_with_disable: true
  # tool_num: 1
  # speed_map: 0=0.000% 255=100.000%
  # off_on_alarm: true

# position and feedrates for manual tool change and probing
atc_manual:
  safe_z_mpos_mm: -1.0
  probe_seek_rate_mm_per_min: 400.0
  probe_feed_rate_mm_per_min: 80.0
  change_mpos_mm: 300.0 0.0 -1.0
  ets_mpos_mm: 500.0 25.0 -10.0
  ets_rapid_z_mpos_mm: -100.0

macros:
  startup_line0: $H ZXY
  startup_line1:
  macro0: 
  macro1:
  macro2:
  macro3:
  after_homing: g0 x400 y100
  after_reset: g20
  after_unlock: g91
```

### Startup Messages

```text
The board does not boot, unable to access it through the WebUI or the web installer.
```

### User Interface Software

WebUI or FluidNC web installer

### What happened?

I bought your "6X CNC Controller" from Elecrow earlier this year. I managed to connect to it and configure it through USB using the web installer, powered by a 20V power supply. It was a bit unstable to get it to boot but I put that down to my ad-hoc power supply. 

I finally installed and connected it in my home made CNC but I am unable to get it to work (it appears to be dead). When diagnosing the issue, I measured the voltage on the pin headers in the center of the board and both the 5V and 3.3V headers have zero volts on them, while the VMot pin measures 24V so it has power. 

I suspect that the voltage regulators, or part of that circuit, is defect so there is no power to the ESP32.

Here is a couple of pictures of the installation: 
[image]
[image]

bdring, 2024-12-11:
What is the circuit mounted above the controller. It looks blue and maybe has a usb connector.

CMarkussen, 2024-12-11:
> Does the controller connect via USB to a PC? Can you see anything on a serial terminal like FluidTerm or the Web Installer?

No, I am unable to connect via a USB. If I use the web installer then it does not detect any USB com ports that it can connect to.

CMarkussen, 2024-12-11:
> What is the circuit mounted above the controller. It looks blue and maybe has a usb connector.

It is a Arduino Micro. It sits on a 3D printed plastic block that is secured through the bolt hole for the expansion boards. It is connected to the a few I/O pins on the 6X CNC controller and it has a serial communicates with another Arduino located by the spindle. I use it to simplify wiring and add some functionality like monitoring the temperature of the spindle and controlling some LEDs.

CMarkussen, 2024-12-11:
FYI. The board with the glowing LEDs is a is a "smoke stopper" that cuts the power if the power draw exceeds 1 amp. It is used to protect circuits from shorts during wiring and testing.

bdring, 2024-12-11:
The arduino micro is a 5v part. That socket is extremely sensitive. It connects directly to the 3.3v io pins.

http://wiki.fluidnc.com/en/hardware/official/6x_CNC_Controller#expansion-module-socket

CMarkussen, 2024-12-11:
> The arduino micro is a 5v part. That socket is extremely sensitive. It connects directly to the 3.3v io pins.
> 
> http://wiki.fluidnc.com/en/hardware/official/6x_CNC_Controller#expansion-module-socket

The Arduino is not connected to the expansion socket but to gpio.4 and gpio.13. I only temporarily use the expansion port location to fasten the Arduino. Also, I have not yet provided power to the Arduino since I have not come that far in the installation. 

My understanding is that the IO pins operate on 5V control logic as described in the "features" and in the "Input" sections on the [Wiki page](http://wiki.fluidnc.com/en/hardware/official/6x_CNC_Controller). My plan is to power the Arduino from the 5V pin header in the 6X CNC controller which should provide a stable voltage. do you think this can cause issues?

FYI: I have ordered the "FluidNC RJ12 Pendant/Display Module" and "FluidDial RJ12 Wiring Kit" that I will use to connect the pendant to the controller.

bdring, 2024-12-11:
The wiki is correct. The expansion port does not have any buffering or level shifting.

If you don't think anything on your end broke it, I can authorize a refund from Elecrow. I would need your name and order number.

CMarkussen, 2024-12-11:
> The wiki is correct. The expansion port does not have any buffering or level shifting.
> 
> If you don't think anything on your end broke it, I can authorize a refund from Elecrow. I would need your name and order number.

I do not see how I could have broken it since I took care not to short anything or connect anything that could cause an electrical spike, so I would appreciate a refund for the board. 

My name is Christian Markussen and the order number is 209278.

Thank you👍

bdring, 2024-12-11:
Refund or replacement?

CMarkussen, 2024-12-12:
> Refund or replacement?

Replacement please

bdring, 2024-12-12:
I contacted Elecrow to send a replacement. They should ship it within a few days.

- I suggest you put heatshink over all shielding. You don't want it to ever contact live circuitry. Conductive pieces of the shielding could also fall off onto the controller.
- Do not connect to the expansion port without approved modules. The modules insure the right voltages and add protection to the ESP32 pins.
- Be very careful about using the controller power to power other devices. It is best to use separate power supplies with a shared ground.

I am closing this issue. Start a new issue if you have questions with the rest of your build