bdring/FluidNC #1399: Feature: External stepper control through RS485

CMarkussen, 2024-12-11:
### Machine Context

3-axis gantry router with [Makerbase MKS SERVO57D NEMA23](https://www.aliexpress.com/item/1005008013104919.html?spm=a2g0o.productlist.main.15.377e7999c8Gsfy&algo_pvid=4d98cff5-a170-4a76-a239-3a1015588e56&aem_p4p_detail=2024121101524410746883037020140002572913&algo_exp_id=4d98cff5-a170-4a76-a239-3a1015588e56-7&pdp_npi=4%40dis%21NOK%21630.90%21630.90%21%21%21400.32%21400.32%21%40211b80e117339107646454096e9505%2112000043250285152%21sea%21NO%21855309643%21X&curPageLogUid=VphJqAcSGPbK&utparam-url=scene%3Asearch%7Cquery_from%3A&search_p4p_id=2024121101524410746883037020140002572913_2) closed loop stepper motors, inductive end-stops, 1.5kW spindle with Huanyang VFD.

### Feature Description

Is it possible to add functionality for controlling the stepper drivers through RS485? 
I am using  [Makerbase MKS SERVO57D NEMA23](https://www.aliexpress.com/item/1005008013104919.html?spm=a2g0o.productlist.main.15.377e7999c8Gsfy&algo_pvid=4d98cff5-a170-4a76-a239-3a1015588e56&aem_p4p_detail=2024121101524410746883037020140002572913&algo_exp_id=4d98cff5-a170-4a76-a239-3a1015588e56-7&pdp_npi=4%40dis%21NOK%21630.90%21630.90%21%21%21400.32%21400.32%21%40211b80e117339107646454096e9505%2112000043250285152%21sea%21NO%21855309643%21X&curPageLogUid=VphJqAcSGPbK&utparam-url=scene%3Asearch%7Cquery_from%3A&search_p4p_id=2024121101524410746883037020140002572913_2) closed loop stepper controllers that can be controlled through RS485, and I am sure there are other controllers that support that as well. This will simplify wiring, free up I/O pins, and can potentially also provide feedback from the stepper controllers for events such stalling or overloading the steppers motors, faults, etc. and allow FluidNC to halt the CNC program. 

### Other Approaches

FluidNC and your control boards already support RS485 communication so it should be possible to implement this. Here is a link to the Github repository for this unit: https://github.com/makerbase-motor/MKS-SERVO57D

### How I Can Help

I will be happy to be a beta-tester for this feature

MitchBradley, 2024-12-11:
You ask "is it possible".  It is indeed possible in the sense that it does not violate the laws of physics.  But, presumably, what you really mean is "will we do it for free?".  The answer to that is "almost certainly no, at least for the forseeable future".  The cost-benefit analysis just does not work.  FluidNC is intended for multi-axis CNC machines where the tool must move in a precise trajectory in N-space, while respecting the individual axes acceleration and speed constraints.  In order to do that, it needs step-by-step control of each axis at fine granularity and short time resolution.

RS485 controlled drivers typically have their own internal motion controller than handles acceleration and deceleration.  They are easy to use for single-axis machines, but doing an accurate multi-axis trajectory with them is extremely complicated.  There are Trinamic app notes that show multi-axis control and, even though some Trinamic drivers have internal motion controllers, the multi-axis setups use an external motion controller with step/direction for the individual drivers.  The timing characteristics of half-duplex multidrop RS485 are not at all favorable for this application.

Redesigning FluidNC's internal motion planning to handle external "smart" motion controllers would be a massive undertaking, requiring many months or more of planning, implementation, testing, documentation, and support.  Support would be especially expensive in light of the many problems that users already have with single-master, single-endpoint RS485 setups.  After it is "done", the work would not be over because people would want it to work with a different smart motion controller, and another and another, ad infinitum.

The only situation in which we would be likely to undertake such a redesign would be if some company wanted it badly enough to fully fund the work, at market rates for our time.  I estimate the true cost at over $100,000.  All you have offered as incentive is "being a beta tester".

CMarkussen, 2024-12-11:
Thank you for your reply. I did not realize the effort required to implement it so I understand your position on this.

closed 2024-12-11 by bdring