bdring/FluidNC #1402: ESP32 security fuse stops install

jappiemike, 2024-12-12:
### Wiki Search Terms

fuse security

### Controller Board

6-pack

### Machine Description

Gantry router with external stepper drivers and 2.2kw spindle

### Configuration file

```yaml
board: 6 Pack
name: 6 Pack External XYZ 10V
meta: 2022-08-20 B. Dring
start:
  must_home: false
  deactivate_parking: false
  check_limits: true
stepping:
  engine: I2S_STATIC
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 160
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 480
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 250.000
      seek_mm_per_min: 1800
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: gpio.35:low
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      # Terminal Block 1
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1: high
        disable_pin: I2SO.0
  y:
    steps_per_mm: 160
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2:100.000
    max_travel_mm: 500
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: True
      mpos_mm: 0.000
      feed_mm_per_min: 250.000
      seek_mm_per_min: 1800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.32:low:pu
      limit_pos_pin: gpio.33:low:pu
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      # Terminal Block 2
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4:high
        disable_pin: I2SO.7
    motor1:
      limit_neg_pin:gpio.16
      hard_limits: true
      pulloff_mm: 5.00
      # Terminal Block 4
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12:high
        disable_pin: I2SO.15
  z:
    steps_per_mm: 320
    max_rate_mm_per_min: 2300.000
    acceleration_mm_per_sec2: 150.000
    max_travel_mm: 107
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 150.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100
    motor0:
      limit_neg_pin: gpio.36:low
      limit_pos_pin: gpio.39:low
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      # Terminal Block 3
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8
probe:
  pin: gpio.26:low
  toolsetter_pin: gpio.4
  check_mode_start: true
control:
  safety_door_pin: gpio.27			
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
sdcard:
  card_detect_pin: NO_PIN                                         
  cs_pin: gpio.5
10V:
  forward_pin: gpio.13
  reverse_pin: gpio.15
  pwm_hz: 3000
  output_pin: gpio.14
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: true
  s0_with_disable: true
  spinup_ms: 1000
  spindown_ms: 1000
  tool_num: 0
  speed_map: 0=0.000% 24000=100.000%
  off_on_alarm: true
macros:
  startup_line0: $sd/run=init.nc
```

### Startup Messages

```text
Unable to get ESP32 to restart any more using fluidterm.
If I try and reinstall the firmware using install-wifi I get:
win64\esptool.exe --chip esp32 --baud 921600 dump_mem 0x3ff5a018 4 SecurityFuses.bin
esptool.py v3.1
Found 1 serial ports
Serial port COM5
Connecting.....
Chip is ESP32-D0WD-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c0:49:ef:4b:2f:58
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Traceback (most recent call last):
  File "esptool.py", line 4582, in <module>
  File "esptool.py", line 4575, in _main
  File "esptool.py", line 4122, in main
  File "esptool.py", line 3217, in dump_mem
PermissionError: [Errno 13] Permission denied: 'SecurityFuses.bin'
[13604] Failed to execute script esptool
esptool failed
Press any key to continue . . .
```

### User Interface Software

Fluidterm

### What happened?

I'm trying to get my rapidchange magazine to work with Fluidnc and UGS. I initially had problems with the X axis behaving strangely and decided to roll back both FluidNC to 3.8.3 and UGS to 2.1.9. Do I just buy another ESP module or can I fix this one.
Cheers
Mike

MitchBradley, 2024-12-12:
Try using webinstaller https://installer.fluidnc.com/

MitchBradley, 2024-12-12:
The error message you report seems to indicate that you do not have write permission in the folder from which you are running the installer script.

closed 2024-12-12 by bdring