bdring/FluidNC #1407: Homing Z problem

FMD14, 2024-12-18:
### Wiki Search Terms

Homing Z

### Controller Board

DLC MKS 32 V2.1

### Machine Description

Laser corexy  et z Neje 30w

### Configuration file

```yaml
board: MKS-DLC32 V2.1
name: CoreXY Laser NJE 30W
meta: 06/11/2024 FMD14
kinematics:
  corexy:
stepping:
  engine: I2S_STATIC
  idle_ms: 0
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
axes:
  shared_stepper_disable_pin: i2so.0
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 25000
    acceleration_mm_per_sec2: 3200
    max_travel_mm: 520
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 300
      seek_mm_per_min: 5000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.35:low
      hard_limits: true
      pulloff_mm: 5
      stepstick:
        step_pin: i2so.1
        direction_pin: i2so.2:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 25000
    acceleration_mm_per_sec2: 3200
    max_travel_mm: 550
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 300
      seek_mm_per_min: 5000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.36:low
      hard_limits: true
      pulloff_mm: 5
      stepstick:
        step_pin: i2so.5
        direction_pin: i2so.6:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
  z:
    steps_per_mm: 80
    max_rate_mm_per_min: 1200
    acceleration_mm_per_sec2: 500
    max_travel_mm: 30
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 300
      seek_mm_per_min: 1000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
    motor0:
      limit_neg_pin: gpio.34:low
      hard_limits: false
      pulloff_mm: 1
      stepstick:
        step_pin: i2so.3
        direction_pin: i2so.4:low
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
i2so:
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17
spi:
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14
sdcard:
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.33:pu:low
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
macros:
  macro0: $SD/Run=lasertest.gcode
  macro1: $SD/Run=home.gcode
coolant:
  flood_pin: NO_PIN
  mist_pin: gpio.04:low
  delay_ms: 0
probe:
  pin: gpio.22
  check_mode_start: true
  toolsetter_pin: NO_PIN
Laser:
  pwm_hz: 5000
  output_pin: gpio.32
  enable_pin: i2so.7
  disable_with_s0: false
  s0_with_disable: false
  tool_num: 0
  speed_map: 0=0.000%  1000=100.000%
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
start:
  must_home: false
```

### Startup Messages

```text
rbl 3.9 [FluidNC v3.9.3 (wifi) '$' for help]
[MSG:INFO: websocket auto report interval set to 300 ms]
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.9.3 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:Corexy_11_2024.yaml]
[MSG:INFO: Machine CoreXY Laser NJE 30W]
[MSG:INFO: Board MKS-DLC32 V2.1]
[MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.12]
[MSG:INFO: SD Card cs_pin:gpio.15 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STATIC Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:0ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Shared stepper disable I2SO.0]
[MSG:INFO: Axis X (0.000,520.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.1 Dir:I2SO.2:low Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.35:low]
[MSG:INFO: Axis Y (0.000,550.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.5 Dir:I2SO.6:low Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.36:low]
[MSG:INFO: Axis Z (0.000,30.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:I2SO.3 Dir:I2SO.4:low Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.34:low]
[MSG:INFO: macro0_pin gpio.33:low:pu]
[MSG:INFO: Kinematic system: CoreXY]
[MSG:INFO: Connecting to STA SSID:Freebox-C30E80]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connected - IP is 192.168.1.52]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: Laser Ena:I2SO.7 Out:gpio.32 Freq:5000Hz Period:8191]
[MSG:INFO: Mist coolant gpio.4:low]
[MSG:INFO: Probe gpio.22]

Grbl 3.9 [FluidNC v3.9.3 (wifi) '$' for help]
ok
[VER:3.9 FluidNC v3.9.3:]
[OPT:MPHS]
[MSG:Machine: CoreXY Laser NJE 30W]
[MSG:Mode=STA:SSID=Freebox-C30E80:Status=Connected:IP=192.168.1.52:MAC=0C-8B-95-1B-E2-2C]
ok

Grbl 3.9 [FluidNC v3.9.3 (wifi) '$' for help]
```

### What happened?

$hz
<Home|MPos:0.000,0.000,-33.000|FS:0,0|Pn:P|WCO:0.000,0.000,0.000>
[MSG:INFO: ALARM: Homing Fail Approach]
ALARM:9

ok

Here is the error message the motor runs a little then the error message

[image]

bdring, 2024-12-18:
You did not say what happened other than the alarm

Your config section for the Z tells the machine to move down ( positive_direction: false) 33 mm (max_travel_mm: 30 * seek_scaler: 1.1) to find the Z homing switch. It looks like your switch is mounted in positive Z.

```yaml
z:
    steps_per_mm: 80
    max_rate_mm_per_min: 1200
    acceleration_mm_per_sec2: 500
    max_travel_mm: 30
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: false
      mpos_mm: 0
      feed_mm_per_min: 300
      seek_mm_per_min: 1000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1
 ```

FMD14, 2024-12-18:
steps_per_mm: 80
    max_rate_mm_per_min: 1200
    acceleration_mm_per_sec2: 500
    max_travel_mm: 30
    soft_limits: true
    homing:
      cycle: 3
      positive_direction: true
      mpos_mm: 0
      feed_mm_per_min: 300
      seek_mm_per_min: 1000
      settle_ms: 500
      seek_scaler: 1.1
      feed_scaler: 1.1

I just changed it's the same 
Can we run the engine without the safety devices?

MitchBradley, 2024-12-18:
Does the motor run in the correct direction - toward the switch?

If you issue a motion command like `G0 Z10`, does the z axis move the correct distance and direction (10 mm upward for that command)?

FMD14, 2024-12-18:
No you have to put 300 to move a lot

MitchBradley, 2024-12-18:
You need to set your configuration correctly.  http://wiki.fluidnc.com/en/support/setup/motion#adjust-steps_per_mm

Read what the wiki says about configuration and adjust everything to match your machine.

FMD14, 2024-12-18:
my assembly is as follows nema 17 motor two G2 20 teeth pulleys with belt which drives a threaded rod of diameter 5 with a pitch of 1mm. When I exceed 800 mm the motor is not happy with the tmc2208 driver. How to calculate the steps with this assembly? To get 40 mm I have to type g0 z140. How to do it?

MitchBradley, 2024-12-18:
Go through the process at http://wiki.fluidnc.com/en/support/setup/motion step by step.  Read everything carefully and make sure you do everything it says.  If the motor is not happy with >800 it means that you are going too fast.

FMD14, 2024-12-20:
here is the new config file that works
[Corexy_12_2024 .txt](https://github.com/user-attachments/files/18213381/Corexy_12_2024.txt)

closed 2024-12-20 by MitchBradley