bdring/FluidNC #1408: Problem: fluidnc versions 3.9.0 and 3.9.3 dual y axis one motor rotates in same direction

jkingdcs, 2024-12-18:
### Wiki Search Terms

none

### Controller Board

rootcnc v3

### Machine Description

foxalien vasto with rootcnc controller, all motors upgraded to closed loop.
dual y axis motors, single motors on x and z
limit switches for x and z axis. y axis has limit switches for one side only

### Input Circuits

```yaml
machine works fine with 3.9.1 and not 3.9.1 or 3.9.3 so I don't think its a wiring error.
```

### Configuration file

```yaml
board: Root Controller ISO
name: VFD hard limits
meta:
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16

#NOTE pin 20 is relay 1
#NOTE pin 21 is relay 2

# BUG status outputs are not set until after first physical movement!!
status_outputs:
  report_interval_ms: 500
# idle seems to work on relays and motor output pins. idle ONLY when nothing else is happening including jogging
# 
#  idle_pin: I2SO.15:high
#run pin is NOT active during jog. IS active on other gcode commands
#  run_pin: I2SO.15:high
#hold pin works when feed hold is pressed i.e. ! and goes back to 0 when ~ is pressed
#  hold_pin: I2SO.15:high
# ALARM DOES NOT SEEM TO WORK. homing fail does not activate. E stop does not activate
  alarm_pin: I2SO.15:high

start:
  must_home: true
  check_limits: false

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 4
  disable_delay_us: 4

axes:
  shared_stepper_disable_pin: NO_PIN

  x:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 500.000
# was 
    max_travel_mm: 422.000
[... 293 lines cut ...]
[MSG:INFO: Connected - IP is 192.168.0.57]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: Status outputs Interval:500 Idle:NO_PIN Cycle:NO_PIN Hold:NO_PIN Alarm:I2SO.15]
[MSG:INFO: ATC:atc_manual]
[MSG:INFO: Huanyang Spindle with atc_manual Tx:gpio.17 Rx:gpio.16 RTS:gpio.4 Baud:9600]
[MSG:INFO: Probe gpio.2]
[MSG:INFO: Toolsetter gpio.14]
```

### User Interface Software

using web gui or gsender in grbl or grbl hal mode

### What happened?

machine works fine with fluidnc 3.9.1 and multiple earlier versions.
But with version 3.9.0 and 3.9.3, if I jog the dual motor y axis in the negative direction i.e. towards the front of the machine both y axis motors turn the correct way BUT when jogging the y axis in the positive direction i.e. towards the back of the machine one of the dual y axis motors (the one on the right hand side of the machine) turns the correct way and the other (on the left side of the machine) turns to move the axis towards the front of the machine skewing the gantry. . i.e. the left X axis motor always turns the same way regardless of the requested y axis travel direction. It is always the same motor turning the wrong way. Reseting does not help, switching on and off does not help. This is 100% repeatable, using webui or gsender for jogging makes no difference. Eventually the left hand y axis hits the homing switch and that is ignored too.  If I keep jogging the y axis eventually it will crash into the bearing support housing and the closed loop stepper driver detects the missing steps and errors out setting the fault pin, but no error is reported by fluidnc. If I hit the reset button then the controller says fault pin on startup.  Z and X (both axis have single motors) work fine. Just downgrading the firmware to 3.9.1 with no config or other changes restores correct operation. The issue is 100% reproducible over multiple upgrade and downgrade cycles.
I therefore do not believe it is a problem with the hardware, wiring or gsender and is a problem triggered by some code changes in fluidnc versions 3.9.3 and 3.9.0.

### GCode File

any Y axis jog at any stepping speed and any length in mm

bdring, 2024-12-18:
Just to clarify

- It it does not work with 3.9.0 and 3.9.3.
- It works with 3.9.1
- It works with versions earlier than 3.9.0

bdring, 2024-12-18:
What closed loop motors are you using?

jkingdcs, 2024-12-18:
> Just to clarify
> 
> * It it does not work with 3.9.0 and 3.9.3.
> * It works with 3.9.1
> * It works with versions earlier than 3.9.0

this is correct

jkingdcs, 2024-12-18:
> What closed loop motors are you using?

nema 23 from stepper online 3nm. motor controller is cl57T v4.1 closed loop drivers

bdring, 2024-12-18:
Is it possible to place an LED on the direction signal of the controller on the axis that is having trouble?

bdring, 2024-12-18:
[image]

jkingdcs, 2024-12-18:
> Is it possible to place an LED on the direction signal of the controller on the axis that is having trouble?

yes, I can put an led, oscilloscope or multimeter

(I can jog for about 10-20mm before the gantry skews so much it jams)

bdring, 2024-12-18:
The direction signal is persistent. just jog a few millimeters. The state of the direction signal will stay until you reverse.

MitchBradley, 2024-12-18:
It is very strange that there is different behavior between 3.9.0 and 3.9.1.  The changes between those versions are in an area that is very far removed from the step/direction generation code.

jkingdcs, 2024-12-18:
Y motor connected to motor socket B on rootcnc : (bad motor always turns same way)
multimeter negative on dir and positive on ref = -4.95V  (motor connector B)

Y motor connected to motor socket Y on rootcnc : (good motor operates as expected)
axis motor 1 multimeter negative on dir and positive on ref = +0V (motor connector Y)

MitchBradley, 2024-12-18:
I presume that the voltage on the good motor socket changes depending on the command direction, while the bad one does not change?

jkingdcs, 2024-12-18:
> I presume that the voltage on the good motor socket changes depending on the command direction, while the bad one does not change?

correct the motor on socket B has -5V dir to ref regardless of the direction of travel requested,.

I just put 3.9.1 back on, switched off and on the power and jogged in the y positive direction and BOTH motors work correctly and both motors have 0V between dir and ref when moving positive Y and -5V when moving negative Y.

jkingdcs, 2024-12-18:
> In the previous test, where the B voltage did not change, was it 3.9.0 or 3.9.3?
3.9.3 
(I had exactly the same behaviour from 3.9.0) but stopped using it when it got pulled from the releases)

jkingdcs, 2024-12-18:
I should mention I am using firmware (3.9.0 3.9.1 and 3.9.3) I compiled from sources in github and not running the pre compiled firmware in https://github.com/bdring/FluidNC/releases. 
The only change I made was to uncomment the 8MB flash option and comment out the 4MB flash option in platformio.ini. the compile environment and version of visual studio code was the same for all.
I am happy to test 3.9.1 and 3.9.3 original firmware downloads from git hub and/or compile with the $MB flash option if that would help

MitchBradley, 2024-12-18:
Yes, please do test the released versions, just to eliminate variables.

jkingdcs, 2024-12-18:
using the original downloads from git hub
3.9.1 works as expected no issues
3.9.3 ALSO works as expected...
im going to try recompiling 3.9.3 but with the 4BM flash option and see if it works or fails.

jkingdcs, 2024-12-18:
UGH this is crazy, 
I switched rootcnc back on with the firmware 3.9.3 downloaded from git hub and the Y axis problem has struck again. I did nothing expect reboot the pc and leave the rootcnc off from sometime.

oliof, 2024-12-18:
sounds like s loose crimp on one of the motor wires to me.

jkingdcs, 2024-12-18:
yep 3.9.3 4MB flash from git hub does not work after the second reboot.
if I have 3.9.1 flash installed, run install-wifi to flash 3.9.3 git hub firmware download and then quickly power cycle the root cnc 
3.9.3 works BUT when I power cycle the second time 3.9.3 fails with skewed axis. All subsequent power cycles 3.9.3 precompiled from git hub does not work. I even tried reflashing 3.9.3 git hub again over the 3.9.3 git hub and it still does not work

jkingdcs, 2024-12-18:
> sounds like s loose crimp on one of the motor wires to me.

just put 3.9.1 from git works first time reboot still works reboot again still works....

ok if I swap the Y and B cables at the rootnc then if its bad wiring the problem will not move from the left motor to the right motor if the problem is the rootnc not outputting the direction correctly then the problem will remain on the left motor...
lets see...

jkingdcs, 2024-12-18:
I swapped B and Y motor connectors at the rootcnc, 
with all 3.9.1 (compiled by me 4BM flash and from github) motors run correctly 
with 3.9.3 from git hub installed the right hand motor now moves the wrong way, if it was bad left hand motor wiring on the left motor then the left motor would still be doing incorrect things regardless of if it is attached to channel B or Y
I believe the B motor channel direction is jammed at -5V Y with 3.9.3 after flashing if you switch off for more than a second or two.
I guess the next step is to see if channels A and C have the same direction problem.
I will also unplug the control lines from the rootnc to the motors and test the voltages with no motor wiring attached.

jkingdcs, 2024-12-18:
b channel direction is locked to -5V with 3.9.3 if hand compiled as 4MB or hand compiled as 8MB or using original git hub download, this is the case even if the motor control wires are disconnected from the rootcnc. Apart from the single time I went from 3.9.1 to 3.9.3 and did a very quick power cycle all other experiments are consistent. I can't get 3.9.3 to work correctly on channel B.
channel B *always* works as expected in every 3.9.1, hand compiled 4MB or hand compiled 8MB or download from git hub.
the problem consistently moves from left motor to right motor and back when I swap the wiring and now the right hand motor is on channel B I cant get the left motor to misbehave.

MitchBradley, 2024-12-19:
We are at somewhat of a disadvantage as I don't have a Root Controller.  Maybe Bart has one but I don't remember him mentioning it.  Nor have I found schematics online.

One thing that comes to mind as a possibility is signal integrity problems causing the I2S shift registers to misread the serial data.  The 3.9 series runs the I2S at a faster clock rate allowing higher pulsing rates.  Bart's designs, which feature careful layout for good I2S signal integrity, have no problems at the faster clock, but who knows for the root controller?  The fact that both Y motors are on the same shift register chip makes this somewhat of a long shot though.

jkingdcs, 2024-12-19:
> We are at somewhat of a disadvantage as I don't have a Root Controller. Maybe Bart has one but I don't remember him mentioning it. Nor have I found schematics online.
>
> One thing that comes to mind [...]

hmm, that's an interesting thought which makes me wonder if its possibly a marginal I2S shift register...
I definitely want to move a y axis motor to channel A and then C to see if they are equally effected.

jkingdcs, 2024-12-19:
Some GOOD progress:
I just tested with one Y axis motor on Y channel (always reliable) and the other on (the previously unused) A channel and quick testing with 3.9.1 shows correct operation.
channel A also seems to work with 3.9.3. So far 3.9.3 has survived 4 power cycles and works correctly. I am now leaving it switched off for longer between power cycles. But this is the most reliable I have ever had 3.9.3...
(I have not tried channel C yet)

mattydboom, 2024-12-19:
This all sounds like my previously reported issue with v3.9.0 here #1364 

That version was untimely pulled, but I was experiencing exactly the same gantry skewing problem, and run a RootCNC ISO v3 with similar set up (except standard high torque and not closed loop steppers).

Bart and I did a bunch of tests over DM to try and get to the bottom but never did pinpoint the exact culprit..... and was fixed (and remains so for me) in all subsequent releases. 

Just thought I would share in case I can be of assistance in trouble shooting with the Root controller, or if my previously shared config file could be used for comparison / testing.

jkingdcs, 2024-12-19:
> This all sounds like my previously reported issue with v3.9.0 here #1364
>
> That version was untimely pulled, but I was experiencing exactly the same gantry skewing problem, and run a RootCNC ISO v3 [...]

thanks for the info.
is 3.9.3 ok for you? which motor control channel is your second y axis motor on?

mattydboom, 2024-12-19:
Uploaded 3.9.3 firmware last night and behaves correctly just like every release (including the Pre releases) for me since 3.9.0.

Yes, my Y2 is wired to B port like yours. See attached pin mapping cheat sheet which may help you check for similarities / differences in your set up. 
[Pin Mapping Record_Cheet Sheet.pdf](https://github.com/user-attachments/files/18196357/Pin.Mapping.Record_Cheet.Sheet.pdf)

MitchBradley, 2024-12-19:
74HC595 does not come close to meeting clock timing with the new, faster I2S stepping code.  Our controller design guidelines specify the use of 74AHCT595 which is about 5x faster than HC.

I will eventually make the clock speed configurable so it can go at the old slower speed, but I have other things to do right now.  I am starting to resent supporting third party hardware that people sell without sponsoring the project.  Ideally, users would also sponsor the project, but the number of sponsors is really low and the typical amount does not begin to pay for one breakfast.

jkingdcs, 2024-12-19:
> 74HC595 does not come close to meeting clock timing with the new, faster I2S stepping code. Our controller design guidelines specify the use of 74AHCT595 which is about 5x faster than HC.
>
> I will [...]

Mitch,
thank you for your time and knowledge, That explains the shift issues I have been having. I'm inclined to replace the shift register chips if I can find a pin compatible one that meets your specs or change controller and go with a 6-pack. I have made a donation to you which I will hope will pay for breakfast.

daxliniere, 2024-12-20:
I have been experiencing the same issue with a DLC32 - it's miserable, right?

I remember years ago when I started using FluidNC that I looked at supporting the project by buying one of Bart's 6-packs, but I recall the price being high once you bought all the modules required. What I _**didn't**_ realise is that in the meantime Bart released a fully integrated version and switched to a less expensive manufacturer (Elecrow). Had I known I could get a high quality controller that supported the project for $49, I would have done it ages ago.

oliof, 2024-12-20:
@daxliniere you can still retire the DLC32 and get the 6x for less than a night out with friends (-;

MitchBradley, 2024-12-20:
The branch that I pushed last night probably fixes the problem.

daxliniere, 2024-12-20:
Oh, yes, absolutely. I have ordered one as I just blew up my DLC32 debugging this issue (haste makes waste).

MitchBradley, 2024-12-20:
Contrary to the hardware design guidelines on the wiki, DLC32 uses 'HC595 instead of 'AHCT595 shift registers.  The max clock rate for AHCT logic is more than 5x faster than HC.  The redesign of the I2S stepping code allowed us to run the I2S bus faster and push the max step rate from 125KHz up to 250KHz, but that pushed the bit clock rate to just past the limit of what HC logic can handle reliably.

Yesterday I figured out a way to get to the 250KHz rate while still using the slower bit clock that is within HC logic specs.  That change also has the effect that, if you use the higher bit clock rate, you can pulse at 500KHz on a controller that uses AHCT595 as specified.

There is now a new configuration item /i2so/max_pulse_us.  Its default value is 2, which results in up to 250KHz pulse rate and works with both HC and AHCT shift registers.  It can also be set to 4, for 125KHz max pulse rate and extra-slow shift registers, and to 1 for 500KHz max pulse rate with properly-designed controllers.

jkingdcs, 2024-12-20:
> Yesterday I figured out a way to get to the 250KHz rate while still using the slower bit clock that is within HC logic specs. That change also has the effect that, if you use the higher bit clock [...]

Thank you Mitch, that's fantastic. I would be very happy to test and report on the results as soon as I am back from visiting my patents for Christmas.

daxliniere, 2024-12-20:
Patent-in-law
The patent trap
Gentle parenting
😁

daxliniere, 2024-12-20:
> Yesterday I figured out a way to get to the 250KHz rate while still using the slower bit clock that is within HC logic specs.  That change also has the effect that, if you use the higher bit clock rate, you can pulse at 500KHz on a controller that uses AHCT595 as specified.

Haha, amazing work, Mitch! You are one tenacious bastard. 😊 For all the undue frustration, I bet that you feel a sense of great accomplishment. (As an ADHD-er, I know that feeling of beating an 'unbeatable' problem.)

daxliniere, 2024-12-20:
In the new year, I'm going to test maxing-out the microstepping on my Servo57C/D drivers with my new 6-pack. 👌

MitchBradley, 2024-12-20:
I'm sure that you will find problems and report them, further adding to my backlog.

daxliniere, 2024-12-20:
That's the spirit, Mitch! 😉

daxliniere, 2024-12-20:
But at least it will be on known hardware.

daxliniere, 2024-12-20:
I guess there will need to be a bulletin about this (required) limit.

To avoid a few support tickets, would it be worth adding a sanity check to the WebUI that when someone has a config that seems would likely break the pulse limit with known hardware, that a warning was displayed?

This would be a combination of # of axes, level of microstepping (>800 steps/mm?), max mm/sec and controller type* (i.e. DLC32, Root.)
(*Controller type would need parsing from text and this check wouldn't help if someone omitted/spelling it wrong).

MitchBradley, 2024-12-20:
With the new code, the only constraint that is not addressed by existing error checks is if you set min_pulse_us to 1 with a controller that has HC chips.  If you leave it at the default of 2 there should be no issue.  The steps_per_mm * max_rate_mm_per_sec check already exists.

I don't know of any reliable way to detect HC chips.  The way I determined that DLC32 uses them was by looking with a microscope.  The MKS schematics don't have part numbers.  You can't depend on particular values for the board field.  People frequently make their own hardware using whatever they feel like using, either not reading the guidelines, ignoring them, or thinking they know better.

Maybe the right answer is not to have the fast option.  I don't know of any drivers that can go that fast.

It is often the case that error checking actually causes as many failures as it prevents.  You can't win, and often you can't even 
break even.

bdring, 2024-12-20:
We have not even fully tested this right? Has anyone with this problem confirmed it fixed it?

MitchBradley, 2024-12-20:
That's right.  We are in the not-so-infrequent situation where people are eager for a fix then cannot test it until much later.

MitchBradley, 2024-12-20:
Now that I know the details, I could probably concoct a failure situation on a DLC32, but I am pretty tired of the whole thing and really want to do something for myself at that moment.

daxliniere, 2024-12-20:
> That's right. We are in the not-so-infrequent situation where people are eager for a fix then cannot test it until much later.

I would love to, but I am currently 3500mi from my CNC with its dead DLC32 controller and the new 6P controller is 5900mi from there.

breiler, 2024-12-21:
I double checked my MKS DLC32 and it also has the `HC595` chip and decided to test this out. I unfortunatley have some inconclusive results...

* I setup my MKS DLC32 to my test rig using the v3.9.3 and was able to reproduce the problems with the direction changing intermittently when running `G0 G91 G10` and `G0 G91 G-10`.

* I then compiled @MitchBradley's branch https://github.com/bdring/FluidNC/tree/BetterI2SClocking The default value of `$i2so/min_pulse_us: 2` worked fine for me with the test running `G0 G91 G10` and `G0 G91 G-10`.

* I then reverted to v3.9.3 and was no longer able to reproduce the problem...

bdring, 2024-12-21:
You could try the branch with a value of 1. See if the problem happens with a faster i2s rate.

breiler, 2024-12-21:
Yeah, I gave that a try as well and it works fine. The problem is that I can no longer reproduce the problem on 3.9.3 anymore either. If you have any suggestions on any setting that I should tweak, let me know. 

This is the configuration that I have been using:
```
board: MKS-DLC32
name: MKS-DLC32
meta: V2.6.9 (2202.11.09) Dax Liniere
stepping: 
  engine: I2S_STREAM
  idle_ms: 150
  pulse_us: 4
  dir_delay_us: 4
  disable_delay_us: 4
  segments: 12

i2so: 
  bck_pin: gpio.16
  data_pin: gpio.21
  ws_pin: gpio.17
  min_pulse_us: 1

spi: 
  miso_pin: gpio.12
  mosi_pin: gpio.13
  sck_pin: gpio.14

sdcard: 
  cs_pin: gpio.15
  card_detect_pin: NO_PIN
  frequency_hz: 8000000

kinematics: 
  Cartesian: 

axes: 
  shared_stepper_disable_pin: I2SO.0
  shared_stepper_reset_pin: NO_PIN
  homing_runs: 2
  x: 
    steps_per_mm: 800.000000
    max_rate_mm_per_min: 1400.000000
    acceleration_mm_per_sec2: 20.000000
    max_travel_mm: 1000.000000
    soft_limits: false
    homing: 
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000000
      feed_mm_per_min: 20.000000
      seek_mm_per_min: 300.000000
      settle_ms: 500
      seek_scaler: 1.100000
      feed_scaler: 1.100000

    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: 'gpio.36:low'
      hard_limits: false
      pulloff_mm: 1.000000
      stepstick: 
        step_pin: I2SO.1
        direction_pin: I2SO.2
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

  y: 
    steps_per_mm: 800.000000
    max_rate_mm_per_min: 1400.000000
    acceleration_mm_per_sec2: 20.000000
    max_travel_mm: 1000.000000
    soft_limits: false
    homing: 
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000000
      feed_mm_per_min: 20.000000
      seek_mm_per_min: 300.000000
      settle_ms: 500
      seek_scaler: 1.100000
      feed_scaler: 1.100000

    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: 'gpio.35:low'
      hard_limits: false
      pulloff_mm: 1.000000
      stepstick: 
        step_pin: I2SO.5
        direction_pin: I2SO.6
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

  z: 
    steps_per_mm: 800.000000
    max_rate_mm_per_min: 1000.000000
    acceleration_mm_per_sec2: 20.000000
    max_travel_mm: 1000.000000
    soft_limits: false
    homing: 
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 0.000000
      feed_mm_per_min: 20.000000
      seek_mm_per_min: 100.000000
      settle_ms: 500
      seek_scaler: 1.100000
      feed_scaler: 1.100000

    motor0: 
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: 'gpio.34:low'
      hard_limits: false
      pulloff_mm: 1.000000
      stepstick: 
        step_pin: I2SO.3
        direction_pin: I2SO.4
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN

control: 
  safety_door_pin: 'gpio.33:low:pu'
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: 'gpio.19:low:pu'
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN

coolant: 
  flood_pin: gpio.0
  mist_pin: gpio.4
  delay_ms: 0

probe: 
  pin: 'gpio.22:low'
  toolsetter_pin: NO_PIN
  check_mode_start: true
  hard_stop: false

macros: 
  startup_line0: 
  startup_line1: 
  Macro0: 
  Macro1: 
  Macro2: 
  Macro3: 
  after_homing: 
  after_reset: 
  after_unlock: 

start: 
  must_home: false
  deactivate_parking: false
  check_limits: true

parking: 
  enable: true
  axis: Z
  target_mpos_mm: 0.000000
  rate_mm_per_min: 800.000000
  pullout_distance_mm: 5.000000
  pullout_rate_mm_per_min: 250.000000

user_outputs: 
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN

user_inputs: 
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN

arc_tolerance_mm: 0.002000
junction_deviation_mm: 0.010000
verbose_errors: true
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
```

MitchBradley, 2024-12-21:
Sometimes clock problems get worse with temperature.  Back when I was doing hardware, we used to heat chips with a heat gun or cool them with freeze spray to try and provoke problems.

MitchBradley, 2024-12-21:
There are no settings in 3.9.3 that change the I2S core clock.  I don't know of any settings that I would expect to make a difference.    It is possible that the Y motor would be the most likely to have an issue since its direction pin is farther down the shift chain, but that is just speculation.

breiler, 2024-12-21:
Thanks for the tips, I'll continue testing later.

MitchBradley, 2024-12-21:
It would be possible to push the clock speed even farther into uncharted territory with this patch to 3.9.3
```diff
diff --git a/FluidNC/esp32/i2s_engine.c b/FluidNC/esp32/i2s_engine.c
index 7ff4d202..7eb94de2 100644
--- a/FluidNC/esp32/i2s_engine.c
+++ b/FluidNC/esp32/i2s_engine.c
@@ -301,7 +301,7 @@ int i2s_out_init(i2s_out_init_t* init_param) {
     i2s_ll_mclk_div_t first_div = { 2, 3, 47 };  // { N, b, a }
     i2s_ll_tx_set_clk(&I2S0, &first_div);
 
-    i2s_ll_mclk_div_t div = { 2, 32, 16 };  // b/a = 0.5
+    i2s_ll_mclk_div_t div = { 2, 0, 0 };  // b/a = 0
     i2s_ll_tx_set_clk(&I2S0, &div);
 
     // Bit clock configuration bit in transmitter mode.
```

breiler, 2024-12-21:
Boldly going where no man has gone before... =)

We are having dinner here and probably a couple of drinks with that, it will probably take a couple of hours before I am able try this out.

MitchBradley, 2024-12-21:
I was unable to make a DLC32 fail, either with the highest easily-achievable clock rate or with a heat gun.

breiler, 2024-12-22:
Me neither. I have tried again with the official 3.9.3 and with your `i2s_ll_mclk_div`-hack. The machine sounds awful with it, but it always steps in the right direction.

* I have set the `idle_ms=255` which increased the heat on my stepper drivers. 
* I recalled that in one of Dax videos he was jogging the Z axis, so I made sure to jog all of the axes back and forth with `G0 G91 X10 Y10 Z10` and `G0 G91 X-10 Y-10 Z-10`
* And have tinkered with `dir_delay_us`, lowering it to 1 (which was in Dax original config).

When I first recreated the issue I must have done something else wrong, I simply can't reproduce the issue anymore... =(

MitchBradley, 2024-12-22:
The awful sound could be a result of the step pulses being randomly corrupted.  The clock hypothesis would affect any bit, not just the direction bits.  Step bit problems would be harder to observe directly.

mattydboom, 2024-12-26:
Just went to use the machine this morning and the problem has reared itself again i.e. Y2 axis shooting off in the wrong direction during homing. So 3.9.3 for me has not solved the issue and I will look to test more later, but for now will return to last release that was stable for me (fluidnc-v3.9.2-pre2.... I think).

[24 more replies]

closed 2025-01-02 by bdring