bdring/FluidNC #1410: Problem: Feed hold sometimes stops with a hard stop

ktand, 2024-12-20:
### Wiki Search Terms

feed hold deceleration
cycle start acceleration
feed hold hard stop

### Controller Board

PiBot 4.9 Plus

### Machine Description

Gantry router with external DM556 drivers for x and y (dual). DM542 for Z. Spindle 10V controlled. FluidNC pendant. 

### Input Circuits

```yaml
feed_hold_pin: gpio.36:low
  cycle_start_pin: gpio.39:low
```

### Configuration file

```yaml
name: Pibot
board: Pibot 4.9
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

junction_deviation_mm: 0.02

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 320
    max_rate_mm_per_min: 3600
    acceleration_mm_per_sec2: 50
    max_travel_mm: 410
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: -205
      positive_direction: false
      seek_mm_per_min: 800
      feed_mm_per_min: 100
      seek_scaler: 1.100
      feed_scaler: 1.100
      settle_ms: 250
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: gpio.32:high
      limit_pos_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.2
        direction_pin: i2so.1
        disable_pin: i2so.0
  y:
    steps_per_mm: 640
    max_rate_mm_per_min: 3600
    acceleration_mm_per_sec2: 50
    max_travel_mm: 432
    homing:
      cycle: 2
      mpos_mm: 216
      positive_direction: true
      seek_mm_per_min: 800
      feed_mm_per_min: 100
      seek_scaler: 1.1
      feed_scaler: 1.1
      settle_ms: 250
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.35:high
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.5
        direction_pin: i2so.4:low
        disable_pin: i2so.7
  z:
    steps_per_mm: 1600
    max_rate_mm_per_min: 800
    acceleration_mm_per_sec2: 50
    max_travel_mm: 100
    homing:
      cycle: 1
      mpos_mm: 100
      positive_direction: true
      seek_mm_per_min: 400
      feed_mm_per_min: 100
      seek_scaler: 1.1
      feed_scaler: 1.1
      settle_ms: 250
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34:high
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.10
        direction_pin: i2so.9
        disable_pin: i2so.8

start:
  must_home: false

probe:
  pin: gpio.33:low
  toolsetter_pin: NO_PIN
  check_mode_start: true
  hard_stop: true

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.36:low
  cycle_start_pin: gpio.39:low
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN

status_outputs:
  run_pin: gpio.12
  hold_pin: gpio.14  

user_outputs:
  digital0_pin: gpio.4

10V:
  forward_pin: NO_PIN
  reverse_pin: NO_PIN
  pwm_hz: 1000
  output_pin: gpio.13
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 4000=0.000% 25000=100.000%
  off_on_alarm: false

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18  

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

uart1:
  txd_pin: gpio.25
  rxd_pin: gpio.27
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1

uart_channel1:
  report_interval_ms: 75
  uart_num: 1
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.9.3 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:config-pibot.yaml]
[MSG:INFO: Machine pibot]
[MSG:INFO: Board Pibot 4.9]
[MSG:INFO: UART1 Tx:gpio.25 Rx:gpio.27 RTS:NO_PIN Baud:1000000]
[MSG:INFO: uart_channel1 created at report interval: 75]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: User Digital Output: 0 on Pin:gpio.4]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-205.000,205.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.32]
[MSG:INFO: Axis Y (-216.000,216.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4:low Disable:I2SO.7]
[MSG:INFO:  Y Pos Limit gpio.35]
[MSG:INFO: Axis Z (0.000,100.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Z Pos Limit gpio.34]
[MSG:INFO: feed_hold_pin gpio.36:low]
[MSG:INFO: cycle_start_pin gpio.39:low]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:OT]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.224]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://pibot.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: Status outputs Interval:500 Idle:NO_PIN Cycle:gpio.12 Hold:gpio.14 Alarm:NO_PIN]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.13 Dir:NO_PIN Fwd:NO_PIN Rev:NO_PIN Freq:1000Hz Period:65535]
[MSG:INFO: Probe gpio.33:low]
```

### User Interface Software

CNCJs

### What happened?

I have wired up external buttons for feed hold and cycle start. When activating feed hold motion sometimes stops with a hard stop (i.e without deceleration). When that happens, the cycle start resumes without acceleration causing missed steps.

The same also happens when activating feed hold from a FluidNC pendant.

This is the terminal output when it happens (after the last Run):

```
<Idle|MPos:-121.150,16.000,-1.271|FS:0,0>
<Idle|MPos:-121.150,16.000,-1.271|FS:0,0>
<Idle|MPos:-121.150,16.000,-1.271|FS:0,0>
<Idle|MPos:-121.150,16.000,-1.271|FS:0,0>
<Idle|MPos:-121.150,16.000,-1.271|FS:0,0>
<Idle|MPos:-121.150,16.000,-1.271|FS:0,0|WCO:-121.150,16.000,0.000>
<Idle|MPos:-121.150,16.000,-1.271|FS:0,0|WCO:-121.150,16.000,0.000>
<Run|MPos:-121.150,13.050,-1.271|FS:1350,18000>
<Run|MPos:-121.150,-29.167,-1.271|FS:2792,18000>
<Run|MPos:-121.150,-56.000,7.449|FS:450,18000>
<Run|MPos:-121.150,-56.000,1.195|FS:231,18000>
<Run|MPos:-162.438,-42.984,0.951|FS:2400,18000>
<Run|MPos:-192.697,7.952,0.884|FS:2333,18000>
<Run|MPos:-174.278,64.591,0.816|FS:2400,18000>
<Run|MPos:-120.812,87.997,0.749|FS:2348,18000|WCO:-121.150,16.000,0.000>
<Run|MPos:-66.125,62.434,0.681|FS:2400,18000>
<Hold:1|MPos:-49.744,6.791,0.615|FS:1862,18000>
<Hold:0|MPos:-52.256,-4.914,0.602|FS:0,18000>
<Run|MPos:-60.819,-23.291,0.579|FS:2400,18000|Ov:100,100,100|A:S>
<Run|MPos:-110.678,-55.233,0.512|FS:2355,18000>
<Run|MPos:-166.841,-39.645,0.445|FS:2400,18000>
<Hold:1|MPos:-192.819,9.137,0.382|FS:796,18000|Ov:100,100,100|A:S>
<Run|MPos:-193.144,15.120,0.376|FS:1272,18000|WCO:-121.150,16.000,0.000>
<Run|MPos:-173.322,65.619,0.314|FS:2400,18000>
<Run|MPos:-119.241,87.973,0.248|FS:2381,18000>
<Hold:1|MPos:-66.053,62.348,0.181|FS:1995,18000|Pn:H>
<Hold:0|MPos:-58.891,52.159,0.167|FS:0,18000|WCO:-121.150,16.000,0.000>
<Hold:0|MPos:-58.891,52.159,0.167|FS:0,18000>
<Run|MPos:-49.487,9.059,0.117|FS:2371,18000>
<Hold:0|MPos:-64.938,-28.986,0.071|FS:0,18000>
<Run|MPos:-65.769,-30.006,0.069|FS:1207,18000>
<Run|MPos:-116.466,-55.845,0.006|FS:2337,18000|WCO:-121.150,16.000,0.000>
<Hold:0|MPos:-137.913,-54.020,-0.019|FS:0,18000>
<Run|MPos:-138.991,-53.753,-0.019|FS:936,18000>
<Hold:1|MPos:-180.816,-24.297,-0.078|FS:541,18000>
<Run|MPos:-190.450,-3.527,-0.103|FS:2341,18000>
<Hold:0|MPos:-189.950,37.222,-0.149|FS:2400,18000>
<Hold:0|MPos:-189.950,37.222,-0.149|FS:2400,18000|Ov:100,100,100|A:S>
<Hold:0|MPos:-189.950,37.222,-0.149|FS:2400,18000>
<Hold:0|MPos:-189.950,37.222,-0.149|FS:2400,18000>
```

### GCode File

G90 G94
G17
G21
T7
M3 S18000
G54
G0 X0 Y-72
Z7.5
G1 Z2.5 F2400
Z1 F800
G2 X0 Y72 Z0.75 I0 J72 F2400
X0 Y-72 Z0.5 I0 J-72
X0 Y72 Z0.25 I0 J72
X0 Y-72 Z0 I0 J-72
X0 Y72 Z-0.25 I0 J72
X0 Y-72 Z-0.5 I0 J-72
X0 Y72 I0 J72
X0 Y-72 I0 J-72
X0 Y72 Z-0.75 I0 J72
X0 Y-72 Z-1 I0 J-72
X0 Y72 I0 J72
X0 Y-72 I0 J-72
X0 Y72 Z-1.25 I0 J72
X0 Y-72 Z-1.5 I0 J-72
X0 Y72 I0 J72
X0 Y-72 I0 J-72
X0 Y72 Z-1.75 I0 J72
X0 Y-72 Z-2 I0 J-72
X0 Y72 I0 J72
X0 Y-72 I0 J-72
X0 Y72 Z-2.25 I0 J72
X0 Y-72 Z-2.5 I0 J-72
X0 Y72 I0 J72
X0 Y-72 I0 J-72
X0 Y72 Z-2.75 I0 J72
X0 Y-72 Z-3 I0 J-72
X0 Y72 I0 J72
X0 Y-72 I0 J-72
X0 Y72 Z-3.25 I0 J72
X0 Y-72 Z-3.5 I0 J-72
X0 Y72 I0 J72
X0 Y-72 I0 J-72
X0 Y72 Z-3.75 I0 J72
X0 Y-72 Z-4 I0 J-72
X0 Y72 I0 J72
X0 Y-72 I0 J-72
X0 Y72 Z-4.25 I0 J72
X0 Y-72 Z-4.5 I0 J-72
X0 Y72 I0 J72
X0 Y-72 I0 J-72
X0 Y72 Z-4.75 I0 J72
X0 Y-72 Z-5 I0 J-72
X0 Y72 I0 J72
X0 Y-72 I0 J-72
X0 Y72 Z-5.25 I0 J72
X0 Y-72 I0 J-72
X0 Y72 I0 J72
X0 Y-72 I0 J-72
X0 Y72 I0 J72
G1 Z7.5
M5
M30

### Other Information

FluidNC 3.9.3 also happens on 3.9.1

MitchBradley, 2024-12-20:
I no longer have time to solve problems on unofficial hardware whose developers do not sponsor FluidNC.  If you can reproduce the problem on official hardware like some flavor of 6-pack board, I will look into it.

ktand, 2024-12-20:
I was able to reproduce this on a standalone ESP32 Dev Kit using RMT stepping, if that matters.

Here is an example of stepper pulse capture when Feed Hold is pressed. The deceleration can clearly be seen.

[image]

Here  Cycle Start is pressed and then after a short while Feed Hold is pressed. The stepper pulses shows the acceleration but there is no deceleration when Feed Hold is pressed and the stepper pulses abruptly stops:

[image]

I used the following configuration:

```
name: DevKitC
board: ESP32-DevKitC
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

junction_deviation_mm: 0.02

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 320
    max_rate_mm_per_min: 3600
    acceleration_mm_per_sec2: 50
    max_travel_mm: 410
    soft_limits: false
    homing:
      cycle: 2
      mpos_mm: -205
      positive_direction: false
      seek_mm_per_min: 800
      feed_mm_per_min: 100
      seek_scaler: 1.100
      feed_scaler: 1.100
      settle_ms: 250
    motor0:
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: gpio.22
        direction_pin: gpio.21
  y:
    steps_per_mm: 640
    max_rate_mm_per_min: 3600
    acceleration_mm_per_sec2: 50
    max_travel_mm: 432
    homing:
      cycle: 2
      mpos_mm: 216
      positive_direction: true
      seek_mm_per_min: 800
      feed_mm_per_min: 100
      seek_scaler: 1.1
      feed_scaler: 1.1
      settle_ms: 250
    motor0:
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: gpio.16
        direction_pin: gpio.17

  z:
    steps_per_mm: 1600
    max_rate_mm_per_min: 800
    acceleration_mm_per_sec2: 50
    max_travel_mm: 100
    homing:
      cycle: 1
      mpos_mm: 100
      positive_direction: true
      seek_mm_per_min: 400
      feed_mm_per_min: 100
      seek_scaler: 1.1
      feed_scaler: 1.1
      settle_ms: 250
    motor0:
      hard_limits: true
      pulloff_mm: 1
      standard_stepper:
        step_pin: gpio.18
        direction_pin: gpio.19

start:
  must_home: false

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: gpio.26:low:pu
  cycle_start_pin: gpio.27:low:pu
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN

status_outputs:
  run_pin: gpio.12
  hold_pin: gpio.14  
```

bdring, 2024-12-21:
I will setup a test fixture with control switches for hold and cycle start. It will take a few hours before I can fit that in my schedule.

breiler, 2024-12-21:
@bdring this sounds similar to to the the PR that I have open for jogging:
https://github.com/bdring/FluidNC/pull/1396

And with a bit of imagination also this issue about jogging: https://github.com/bdring/FluidNC/issues/902

bdring, 2024-12-21:
I am able to replicate the problem. I concur with @breiler observations. This seems to be due to the tiny segments coming from the arcs. It does not happen with longer segments.

bdring, 2024-12-27:
@ktand 

I think this PR fixes the problem. Do you know how to compile the code from a PR? It would be great if you could test it.

ktand, 2024-12-27:
@bdring

Yes. I will try it a.s.a.p. Thanks!

ktand, 2024-12-27:
@bdring 

I can confirm that this solves the problem. I have tested feed hold/cycle start hundreds of times without any problems. Thank you very much!

closed 2025-01-02 by bdring