bdring/FluidNC #1413: Problem: Bootloader not active - Try holding down the BOOT switch

WhallaNar, 2024-12-27:
### Wiki Search Terms

Bootloader not active
booting
booting errors

### Controller Board

6x

### Machine Description

simply trying to get this controller to move one stepper motor

### Configuration file

```yaml
N/a
```

### Startup Messages

```text
esptool.js
Serial port
```

### What happened?

Trying to connect to web installer -

breiler, 2024-12-27:
We are going to need a bit more information than this.

What controller do you mean by "6x"?
* [6 Pack Universal Controller](http://wiki.fluidnc.com/en/hardware/existing_hardware#h-6-pack-universal-controller) 
* [6x CNC Controller](http://wiki.fluidnc.com/en/hardware/existing_hardware#h-6x-cnc-controller)

Did you try to press the boot-switch as the message suggested?

WhallaNar, 2024-12-27:
This is the 6x CNC Controller.

I tried disconnecting, removing power, and restarting the process while holding the boot button but i keep getting the same error. I tried this a few times

breiler, 2024-12-27:
What operating system and browser are you using?

Are you you sure that you are connecting to the right serial device? Have you installed the serial drivers?

WhallaNar, 2024-12-27:
I am using windows 11 and google chrome.

I am not sure if it is the correct serial device, where do i find the serial drivers? Searched google and the wiki but don’t see drivers mentioned.

BlueOrangeLive, 2024-12-27:
The main power must be on to to power the USB chip. You will not get a connection to your computer if the main power is not on.

WhallaNar, 2024-12-27:
I have the board connected to a benchtop power supply set to 12v. LED 1 is lit.

BlueOrangeLive, 2024-12-27:
You may need to install the driver for the Silicon Labs CP2102 USB driver.

bdring, 2024-12-27:
When you click the connect button does the popup window say, "No compatible devices found" or it it give you a list of ports to select, like this?

[image]

WhallaNar, 2024-12-28:
I installed the driver and I'm now getting one step further, but encountering another issue.
[image]

bdring - it does come up in that popup exactly as your photo shows.

bdring, 2024-12-28:
I am not sure what is going on. Try using the release from github.

[Use this release](https://github.com/bdring/FluidNC/releases/tag/v3.9.3)

[See this wiki page for help](http://wiki.fluidnc.com/en/installation#using-pre-compiled-files)

WhallaNar, 2024-12-28:
Looks like this is giving more luck - is there any way to determine which COM port to use?
[image]

bdring, 2024-12-28:
The easiest way is to unplug all other com ports.

MitchBradley, 2024-12-28:
You should use COM3 because the USB serial port chip on a 6x is made by "SiLabs", hence the name "\Device\SiLabser1"

Some PCs have one or more onboard COM ports that communicate with devices that are built-in to the PC.  Such devices will not disappear even when you unplug every COM1 could be one of those.

In any case, the installation succeeded.  The prompt comes from FluidTerm, which the installer runs at the end to help you configure the board.

WhallaNar, 2024-12-29:
I ran the install and seemed to be fine. I then uploaded the config with fluidterm then I was able to connect to the devices WIFI network and access the web interface. But now when i try to access the web interface i just end up with a blank screen after the "Captive portal page". When i was able to access this screen my stepper motor is was not moving when i try jogging the axis. I am pretty sure i have everything wired correctly so I'm wondering if it could be a config issue? I only have one driver and motor wired to the controller, does that matter? I will paste my config file below.

board: 6x
name: 6x Default
stepping:
  engine: I2S_static
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 4
  disable_delay_us: 0
  segments: 6

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.2:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.26:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  z:
    steps_per_mm: 800.000
[... 165 lines cut ...]
  # output_pin: gpio.4
  # enable_pin: gpio.12
  # disable_with_s0: false
  # s0_with_disable: true
  # tool_num: 1
  # speed_map: 0=0.000% 255=100.000%
  # off_on_alarm: true

# #begin 10V
# 10V:
  # forward_pin: gpio.15
  # reverse_pin: gpio.14
  # pwm_hz: 5000
  # output_pin: gpio.13
  # enable_pin: NO_PIN
  # direction_pin: NO_PIN
  # disable_with_s0: false
  # s0_with_disable: true
  # spinup_ms: 0
  # spindown_ms: 0
  # tool_num: 0
  # speed_map: 0=0.000% 1000=0.000% 24000=100.000%
  # off_on_alarm: false

[image]

MitchBradley, 2024-12-29:
Use FluidTerm to debug motion problems according to http://wiki.fluidnc.com/en/support/setup/motion

bdring, 2025-01-02:
I formatted your latest post to make it readable.

closed 2025-01-11 by bdring