bdring/FluidNC #1414: Problem:   two boards are gone, there is no more any output to the A4988

SirGit1646, 2024-12-28:
### Wiki Search Terms

no output to the motors

### Controller Board

MKS DLC32 V2.1

### Machine Description

DLC32 V2.1  Driver A4988,  Motors no more moving at all , Spindle output death

### Configuration file

```yaml
Config File, your sample Config , everything installed with Web Installer, change to Config file from  a working FLUIDNC machine no success
```

### Startup Messages

```text
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$SS
<Idle|MPos:-89.488,0.000,0.000|FS:0,0>
```

### User Interface Software

WEB UI- Dashboard,  bCNC, UGS

### What happened?

I can install, I can change  Config file , but no output to the motor or Spindle PWM.
I change  back to FLUIDNC 3.8.0  no success
I change  back to GRBL32 no success

### Other Information

COMMANDS output

$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$SS
<Idle|MPos:-89.488,0.000,0.000|FS:0,0>
[MSG:ERR: Showing startup log from previous panic]
[MSG:INFO: FluidNC v3.9.3 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine Default (Test Drive no I/O)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-150.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-120.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-40.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: AP SSID Unimat3 IP 192.168.0.254 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

SirGit1646, 2024-12-28:
I do not understand why reinstalling GBRL32 or FLUIDNC 3.8.0  the boards are no more working , two brand new  boards tested with the original  GBRL32 before  installing FLUIDNC

bdring, 2024-12-28:
You need to create a config file for you controller and machine.  See the wiki

SirGit1646, 2024-12-28:
Thanks for your answer,I use the config file from a running dlc32  with
fluidnc 3.8.0, but also no success.
regards
Sigi

breiler, 2024-12-28:
It is impossible to debug with this little information.

Share the configuration that you think is supposed to work. Share the the startup log when trying to boot the controller with that configuration. Take a picture of your controller setup.

SirGit1646, 2024-12-28:
thanks I will do so, and mail it to you

breiler, 2024-12-28:
Mail will not work, you will have to attach them to this GitHub issue from the web page.

bdring, 2024-12-28:
Did you set the new config as the active config with the Installer or via `$Config/Filename=yourfile` ?

SirGit1646, 2024-12-28:
yeah I did, I simply overwrite the existing with the config of the working
one

SirGit1646, 2024-12-28:
the funny thing is I can install I can change the config but no success at
all, Wifi is working usb connection is working, non of my dlc32 , tinybee
ever gave me trouble

bdring, 2024-12-28:
This is wrong in your startup messages.

```
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine Default (Test Drive no I/O)]
```

You need to post startup messages that are correct.

SirGit1646, 2024-12-28:
[temporär.txt](https://github.com/user-attachments/files/18266621/temporar.txt)

bdring, 2024-12-28:
I made your post readable.

It looks like FluidNC is working fine. If your motors are not moving, you have a hardware problem.

SirGit1646, 2024-12-28:
no , I don't have, I just started  my Mill with the setup I did mail to you

bdring, 2024-12-28:
It works now?

SirGit1646, 2024-12-28:
no it ist the board with fluidnc 3.8.0

bdring, 2024-12-28:
Please do not use email to respond to Github issues. 

Use the latest version of FluidNC, v3.9.3.

We are currently not supporting MKS DLC32 because it uses non compliant I2S chips. We might have a solution with  [this branch](https://github.com/bdring/FluidNC/tree/BetterI2SClocking), but we are awaiting users to test it. That may take a few more weeks.

SirGit1646, 2024-12-28:
yeah but I installed 3.8.0 and can not get it back to work, is it possible
3.9.3 messed up the Board, then even installing back GRBL32 got the board
never to work

bdring, 2024-12-28:
The only startup messages from 3.9.3 we have seen show user errors with the config file.

SirGit1646, 2024-12-28:
yeah I changed back to 3.8.0 with the working .yaml but also no success

bdring, 2024-12-28:
The only way FluidNC can break a controller is with a bad config file. For example, you create an output on a shorted pin.

Ask MKS what the problem is.

SirGit1646, 2024-12-28:
as did say I installed back v3.8.0 with the exact same config.yaml but
still not working

bdring, 2024-12-28:
OK, I will wait for MKS or other MKS users to respond.

SirGit1646, 2024-12-28:
ok please let me know if get any info, for the moment I would not install
fluidnc...is too scary 😂

bdring, 2024-12-28:
Just watch for new releases and read the release notes.

closed 2024-12-28 by bdring

SirGit1646, 2024-12-28:
thanks, I'm soo sorry until now fluidnc did work perfectly