bdring/FluidNC #1415: Problem: Unable to access Fluid WebUI

CMarkussen, 2024-12-28:
### Wiki Search Terms

webui access
webui trouble
webui Q&A
http://wiki.fluidnc.com/en/features/webui

### Controller Board

6-pack

### Machine Description

3-axis gantry router with Makerbase MKS SERVO57D NEMA23 closed loop stepper motors, inductive end-stops, 1.5kW spindle with Huanyang VFD

### Input Circuits

```yaml
# Connections
######################################################
# Function        Motor       End-stop pin
# X-axis        : Motor1      gpio.35
# Y-axis left   : Motor2      gpio.34
# Y-axis right  : Motor3      gpio.39
# Z-axis        : Motor4      gpio.36
# Rotary        : Motor5      gpio.????
# Free          : Motor6      

# Enable - yellow
# Step - Green
# direction - White

# RS485 / UART1 : Pin 15 (TX), 16 (RX), 14(RTS)
# Expansion port: Pin 25 (RX), 27 (TX)

# Input           Pin
# Touch probe   : gpio.2
# Start button  : gpio.26
# Pause button  : gpio.33 
# Fault		: gpio.32

# Outputs         Pin
# Cooling pump  : gpio.12
# Cooling Fan   : gpio.4
# Alarm pin     : gpio.13
# RTS (?)       : gpio.14
```

### Configuration file

```yaml
I have been testing with a minimal configuration file with only one stepper motor:
board: 6 Pac
name: Test
verbose_errors: true
start:
  must_home: false
  deactivate_parking: false
  check_limits: false
axes:
  x:
    steps_per_mm: 400
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 100
    max_travel_mm: 300
    soft_limits: true
    homing:
      cycle: 2
    motor0:
      limit_all_pin: NO_PIN
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.34
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: i2so.5
        direction_pin: i2so.4
        disable_pin: i2so.7
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0
probe:
  pin: gpio.2:pu:low
  toolsetter_pin: NO_PIN
  check_mode_start: true
  hard_stop: false
sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 8000000
control:
  feed_hold_pin: NO_PIN
  cycle_start_pin: gpio.26:pu:low
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN
spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17
uart1:
  txd_pin: gpio.15
  rxd_pin: gpio.16
  baud: 9600
  mode: 8N1
uart2:
  txd_pin: gpio.27
  rxd_pin: gpio.25
  rts_pin: NO_PIN
  cts_pin: NO_PIN
  baud: 1000000
  mode: 8N1
uart_channel1:
  uart_num: 2
  report_interval_ms: 75
  message_level: Verbose
```

### Startup Messages

```text
$Startup/Show
[MSG:INFO: FluidNC v3.9.3 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:test config.yaml]
[MSG:INFO: Machine Test]
[MSG:INFO: Board 6 Pack]
[MSG:INFO: UART1 Tx:gpio.15 Rx:gpio.16 RTS:NO_PIN Baud:9600]
[MSG:INFO: UART2 Tx:gpio.27 Rx:gpio.25 RTS:NO_PIN Baud:1000000]
[MSG:INFO: uart_channel2 created at report interval: 75]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-300.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  X Pos Limit gpio.34]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: cycle_start_pin gpio.26:low:pu]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:Jonas]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.86.29]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: Probe gpio.2:low:pu]
ok
```

### User Interface Software

WebUI and FluidNC web installer

### What happened?

I have connected the board to the inputs and outputs signal wires. The WebUI was working, I was exploring the UI and setting up my preferences. I made some changes to the fluid configuration file and after rebooting the board, it would not load the WebUI using address fluidnc.local. I have performed a complete re-install of Fluid using the Fluid web installer. Everything seems to work as expected but the WebUI will not load. I have also tried to access it directly through the IP address as reported through the Configure WiFi tab in the web installer but this presents a different user interface (see attached screen shot). I have used a minimal configuration for testing purposes but this has not helped. How can I fix this so I get access to the WebUI through the fluidnc.local address? 
[image]
[image]
[image]
[image]
[image]

### GCode File

NA

breiler, 2024-12-28:
The address "fluidnc.local" is broadcasted from the FluidNC controller using the mDNS protocol. I can think of a couple of reasons why this doesn't work for you:

* If the address is was used by another controller it will take some time for it to be unregistered the mDNS registry on your computer.
* Your router may apply some filtering which ports that should be available between different clients on your WiFi. However, since you are able to access the WebUI through the IP directly this is unlikely. 
* Your computer has a firewall that is blocking mDNS traffic.

What happens if you try to ping the controller using CMD `ping fluidnc.local`?

MitchBradley, 2024-12-28:
This is WebUI version 3.

[image]

Some older versions of Windows do not support mDNS so fluidnc.local will not work.  Recent versions, including Windows 10 and 11, support mDNS.  In my experience, fluidnc.local usually works but not always, for reasons unknown.  For that reason, on my production machines I configure a static IP address so I can always access via a known IP as a fallback.

oliof, 2024-12-28:
On some wifi systems, the mDNS domain is not . local but .lan -- for example on Google/Nest WiFi. Since your address range is in the 192.168.86.x range just as on my Google WiFi network, try fluidnc.lan; maybe it works.

CMarkussen, 2024-12-28:
Thanks for the reply guys. The fluid.local address has started to respond again for some reason. I do not know why but maybe it has something to do with my router, it is a bit random. I was confused when the WebUI appeared and thought something was wrong. I could not find any info on the FluidNC wiki. Everything appears to be working fine now, Thank you for your help.

closed 2024-12-28 by CMarkussen