bdring/FluidNC #1416: Feature: Dimmer support for AC motors

VeryFastSnail, 2024-12-29:
### Machine Context

Machine is upgraded chineese CNC machine with MKS DLC32 v2.1 controller

### Feature Description

I'm experimenting on with AC motors, MAKITA clones for example 10k-30k variant. PWM speed control can only be done with additional micro controller that reads in the PWM of spindle (5v TTL) and converts it to dimmers pwm.

Bigger dimmers that support bigger power routers needs zero cross pin for the library. https://github.com/RobotDynOfficial/RBDDimmer

It would be great to have feature where pwm and zero cross would work in conjunction to control the speed. 

### Other Approaches

There are all in one modules that would work, but those are expensive and low power, will buy one and take a look.

### How I Can Help

I can provide some code and what I found out experimenting on my own machine. Glad to do some coding myself if needed.

MitchBradley, 2024-12-29:
> And why should there be two micros in one machine doing the same job.

Because the FluidNC code is already bloated with features that only a small number of people use.  The developers are overwhelmed with support requests from people who rarely donate to help up defray our costs.  Developing code for and supporting hardware that we do not have is especially difficult.  Delegating code for rarely-used hardware to an external MCU makes it clear that it is Somebody Else's Problem.

bdring, 2024-12-29:
Try this.

https://cnc4pc.com/c47-a-c-speed-controller-relay-board.html?srsltid=AfmBOopa8IycqV3-UcJEXftEA_0w50a-QCj4_7uwlDS5XMO3AZf00fIJ

Or this.

https://vhipe.com/

Dimmers chop the power, so they get very weak at slower speeds. The Superpid is closed loop, so it maintains power at slower speeds.

Both have dangerous exposed voltages.

VeryFastSnail, 2024-12-30:
> Because the FluidNC code is already bloated with features that only a small number of people use. The developers are overwhelmed with support requests from people who rarely donate to help up defray [...]

Sounds right, in this case I will make all code available with tutorial how to have it setup in fluidnc so others can recreate. Even if there are only few people who will use it.

VeryFastSnail, 2024-12-30:
> Try this.
>
> https://cnc4pc.com/c47-a-c-speed-controller-relay-board.html?srsltid=AfmBOopa8IycqV3-UcJEXftEA_0w50a-QCj4_7uwlDS5XMO3AZf00fIJ
>
> Or this.
>
> https://vhipe.com/
>
> Dimmers chop the power, so [...]

I already have this solved with external micro, the only thing I want now is to have really stable PWM control in noisy environment, since brushed motors create EM radiation.  

Thanks for the information. Will take a look

closed 2024-12-30 by VeryFastSnail