bdring/FluidNC #1417: Problem: I can not compile the source after enabling Authentication

alidemyr, 2024-12-29:
### Wiki Search Terms

I searched about how to enable authentication

### Controller Board

MKS DLC32

### Machine Description

I compile with wifi environment v3..9.3 and earlier past five versions unsuccessfully.

### Configuration file

```yaml
// Copyright (c) 2012-2016 Sungeun K. Jeon for Gnea Research LLC
// Copyright (c) 2009-2011 Simen Svale Skogsrud
// Copyright (c) 2021 -	Bart Dring
// Use of this source code is governed by a GPLv3 license that can be found in the LICENSE file.

#pragma once

// This file contains compile-time configuration choices.  Most users will not need
// to directly modify these, but they are here for unusual needs, i.e.
// performance tuning or adjusting to non-typical machines.

// IMPORTANT: Any changes here require recompilation.

/*
ESP 32 Notes

Some features should not be changed. See notes below.

*/

// It is no longer necessary to edit this file to choose
// a machine configuration; edit machine.h instead
// machine.h is #included below, after some definitions
// that the machine file might choose to undefine.

const int MAX_N_AXIS = 6;

const int MAX_MESSAGE_LINE = 256;

// Axis array index values. Must start with 0 and be continuous.
// Note: You set the number of axes used by changing MAX_N_AXIS.
// Be sure to define pins or servos in the machine definition file.
const int X_AXIS = 0;  // Axis indexing value.
const int Y_AXIS = 1;
const int Z_AXIS = 2;
const int A_AXIS = 3;
const int B_AXIS = 4;
const int C_AXIS = 5;

const int X2_AXIS = (X_AXIS + MAX_N_AXIS);
const int Y2_AXIS = (Y_AXIS + MAX_N_AXIS);
const int Z2_AXIS = (Z_AXIS + MAX_N_AXIS);
const int A2_AXIS = (A_AXIS + MAX_N_AXIS);
const int B2_AXIS = (B_AXIS + MAX_N_AXIS);
const int C2_AXIS = (C_AXIS + MAX_N_AXIS);

const int SUPPORT_TASK_CORE = 0;  // Reference: CONFIG_ARDUINO_RUNNING_CORE = 1

// Serial baud rate
// OK to change, but the ESP32 boot text is 115200, so you will not see that is your
// serial monitor, sender, etc uses a different value than 115200
const int BAUD_RATE = 115200;

//Connect to your local AP with these credentials
//#define CONNECT_TO_SSID  "your SSID"
[... 403 lines cut ...]
FluidNC/src/WebUI/Authentication.h: In constructor 'AuthPasswordSetting::AuthPasswordSetting(const char*, const char*, const char*)':
FluidNC/src/WebUI/Authentication.h:19:9: error: type 'StringSetting' is not a direct base of 'AuthPasswordSetting'
         StringSetting(description, WEBSET, WA, NULL, name, defVal, MIN_LOCAL_PASSWORD_LENGTH, MAX_LOCAL_PASSWORD_LENGTH) {}
         ^~~~~~~~~~~~~
FluidNC/src/WebUI/Authentication.h:19:36: error: 'WEBSET' was not declared in this scope
         StringSetting(description, WEBSET, WA, NULL, name, defVal, MIN_LOCAL_PASSWORD_LENGTH, MAX_LOCAL_PASSWORD_LENGTH) {}
                                    ^~~~~~
FluidNC/src/WebUI/Authentication.h:19:36: note: suggested alternative: 'L_SET'
         StringSetting(description, WEBSET, WA, NULL, name, defVal, MIN_LOCAL_PASSWORD_LENGTH, MAX_LOCAL_PASSWORD_LENGTH) {}
                                    ^~~~~~
                                    L_SET
FluidNC/src/WebUI/Authentication.h:19:44: error: 'WA' was not declared in this scope
         StringSetting(description, WEBSET, WA, NULL, name, defVal, MIN_LOCAL_PASSWORD_LENGTH, MAX_LOCAL_PASSWORD_LENGTH) {}
                                            ^~
FluidNC/src/WebUI/Authentication.h: In member function 'Error AuthPasswordSetting::setStringValue(std::string_view)':
FluidNC/src/WebUI/Authentication.h:29:37: error: incomplete type 'StringSetting' used in nested name specifier
               return StringSetting::setStringValue(s);
                                     ^~~~~~~~~~~~~~
*** [.pio\build\wifi\src\src\FluidPath.cpp.o] Error 1
========================================================================================== [FAILED] Took 8.45 seconds ==========================================================================================

Environment    Status    Duration
-------------  --------  ------------
wifi           FAILED    00:00:08.448
===================================================================================== 1 failed, 0 succeeded in 00:00:08.448 =====================================================================================

MitchBradley, 2024-12-29:
The authentication code has not been maintained or tested for several years.  We are not supporting it anymore so if you need to use it, it will be up to you to figure out how to fix it.

alidemyr, 2024-12-29:
> The authentication code has not been maintained or tested for several years. We are not supporting it anymore so if you need to use it, it will be up to you to figure out how to fix it.

Hello!

Thank you for your quick reply. I built a machine at the shared workshop and I don't want to let unauthorized someone to use the laser cutter because of the safety issues. I reassembled the usb port of the board and I only use the machine with TCP. I set a statical IP and clever students discovered that while I were controlling the machine with LightBurn. I would appreciate it if you could figure it out and share with anyone who needs that layer of security. 

Thanks in advance,
Ali.

MitchBradley, 2024-12-29:
Spend $10 on an old wifi router and give the machine its own network for which only you know the credentials.

alidemyr, 2024-12-29:
> Spend $10 on an old wifi router and give the machine its own network for which only you know the credentials.

alidemyr, 2024-12-29:
> Spend $10 on an old wifi router and give the machine its own network for which only you know the credentials.

The problem is anyone can connect the machine who knows the device IP. I want to put a password to WebUI to prevent people who connect to the same WiFi. A router could fix it but everyone is using own PC. If there were a stationary PC, I could use an external modem or I could use the machine with AP mode, instead of STA.

I hope I did not misunderstand your solution.

Thank you,
Ali.

MitchBradley, 2024-12-29:
The dedicated wifi router will have its own ssid and password so random people will not be able to connect to that network. The fluidnc IP will only exist on that network. 

For a little extra obscurity, you could make it a hidden SSID.

MitchBradley, 2024-12-29:
The compiled-in authentication feature would not work with TCP/Lightburn anyway.  It only applies to WebUI and does nothing to protect against the TCP streaming channel.  Doing security right is very difficult and the old WebUI authentication is not very secure.  That is the main reason why we decided to drop it.  We don't have the time or skill to do proper security, so if you need it, it is best to wrap the FluidNC system behind a gateway - i.e. a dedicated router - whose security was implemented by security specialists.

closed 2024-12-29 by MitchBradley

alidemyr, 2024-12-29:
Thank you for your reply. When I saw authentication future, I jus wondered if I could use it to solve my problem with onboard future without adding extra hardware. I will consider your guess as a solution of my problem. If I could enable authentication, I could have kept using with WebUI. Maybe I can try to find a handshake algorithm during connection.

Thanks in advance, 
Ali

MitchBradley, 2024-12-29:
The extra hardware costs a few dollars at most, and many people have old routers that have been taken out of service, than can be had for nothing.  The time I have already spent explaining this is substantially more costly.

alidemyr, 2024-12-29:
So, I tried to set a password to Bluetooth connection. Esp32 does not support setting a pass key also. I will try to figure out a handshake algorithm. If I succeed in, I would share my work. Thank you for giving your time.