bdring/FluidNC #1420: Problem: FluidNC serial connection

sergiobeltrao, 2024-12-31:
### Wiki Search Terms

Serial connection, serial, g-code sender, serial connection

### Controller Board

I'm trying to build a custom controller board

### Machine Description

It's going to be a 3 axis TB6600HG driven CNC

### Configuration file

```yaml
board: None
name: Configuracao Padrao
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 12
kinematics: {}
axes:
  homing_runs: 2
  x:
    steps_per_mm: 80
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    soft_limits: true
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: gpio.26
        direction_pin: gpio.32
        disable_pin: gpio.18
  y:
    steps_per_mm: 80
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    soft_limits: true
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: gpio.27
        direction_pin: gpio.33
        disable_pin: gpio.13
  z:
    steps_per_mm: 80
    max_rate_mm_per_min: 1000
    acceleration_mm_per_sec2: 25
    max_travel_mm: 1000
    soft_limits: true
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1
      standard_stepper:
        step_pin: gpio.23
        direction_pin: gpio.25
        disable_pin: gpio.19
coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0
macros: {}
start:
  must_home: true
  deactivate_parking: false
  check_limits: true
parking:
  enable: false
  axis: Z
  target_mpos_mm: -5
  rate_mm_per_min: 800
  pullout_distance_mm: 5
  pullout_rate_mm_per_min: 250
user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN
user_inputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN
arc_tolerance_mm: 0.002
junction_deviation_mm: 0.01
verbose_errors: true
report_inches: false
enable_parking_override_control: false
use_line_numbers: false
planner_blocks: 16
```

### Startup Messages

```text
Grbl 3.9 [FluidNC v3.9.3 (wifi) '$' for help]
$Startup/Show
[MSG:INFO: FluidNC v3.9.3 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:INFO: Machine Configuracao Padrao]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.26 Dir:gpio.32 Disable:gpio.18]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.27 Dir:gpio.33 Disable:gpio.13]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:gpio.23 Dir:gpio.25 Disable:gpio.19]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:SobriaHorda]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connecting.]
[MSG:INFO: Connected - IP is 192.168.11.53]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
```

### User Interface Software

UGS

### What happened?

Hi, I have FluidNC v3.9.3 installed on my ESP32, and I'm trying to use UGS v2.1.9 to communicate with it via USB. However, every time I try, I get the following message:

'*** Connecting to jserialcomm://COM7:115200
*** Connection closed'

The TCP connection works fine. What am I missing? Do I have to configure something to be able to connect like that?

bdring, 2024-12-31:
Can you connect with the Web Installer or FluidTerm?

sergiobeltrao, 2024-12-31:
> Can you connect with the Web Installer or FluidTerm?

The web installer works fine, and I am also able to communicate with FluidNC through UGS using the TCP connection. However, when I try to do the same via serial, it does not work, and I've got the "Connection closed" error.

bdring, 2024-12-31:
Do you have FluidNC selected in UGS.

sergiobeltrao, 2024-12-31:
> Do you have FluidNC selected in UGS.

I do
Firmware: FluidNC
Port: COM7 (It's the right one, and the only one connected to the PC)
Port: 115200

sergiobeltrao, 2024-12-31:
Okay, this is going to sound a bit stupid, but I just realized that I was connecting with the web installer (to edit and check the configurations) before trying to connect with UGS, which means the serial connection was already occupied and that's why I couldn't open another one. Sorry

bdring, 2024-12-31:
No worries

closed 2024-12-31 by bdring