bdring/FluidNC #1423: Problem: 6x cnc controller, says controller does not appear to have fluid installed

Thereferee2, 2025-01-02:
### Wiki Search Terms

fluidnc installation

### Controller Board

6x cnc controller v1.2

### Machine Description

queenbee ultra 750 x 1000

### Configuration file

```yaml
board: 6x
name: 6x Eds xyyz 
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 90.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 590.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.26:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  y:
    steps_per_mm: 90.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 755.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

    motor1:
      limit_neg_pin: gpio.32:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

  z:
    steps_per_mm: 90.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 140.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low    
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 5.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0
       
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5
  
probe:
  pin: gpio.39:low
  toolsetter_pin: gpio.36:low
  
# Using MOSFETs (Check Spindle Pin Usage
coolant:
  flood_pin: gpio.12
  mist_pin: gpio.4
  delay_ms: 0

start:
  must_home: false
  
## Begin H100  
uart1:
  txd_pin: gpio.15
  rxd_pin: gpio.16
  rts_pin: gpio.14
  baud: 9600
  mode: 8N1

H100:
  uart_num: 1
  modbus_id: 1
  tool_num: 0
  speed_map: 0=0% 0=25% 6000=25% 24000=100%
  off_on_alarm: false
```

### Startup Messages

```text
not getting a startup message
```

### User Interface Software

web installer, fluidterm

### What happened?

6x cnc controller, says controller "does not appear to have fluid installed", I reinstall, it says installed hit continue and it goes back to fluidnc "does not appear to be installed".  tried installing via fluidterm, loads then gives msg error reset.....If i power the unit off for 15-30min then go back into web installer it open and allows me to access the board along with config files etc.  after 5min it drops out and goes back to "does not appear to have fluid installed"...machine ran yesterday
[image]
[image]

bdring, 2025-01-02:
Can you show a screenshot of FluidTerm? If you get no response in FluidTerm send CTRL+R to reset it or click the reset button on the controller. I want see some text from FluidTerm.

Thereferee2, 2025-01-02:
[image]

bdring, 2025-01-02:
Send a question mark on that terminal every 20 minutes until it stops responding or you believe FluidNC is working OK with FluidTerm.

Thereferee2, 2025-01-02:
Ok

Thereferee2, 2025-01-04:
Shut it down overnight....ran ? 8 times today over a couple hours on fluidterm with no errors.   I will run a test project tomorrow to see if it holds...thx

Thereferee2, 2025-01-04:
Ran for 2 hours no issues....issue closed...thx

closed 2025-01-11 by bdring