bdring/FluidNC #1426: "Skipping configuration file due to panic" - Setting up 6x_CNC_Controller

daxliniere, 2025-01-14:
### Wiki Search Terms

N/A

### Controller Board

6x_CNC_Controller V1.2

### Machine Description

3-axis gantry mill

### Configuration file

```yaml
board: 6x
name: 6x Default
meta: Settings v1.0.0 (2025-01-14) Dax Liniere

stepping:
  engine: I2S_STREAM
  idle_ms: 254
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 301.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 600.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: gpio.32
      limit_all_pin: NO_PIN
      hard_limits: true #??? was false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 186.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 600.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: gpio.26
      limit_all_pin: NO_PIN
      hard_limits: true #??? was false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 50.000 # was 100
    max_travel_mm: 100.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.2
      limit_all_pin: NO_PIN
      hard_limits: true #??? was false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

10V:
 forward_pin: gpio.14
 reverse_pin: NO_PIN
 pwm_hz: 5000
 output_pin: gpio.13
 enable_pin: NO_PIN
 direction_pin: NO_PIN
 disable_with_s0: false
 s0_with_disable: true
 spinup_ms: 0
 spindown_ms: 0
 tool_num: 0
 speed_map: 0=0.000% 1000=0.000% 24000=100.000% #update later with speed maps inc minimum 6000rpm
 off_on_alarm: false

probe:
  pin: gpio.34:low
  check_mode_start: true
#  toolsetter_pin: gpio.36:low

control:
  safety_door_pin: gpio.39:low:pu
  cycle_start_pin: NO_PIN
  feed_hold_pin: NO_PIN
  reset_pin: NO_PIN
  macro0_pin: gpio.36:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

# Using MOSFETs (Check Spindle Pin Usage
# coolant:
  # flood_pin: gpio.12
  # mist_pin: gpio.4
  # delay_ms: 0

coolant:
  flood_pin: gpio.12
# continuous air
  mist_pin:  gpio.4
# pulsed air 
  delay_ms: 0
  
parking:
  enable: true
  axis: Z
  pullout_distance_mm: 5.000
  pullout_rate_mm_per_min: 250.000
  target_mpos_mm: 0.000
  rate_mm_per_min: 800.000

#user_outputs:
#  analog0_pin: NO_PIN
#  analog1_pin: NO_PIN
#  analog2_pin: NO_PIN
#  analog3_pin: NO_PIN
#  analog0_hz: 5000
#  analog1_hz: 5000
#  analog2_hz: 5000
#  analog3_hz: 5000
#  digital0_pin: NO_PIN
#  digital1_pin: NO_PIN
#  digital2_pin: NO_PIN
#  digital3_pin: NO_PIN

start:
  must_home: false
  
macros:
  startup_line0:
  startup_line1:
  macro0: $H
#unlock and home XYZ
  macro1: G38.2 G91 Z-46 F100&G0 Z1&G38.2 G91 Z-2 F50&G10 L20 P1 Z4.985&G10 L20 P2 Z4.985&G10 L20 P3 Z4.985&G10 L20 P4 Z4.985&G10 L20 P5 Z4.985&G10 L20 P6 Z4.985&G53 G0 Z0&G90
#z-probe
  macro2:
  macro3: $SD/Run=lasertest.gcode

# #begin PWM
# pwm:
  # pwm_hz: 5000
  # direction_pin: NO_PIN
  # output_pin: gpio.13
  # enable_pin: gpio.14
  # disable_with_s0: false
  # s0_with_disable: true
  # spinup_ms: 0
  # spindown_ms: 0
  # tool_num: 0
  # speed_map: 0=0.000% 10000=100.000%
  # off_on_alarm: false

# #begin Laser
# Laser:
  # pwm_hz: 5000
  # output_pin: gpio.4
  # enable_pin: gpio.12
  # disable_with_s0: false
  # s0_with_disable: true
  # tool_num: 1
  # speed_map: 0=0.000% 255=100.000%
  # off_on_alarm: true
```

### Startup Messages

```text
Resetting MCU
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.9.4 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:config.yaml]
[MSG:ERR: Configuration parse error on line 71: Expected a float value like 123.456]
[MSG:INFO: Machine 6x Default]
[MSG:INFO: Board 6x]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:254ms]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40108b5e  PS      : 0x00060230  A0      : 0x800eddce  A1      : 0x3ffb2200
A2      : 0x3ffbddc0  A3      : 0x3ffc69ac  A4      : 0x3f402ed1  A5      : 0x00000001
A6      : 0x00000000  A7      : 0x003fffff  A8      : 0x00000000  A9      : 0x3ffb21e0
A10     : 0x00000001  A11     : 0x00000004  A12     : 0x0000000a  A13     : 0x00000013
A14     : 0x3ffb225c  A15     : 0x00000000  SAR     : 0x00000013  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x4008d5e9  LEND    : 0x4008d5f9  LCOUNT  : 0xffffffff

Backtrace: 0x40108b5b:0x3ffb2200 0x400eddcb:0x3ffb2240 0x40129a2a:0x3ffb2290

ELF file SHA256: b379c673bc8e0bd1

Rebooting...
[MSG:INFO: uart_channel0 created]
[MSG:RST]
[MSG:INFO: FluidNC v3.9.4 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:ERR: Skipping configuration file due to panic]
[MSG:INFO: Using default configuration]
[MSG:INFO: Axes: using defaults]
[MSG:INFO: Machine Default (Test Drive)]
[MSG:INFO: Board None]
[MSG:INFO: Stepping:RMT Pulse:4us Dsbl Delay:0us Dir Delay:0us Idle Delay:255ms]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Y (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Axis Z (-1000.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: STA SSID is not set]
[MSG:INFO: AP SSID FluidNC IP 192.168.0.1 mask 255.255.255.0 channel 1]
[MSG:INFO: AP started]
[MSG:INFO: WiFi on]
[MSG:INFO: Captive Portal Started]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]

Grbl 3.9 [FluidNC v3.9.4 (wifi) '$' for help]
[MSG:ERR: Configuration is invalid. Check boot messages for ERR's.]
```

### User Interface Software

FluidTerm

### What happened?

[MSG:ERR: Configuration is invalid. Check boot messages for ERR's.] -> [MSG:ERR: Skipping configuration file due to panic]

bdring, 2025-01-14:
Try removing the trailing comments. There are several.

[image]

[image]

daxliniere, 2025-01-14:
ahhh, sorry! Will do now. Thanks for the tip.

(Perhaps the error message I got could go into the wiki at some point)

daxliniere, 2025-01-14:
Yep, you got it. (Seems to not like it on second line under some circumstances)

closed 2025-01-14 by bdring