bdring/FluidNC #1427: Problem: Probing was skipped randomly, for the 1st tool change, before going for tool change in atc_manual macro.

ikhwan-ma, 2025-01-15:
### Wiki Search Terms

probing skipped atc manual

### Controller Board

6-pack external

### Machine Description

Gantry router with external HBS860H all closed loop, dual X motor, single Y, single Z, Fuling/Fohlinn DB600 VFD 3.7KW, 3.5KW 1200Hz 8pole ATC spindle.

### Configuration file

```yaml
board: 6 Pack External
name: 6 Pack Ext XXYZ Fuling ATC
meta: Last changed 23/11/2024
use_line_numbers: true

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 850.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

    motor1:
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

  y:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 650.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 500.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.32
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8
    
    
  z:
    steps_per_mm: 320.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 200.000
    max_travel_mm: 90.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 250.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.33
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15
        
probe:
  pin: gpio.36:low
  toolsetter_pin: gpio.14:low
  check_mode_start: true
  hard_stop: false
  
control:
  safety_door_pin: gpio.13:low
  # safety_door_pin pair with probe overtravel sw
  cycle_start_pin: gpio.12
  feed_hold_pin: gpio.15
        
i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

start:
  must_home: true
  
coolant:
  flood_pin: i2so.25
  mist_pin: i2so.24
  delay_ms: 0

user_outputs:
  #P0 - ATC tool release
  digital0_pin: i2so.27
  #P1 - TLO Probe Blower
  digital1_pin: i2so.26

10V:
  output_pin: gpio.26
  forward_pin: gpio.4
  reverse_pin: gpio.16
  pwm_hz: 5000
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 10000
  spindown_ms: 10000
  tool_num: 0
  speed_map: 0=0.000% 1800=10.000% 18000=100.000%
  off_on_alarm: false
  atc: atc_manual

atc_manual:
  safe_z_mpos_mm: -1.0000
  probe_seek_rate_mm_per_min: 300.0000
  probe_feed_rate_mm_per_min: 100.0000
  change_mpos_mm: -775.000 -315.000 -1.000
  ets_mpos_mm: 0.000 -75.000 -85.000
  ets_rapid_z_mpos_mm: -10.0000
```

### Startup Messages

```text
[MSG:INFO: FluidNC v3.9.4 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is spiffs]
[MSG:INFO: Configuration file:6P Ext XXYZ FUL.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine 6 Pack Ext XXYZ Fuling ATC]
[MSG:INFO: Board 6 Pack External]
[MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21Min Pulse:2us]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:I2S_STREAM Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]
[MSG:INFO: User Digital Output: 0 on Pin:I2SO.27]
[MSG:INFO: User Digital Output: 1 on Pin:I2SO.26]
[MSG:INFO: Axis count 3]
[MSG:INFO: Axis X (-850.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.2 Dir:I2SO.1 Disable:I2SO.0]
[MSG:INFO:  X Neg Limit gpio.34]
[MSG:INFO:   Motor1]
[MSG:INFO:     standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]
[MSG:INFO:  X2 Neg Limit gpio.39]
[MSG:INFO: Axis Y (-650.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]
[MSG:INFO:  Y Pos Limit gpio.32]
[MSG:INFO: Axis Z (-90.000,0.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]
[MSG:INFO:  Z Pos Limit gpio.33]
[MSG:INFO: safety_door_pin gpio.13:low]
[MSG:INFO: feed_hold_pin gpio.15]
[MSG:INFO: cycle_start_pin gpio.12]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:ASF]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.201]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://ASFRouter.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: ATC:atc_manual]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.26 Dir:NO_PIN Fwd:gpio.4 Rev:gpio.16 Freq:5000Hz Period:8191 with atc_manual]
[MSG:INFO: Flood coolant I2SO.25]
[MSG:INFO: Mist coolant I2SO.24]
[MSG:INFO: Probe gpio.36:low]
[MSG:INFO: Toolsetter gpio.14:low]
ok
```

### User Interface Software

UGS 2.1.9

### What happened?

** All axes homed **
1- load my nc file.
2- typed in G56 - my work offset
3- M61 Q10 - load the 1st tool used in the program
4- Hit Start button
5- Reached next tool change. Program paused, spindle turning down.

 **++normal case++**
**6- Z moves up, spindle move on to fixed probe for tool length offset probing, and then go to tool change location.**
7- Insert new tool, hit cycle start button, spindle goes to fixed probe to complete tool length offset probing.
8- Comes back to the last location at the last tool height, and continue with the rest of the program. I smiled.
9- Subsequent tool changes exhibit no problems at all.

**<<abnormal case>>**
**6- Z moves up, spindle move on to tool change location.**
7- Insert new tool, hit cycle start button, spindle goes to fixed probe to complete tool length offset probing.
8- Comes back to the last location at the wrong tool height, and crashed on the work piece. I died.

This <<abnormal case>> almost always happen when you ran the same program for the next part (2nd, 3rd.. on), regardless if you hit the soft reset button or not. But a power cycle will confirm the ++normal case++ to happen on the fresh run of the program, but not on the subsequent run of the program. Not without a power cycle.

bdring, 2025-01-15:
It will be hard for me to help if I cannot repeat the problem, so...

Attach a gcode file that exhibits the problem. The smallest and simplest example would be great.

What do you normally do between jobs? Do you manually send any gcode, like to return the last tool to the rack?

You mentioned a soft reset button and a start button. Are these UGS buttons?

If you do further testing check some values between jobs. See if anything looks wrong.

- $G ; to show the current work offset and tool number
- D#<_selected_tool>
- D#<_current_tool>
- $# ; see the tool length offset

ikhwan-ma, 2025-01-15:
Thanks for the quick reply.
It does not only happen with a certain program file, it also happened on others too. But i will attached the gcode file next.

> What do you normally do between jobs? Do you manually send any gcode, like to return the last tool to the rack?
i don't have a tool rack, but i change tool bits at the tool change location preset. Once a program finishes, i only raised the spindle up and jog the spindle aside to clear the work piece.
I entered G56 (to come back to the work offset because it usually default back to G54) and G49 to clear any tool length offset. I also hit the Soft Reset button on UGS panel.

> If you do further testing check some values between jobs. See if anything looks wrong.
I usually observe the Z height on every completed tool offset probe height. If the height between the work piece and tool bit seems off (during spindle turning up), I hit the (wired) pause button, and stopped the program from continuing, that's if it doesn't crashed on to work piece.

Thanks for tips given, I'll use the commands to check on it further.

ikhwan-ma, 2025-01-15:
N100 (Exported by FreeCAD)
N110 (Post Processor: script_module)
N120 (Output Time:2025-01-14 17:18:28.012247)
N130 (Begin preamble)
N140 G17 G90
N150 G21
N160 (Begin operation: Fixture)
N170 (Path: Fixture)
N180 G56
N190 (Finish operation: Fixture)
N200 (Begin operation: TC: 10mm 2F EM)
N210 (Path: TC: 10mm 2F EM)
N220 (TC: 10mm 2F EM)
N230 (Begin toolchange)
N240 M6 T10
N250 M3 S12500
N260 G4 P8
N270 (Finish operation: TC: 10mm 2F EM)
N280 (Begin operation: Profile009)
N290 (Coolant On:Mist)
N300 M7
N310 (Path: Profile009)
N320 (Profile009)
N330 (Compensated Tool Path. Diameter: 10.0)
N340 G0 Z5.000
N350 G0 X-17.850 Y67.000
N360 G0 Z3.000
N370 G1 X-17.850 Y67.000 Z-1.050 F350.000
N380 G1 X-17.850 Y-1.000 Z-1.050 F1050.000
N390 G1 X-25.970 Y-1.000 Z-1.050 F1050.000
N400 G1 X-25.970 Y67.000 Z-1.050 F1050.000
N410 G1 X-17.850 Y67.000 Z-1.050 F1050.000
N420 G0 Z5.000
N430 G0 Z5.000
N440 (Finish operation: Profile009)
N450 (Coolant Off:Mist)
N460 M9
N470 (Begin operation: TC: 4mm Vbit 90deg)
N480 (Path: TC: 4mm Vbit 90deg)
N490 (TC: 4mm Vbit 90deg)
N500 (Begin toolchange)
N510 M6 T18
N520 M3 S14500
N530 G4 P8
N540 (Finish operation: TC: 4mm Vbit 90deg)
N550 (Begin operation: Deburr)
N560 (Coolant On:Mist)
N570 M7
N580 (Path: Deburr)
N590 (Deburr)
N600 G0 Z5.000
N610 G0 X-174.270 Y-6.000
N620 G0 Z3.000
N630 G1 X-174.270 Y-6.000 Z-1.000 F350.000
N640 G1 X-174.270 Y72.000 Z-1.000 F1200.000
N650 G2 X-173.770 Y72.500 Z-1.000 I0.500 J0.000 K0.000 F1200.000
N660 G1 X-165.150 Y72.500 Z-1.000 F1200.000
N670 G2 X-164.650 Y72.000 Z-1.000 I0.000 J-0.500 K0.000 F1200.000
N680 G1 X-164.650 Y-6.000 Z-1.000 F1200.000
N690 G2 X-165.150 Y-6.500 Z-1.000 I-0.500 J0.000 K0.000 F1200.000
N700 G1 X-173.770 Y-6.500 Z-1.000 F1200.000
N710 G2 X-174.270 Y-6.000 Z-1.000 I0.000 J0.500 K0.000 F1200.000
N720 G0 Z5.000
N730 G0 Z5.000
N740 G0 X-137.430 Y72.000
N750 G0 Z3.000
N760 G1 X-137.430 Y72.000 Z-1.000 F350.000
N770 G1 X-137.430 Y-6.000 Z-1.000 F1200.000
N780 G2 X-137.930 Y-6.500 Z-1.000 I-0.500 J0.000 K0.000 F1200.000
N790 G1 X-146.550 Y-6.500 Z-1.000 F1200.000
N800 G2 X-147.050 Y-6.000 Z-1.000 I0.000 J0.500 K0.000 F1200.000
N810 G1 X-147.050 Y72.000 Z-1.000 F1200.000
N820 G2 X-146.550 Y72.500 Z-1.000 I0.500 J0.000 K0.000 F1200.000
N830 G1 X-137.930 Y72.500 Z-1.000 F1200.000
N840 G2 X-137.430 Y72.000 Z-1.000 I0.000 J-0.500 K0.000 F1200.000
N850 G0 Z5.000
N860 G0 Z5.000
N870 G0 X-110.690 Y-6.000
N880 G0 Z3.000
N890 G1 X-110.690 Y-6.000 Z-1.000 F350.000
N900 G2 X-111.190 Y-6.500 Z-1.000 I-0.500 J0.000 K0.000 F1200.000
N910 G1 X-119.810 Y-6.500 Z-1.000 F1200.000
N920 G2 X-120.310 Y-6.000 Z-1.000 I0.000 J0.500 K0.000 F1200.000
N930 G1 X-120.310 Y72.000 Z-1.000 F1200.000
N940 G2 X-119.810 Y72.500 Z-1.000 I0.500 J0.000 K0.000 F1200.000
N950 G1 X-111.190 Y72.500 Z-1.000 F1200.000
N960 G2 X-110.690 Y72.000 Z-1.000 I0.000 J-0.500 K0.000 F1200.000
N970 G1 X-110.690 Y-6.000 Z-1.000 F1200.000
N980 G0 Z5.000
N990 G0 Z5.000
N1000 G0 X-83.950 Y72.000
N1010 G0 Z3.000
N1020 G1 X-83.950 Y72.000 Z-1.000 F350.000
N1030 G1 X-83.950 Y-6.000 Z-1.000 F1200.000
N1040 G2 X-84.450 Y-6.500 Z-1.000 I-0.500 J0.000 K0.000 F1200.000
N1050 G1 X-93.070 Y-6.500 Z-1.000 F1200.000
N1060 G2 X-93.570 Y-6.000 Z-1.000 I0.000 J0.500 K0.000 F1200.000
N1070 G1 X-93.570 Y72.000 Z-1.000 F1200.000
N1080 G2 X-93.070 Y72.500 Z-1.000 I0.500 J0.000 K0.000 F1200.000
N1090 G1 X-84.450 Y72.500 Z-1.000 F1200.000
N1100 G2 X-83.950 Y72.000 Z-1.000 I0.000 J-0.500 K0.000 F1200.000
N1110 G0 Z5.000
N1120 G0 Z5.000
N1130 G0 X-57.210 Y-6.000
N1140 G0 Z3.000
N1150 G1 X-57.210 Y-6.000 Z-1.000 F350.000
N1160 G2 X-57.710 Y-6.500 Z-1.000 I-0.500 J0.000 K0.000 F1200.000
N1170 G1 X-66.330 Y-6.500 Z-1.000 F1200.000
N1180 G2 X-66.830 Y-6.000 Z-1.000 I0.000 J0.500 K0.000 F1200.000
N1190 G1 X-66.830 Y72.000 Z-1.000 F1200.000
N1200 G2 X-66.330 Y72.500 Z-1.000 I0.500 J0.000 K0.000 F1200.000
N1210 G1 X-57.710 Y72.500 Z-1.000 F1200.000
N1220 G2 X-57.210 Y72.000 Z-1.000 I0.000 J-0.500 K0.000 F1200.000
N1230 G1 X-57.210 Y-6.000 Z-1.000 F1200.000
N1240 G0 Z5.000
N1250 G0 Z5.000
N1260 G0 X-40.090 Y72.000
N1270 G0 Z3.000
N1280 G1 X-40.090 Y72.000 Z-1.000 F350.000
N1290 G2 X-39.590 Y72.500 Z-1.000 I0.500 J0.000 K0.000 F1200.000
N1300 G1 X-30.970 Y72.500 Z-1.000 F1200.000
N1310 G2 X-30.470 Y72.000 Z-1.000 I0.000 J-0.500 K0.000 F1200.000
N1320 G1 X-30.470 Y-6.000 Z-1.000 F1200.000
N1330 G2 X-30.970 Y-6.500 Z-1.000 I-0.500 J0.000 K0.000 F1200.000
N1340 G1 X-39.590 Y-6.500 Z-1.000 F1200.000
N1350 G2 X-40.090 Y-6.000 Z-1.000 I0.000 J0.500 K0.000 F1200.000
N1360 G1 X-40.090 Y72.000 Z-1.000 F1200.000
N1370 G0 Z5.000
N1380 G0 Z5.000
N1390 G0 X-13.350 Y-6.000
N1400 G0 Z3.000
N1410 G1 X-13.350 Y-6.000 Z-1.000 F350.000
N1420 G1 X-13.350 Y72.000 Z-1.000 F1200.000
N1430 G2 X-12.850 Y72.500 Z-1.000 I0.500 J0.000 K0.000 F1200.000
N1440 G1 X-4.230 Y72.500 Z-1.000 F1200.000
N1450 G2 X-3.730 Y72.000 Z-1.000 I0.000 J-0.500 K0.000 F1200.000
N1460 G1 X-3.730 Y-6.000 Z-1.000 F1200.000
N1470 G2 X-4.230 Y-6.500 Z-1.000 I-0.500 J0.000 K0.000 F1200.000
N1480 G1 X-12.850 Y-6.500 Z-1.000 F1200.000
N1490 G2 X-13.350 Y-6.000 Z-1.000 I0.000 J0.500 K0.000 F1200.000
N1500 G0 Z5.000
N1510 G0 Z5.000
N1520 (Finish operation: Deburr)
N1530 (Coolant Off:Mist)
N1540 M9
N1550 (Begin postamble)
N1560 M5
N1570 G17 G90
N1580 M2

bdring, 2025-01-15:
I doubt this is the problem, but I don't recommend the use of M2 or M30 at the end of a file with grbl style firmwares.

It can reset some things like modal values that you may have set before running the file or while running the file.

https://linuxcnc.org/docs/2.8/html/gcode/m-code.html#mcode:m2-m30

I can try to reproduce your problem tomorrow.

ikhwan-ma, 2025-01-15:
I got you.. thanks for the update. I'll wait for your result then. my machine is taking a break after a long run, just to check everything.

bdring, 2025-01-15:
The best way to reset before a new job is to send M6T0. Manually sending G49, etc, is problematic if everything is not done perfectly.

It need to reconfigure a machine to with a toolsetter. I won't be able for about 4-5 hours.

bdring, 2025-01-15:
It works by comparing new tools to the one that was used to set the work zero. If you run a second job, you need to let the system know that the original work zero is no longer valid. You do this by sending M6T0.

Here is the workflow you should be doing.

- First Job (no tool changes have been done since turn on)

  - Zero the workpiece (this is all manual)
    - Install a tool send M6T\<tool_num\>. If the tool was already installed send M61Q\<tool_num\>
    - Select the coordinate system that your file uses G54 to G59
    - Jog, probe and whatever you do to set the work zero.
  - Run the job.

- Second job

  - Send M6T0 to reset the ATC.
  - Zero the workpiece
    - **Install a tool:** Send M6T\<tool_num\>. If the tool was already installed, send M61Q\<tool_num\>
    - Select the coordinate system that your file uses G54 to G59
    - Jog, probe and whatever you do to set the work zero.

ikhwan-ma, 2025-01-16:
In my case here, it's not a 'new job', it's the same job, repeated ones because we have lots of parts to be machined. But I understand what you're suggesting.

I did just the same as your were suggesting in the first job run. No problem with this.

On the **next or new job**, now I need to send M6T0 (to reset the ATC). Cool. 

Then I need to set the right tool for the job, so M61Q<tool_num> or M6T<tool_num> doesn't matter right because it's a manual ATC..? But it's getting the right tool for 1st operation in the job again.

> The best way to reset before a new job is to send M6T0. Manually sending G49, etc, is problematic if everything is not done perfectly.

So, when do I need to send G49 to clear the previous tool offset..? I've seen from my observation, where I sent a G49, after a soft reset (from UGS), and the Z height do changed back to it's zeroed value. So, the soft reset didn't clear tool length offset, just to let you know this.

bdring, 2025-01-16:
It is the same job as far as the gcode. If you zero the new workpiece is is basically a new job for the controller.  

After the M6T0....M61Qx will change the current tool number. M6Tx will move to the tool change location and change the tool number. 

You should not send a G49. Let the firmware control the TLO. If you make a mistake. you could slam a tool. This would happen when you cleared the TLO and the job started with a longer tool than the Z zero was set with. If the gcode sees an M6Tx where X was the same as the previous tool, it skips the toolsetter to save time. It assumes the TLO is correct.

ikhwan-ma, 2025-01-17:
Okay.. i learned a couple of new things on the subject. Thanks for the update. I haven't got any batch job for the moment, but I already make a note on the commands and its sequence. I'll come back to update the status once I got the chance to do so.