bdring/FluidNC #1428: 6x CNC Controller died

daxliniere, 2025-01-15:
### Wiki Search Terms

N/A

### Controller Board

6x_CNC_Controller V1.2

### Machine Description

3-axis gantry mill
Switches: X- X+ Y- Y+ Z+ (limits), door (e-stop), macro_1.
Outputs: 3x Servo57C/D, 2x coolant (to Arduino)

### Input Circuits

```yaml
Limits are all NC, door and macro are NO. All wired directly to 6x_CNC
```

### Configuration file

```yaml
board: 6x
name: 6x Default
meta: Settings v1.0.0 (2025-01-14) Dax Liniere

stepping:
  engine: I2S_STREAM
  idle_ms: 254
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 301.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 600.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35
      limit_pos_pin: gpio.32
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
#        step_pin: I2SO.2
#        direction_pin: I2SO.1:low
#        disable_pin: I2SO.0
        step_pin: I2SO.13
        direction_pin: I2SO.12:low
        disable_pin: I2SO.15

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 3000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 186.000
    soft_limits: true
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 600.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: gpio.26
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
#        step_pin: I2SO.5
#        direction_pin: I2SO.4:low
#        disable_pin: I2SO.7
        step_pin: I2SO.18
        direction_pin: I2SO.17:low
        disable_pin: I2SO.16

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 50.000
    max_travel_mm: 100.000
# was 46.000
    soft_limits: true
    homing:
      cycle: 1
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 20.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: gpio.2
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9:low
        disable_pin: I2SO.8

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

10V:
 forward_pin: gpio.14
 reverse_pin: NO_PIN
 pwm_hz: 5000
 output_pin: gpio.13
 enable_pin: NO_PIN
 direction_pin: NO_PIN
 disable_with_s0: false
 s0_with_disable: true
 spinup_ms: 0
 spindown_ms: 0
 tool_num: 0
 speed_map: 0=0.000% 1000=0.000% 24000=100.000%
#update later with speed maps inc minimum 6000rpm
 off_on_alarm: false

probe:
  pin: gpio.34:low
  check_mode_start: true
#  toolsetter_pin: gpio.36:low

control:
  safety_door_pin: gpio.39:low
  #needs external PU resistor
  cycle_start_pin: NO_PIN
  feed_hold_pin: NO_PIN
  reset_pin: NO_PIN
  macro0_pin: gpio.36:low
  #needs external PU resistor
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

coolant:
  flood_pin: gpio.12
# continuous air
  mist_pin:  gpio.4
# pulsed air 
  delay_ms: 0
  
parking:
  enable: true
  axis: Z
  pullout_distance_mm: 5.000
  pullout_rate_mm_per_min: 250.000
  target_mpos_mm: 0.000
  rate_mm_per_min: 800.000

#user_outputs:
#  analog0_pin: NO_PIN
#  analog1_pin: NO_PIN
#  analog2_pin: NO_PIN
#  analog3_pin: NO_PIN
#  analog0_hz: 5000
#  analog1_hz: 5000
#  analog2_hz: 5000
#  analog3_hz: 5000
#  digital0_pin: NO_PIN
#  digital1_pin: NO_PIN
#  digital2_pin: NO_PIN
#  digital3_pin: NO_PIN

start:
  must_home: false
  
macros:
  startup_line0:
  startup_line1:
  macro0: $H
#unlock and home XYZ
  macro1: G38.2 G91 Z-46 F100&G0 Z1&G38.2 G91 Z-2 F50&G10 L20 P1 Z4.985&G10 L20 P2 Z4.985&G10 L20 P3 Z4.985&G10 L20 P4 Z4.985&G10 L20 P5 Z4.985&G10 L20 P6 Z4.985&G53 G0 Z0&G90
#z-probe
  macro2:
  macro3: $SD/Run=lasertest.gcode

# #begin PWM
# pwm:
  # pwm_hz: 5000
  # direction_pin: NO_PIN
  # output_pin: gpio.13
  # enable_pin: gpio.14
  # disable_with_s0: false
  # s0_with_disable: true
  # spinup_ms: 0
  # spindown_ms: 0
  # tool_num: 0
  # speed_map: 0=0.000% 10000=100.000%
  # off_on_alarm: false

# #begin Laser
# Laser:
  # pwm_hz: 5000
  # output_pin: gpio.4
  # enable_pin: gpio.12
  # disable_with_s0: false
  # s0_with_disable: true
  # tool_num: 1
  # speed_map: 0=0.000% 255=100.000%
  # off_on_alarm: true
```

### Startup Messages

```text
I don't have it on hand, but all was normal.
```

### User Interface Software

UGS

### What happened?

Controller seems to be dead. It is no longer causing axes to step, but can still be programmed, communicated-with, etc. (i.e. it doesn't know it's not stepping.)

Installation and configuration began last night. Last night and today controller was showing these LEDs:
[image]
Motor 2 (Y) never had Enable LED on and the axis was locked when not moving, but was moving correctly.

Shortly it was showing this and axes weren't working:
[image]
Only axes 1+2+3 are configured

MitchBradley, 2025-01-15:
Ouch.  Based on all the lights being on, my initial suspicion is that the shift register chips died.

daxliniere, 2025-01-15:
Yeah, I was pretty bummed.
I have some a basic SMD hot air station, is it worth trying to replace them?

daxliniere, 2025-01-15:
The parts are cheap, but the handling fee makes it a bit crazy.
[image]

MitchBradley, 2025-01-15:
If you have a logic analyzer you could first do a test to see if they are getting an I2S signal.  If not, the problem could be elsewhere.

The way to check with a LA is to tack wires onto the i2s gpios and look for clocks and data.

MitchBradley, 2025-01-16:
If you have only a scope, you can look at them individually.  bck/gpio.22 should have a fast clock in the 32 Mhz range.  ws/gpio.17 is slower, in the 1 MHz range.  data/gpio.21 has a waveform that depends on what is being sent to the chips, with a minimum pulse width of one full cycle of bck.

MitchBradley, 2025-01-16:
If you are using a recent version of FluidNC, you can control the clock rate with min_pulse_us in the i2s section.

bdring, 2025-01-16:
I can authorize Elecrow to replace it if needed.

daxliniere, 2025-01-16:
@MitchBradley Unfortunately a multimeter is all I have in terms of test gear.

daxliniere, 2025-01-16:
@bdring, yes, perhaps a replacement would be best. :(

bdring, 2025-01-17:
What is your order number with elecrow?

daxliniere, 2025-01-17:
Order# 237616

bdring, 2025-01-17:
Contact them about getting a getting a new one. I have authorized a replacement. 

I am closing this issue.

closed 2025-01-17 by bdring

daxliniere, 2025-01-18:
Thank you for sorting this so quickly, Bart. 🙏