bdring/FluidNC #1432: Homing doesn't work

qulet, 2025-01-21:
### Wiki Search Terms

FluidNC Limit Switch and Homing Setup

### Controller Board

ESP- WROM 32 4x TB6600 Stepper Motor Driver 4x limit swich optoisolator

### Machine Description

Router bramowy

### Input Circuits

```yaml
LTV847 optoisolator to esp32 pin
```

### Configuration file

```yaml
name: "ESP32 V4"
board: "ESP32"
meta: "N_ver"

stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  homing_runs: 3
  x:
    steps_per_mm: 1261.95
    max_rate_mm_per_min: 1088.8
    acceleration_mm_per_sec2: 230
    max_travel_mm: 1000
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: true
      mpos_mm: 10
      seek_mm_per_min: 800
      feed_mm_per_min: 500
      seek_scaler: 1.5
      feed_scaler: 1.5
    
    motor0:
      limit_neg_pin: gpio.34:low
      hard_limits: true
      pulloff_mm: 10
      stepstick:
        direction_pin: gpio.32
        step_pin: gpio.33

  y:
    steps_per_mm: 1261.95
    max_rate_mm_per_min: 1099.8
    acceleration_mm_per_sec2: 230
    max_travel_mm: 1000
    homing:
      cycle: 1
      allow_single_axis: true
      positive_direction: true
      mpos_mm: -10
      seek_mm_per_min: 800
      feed_mm_per_min: 500
      seek_scaler: 1.5
      feed_scaler: 1.5
    
    motor0:
      limit_neg_pin: gpio.39:low
      hard_limits: true
      pulloff_mm: 10
      stepstick:
        direction_pin: gpio.27
        step_pin: gpio.26

  z:
    steps_per_mm: 1600
    max_rate_mm_per_min: 800
    acceleration_mm_per_sec2: 192.1875
    max_travel_mm: 200
    homing:
      cycle: 0
      allow_single_axis: true
      positive_direction: true
      mpos_mm: -10
      seek_mm_per_min: 400
      feed_mm_per_min: 200
      seek_scaler: 1.5
      feed_scaler: 1.5
    
    
    motor0:
      limit_neg_pin: gpio.36:low
      hard_limits: true
      pulloff_mm: 10
      stepstick:
        direction_pin: gpio.25
        step_pin: gpio.12

  a:
    steps_per_mm: 1114.6
    max_rate_mm_per_min: 1345.2
    acceleration_mm_per_sec2: 448.4787
    max_travel_mm: 1000000
    homing:
      cycle: 3
      allow_single_axis: true
      positive_direction: true
      mpos_mm: -10
      seek_mm_per_min: 700
      feed_mm_per_min: 300
      seek_scaler: 1.5
      feed_scaler: 1.5
    
    
    motor0:
      limit_neg_pin: gpio.35:high
      hard_limits: true
      pulloff_mm: 10                                 
      stepstick:
        direction_pin: gpio.21
        step_pin: gpio.22

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  cs_pin: gpio.5
  card_detect_pin: NO_PIN
  frequency_hz: 8000000
  
coolant:
  flood_pin: gpio.14:low:pu

probe:
  pin: gpio.15:low:pu

user_outputs:
  digital0_pin: NO_PIN

10V:
  forward_pin: gpio.2:low:pu
  reverse_pin: NO_PIN
  pwm_hz: 5000
  output_pin: gpio.13
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0          
  tool_num: 0
  speed_map: 0=0.000% 1000=0.000% 12000=50% 24000=100.000%
  off_on_alarm: falseue000=25% 24000=100%
```

### Startup Messages

```text
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
[MSG:DBG: /macrocfg.json not found]
[MSG:DBG: 404.htm not found]
[MSG:DBG: /preferences2.json not found]
[MSG:DBG: 404.htm not found]
$SS
<Alarm|MPos:0.000,0.000,0.000,0.000|FS:0,0|Pn:XYZ|WCO:-15.812,10.055,65.651,-170.000>
[MSG:DBG: /preferences.json not found]
[MSG:DBG: 404.htm not found]
[MSG:INFO: FluidNC v3.8.3 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:4axis_v4_10V.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine ESP32 V4]
[MSG:INFO: Board ESP32]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (-990.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.33 Dir:gpio.32 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.34:low]
[MSG:INFO: Axis Y (-1010.000,-10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.27 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.39:low]
[MSG:INFO: Axis Z (-210.000,-10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.25 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.36:low]
[MSG:INFO: Axis A (-1000010.000,-10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.22 Dir:gpio.21 Disable:NO_PIN]
[MSG:INFO:  A Neg Limit gpio.35]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:Orange_Swiatlowod_D290]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connecting...]
[MSG:INFO: Connecting....]
[MSG:INFO: Connected - IP is 192.168.1.22]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.13 Dir:NO_PIN Fwd:gpio.2:low:pu Rev:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant gpio.14:low:pu]
[MSG:INFO: Probe gpio.15:low:pu]
ok
<Alarm|MPos:0.000,0.000,0.000,0.000|FS:0,0|Pn:XYZ|Ov:100,100,100|A:F>
$HX
[MSG:DBG: Homing Cycle X]
[MSG:DBG: Starting from 0.000,0.000,0.000]
[MSG:DBG: Planned move to 10.000,0.000,0.000 @ 500.000]
[MSG:DBG: CycleStop PrePulloff]
[MSG:INFO: ALARM: Homing Fail Pulloff]
ALARM:8
Homing fail. Pull off travel failed to clear limit switch. Try increasing pull-off setting or check wiring.
ok
<Alarm|MPos:10.000,0.000,0.000,0.000|FS:0,0|Pn:XYZ>
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG:  X Neg Limit 1]
[MSG:DBG: Limit switch tripped for X motor 0]
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG:  X Neg Limit 1]
[MSG:DBG: Limit switch tripped for X motor 0]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG: Limit switch tripped for Z motor 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Limit switch tripped for Y motor 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Limit switch tripped for Y motor 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Limit switch tripped for Y motor 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Limit switch tripped for Y motor 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Limit switch tripped for Y motor 0]
```

### What happened?

t has limit switches at the ends of the +X,+Y,+Z axes. I have set homig for each axis separately.
After starting $HX, $HY, $HZ, the axes move in the + direction towards the switch for a distance of pulloff_mm=10.
Increasing this value causes the desired movement, but when the switch is activated, the axis movement does not stop and a collision occurs....
All other functions work ok.
Jog moves the axes in the right direction.
What can I do to make homing work?

bdring, 2025-01-21:
It looks like your switches are inverted and reading backwards.

See this wiki page. 

http://wiki.fluidnc.com/en/support/setup/limit_switches

Also send a photo of your controller.

qulet, 2025-01-25:
Fluid is my alternative CNC milling machine control. My original control is the Mach 3 program via the LPT port. There, the end switches at the ends of the axes work flawlessly.  There is an error in the configuration file in the Fluid control and I cannot solve it....

[image]
[image]

bdring, 2025-01-25:
It appears that the switches are reporting as contacted before you try to home. Try inverting the logic. Change the ':low' to ':high' on all your switches in the config file.

bdring, 2025-01-25:
http://wiki.fluidnc.com/en/support/setup/limit_switches#test-switches

Look at the status `<Alarm|MPos:0.000,0.000,0.000,0.000|FS:0,0|Pn:XYZ|Ov:100,100,100|A:F>
$HX` pn:XYZ means XYZ switches are active. before homing.

qulet, 2025-01-27:
Alarms are the result of manual triggering of limit switches. This is to show that they work properly in Jog mode, but there is a problem in homing mode...

Changing the low logic to high causes a message at the start about their activation....

bdring, 2025-01-27:
2 things.  Please respond completely to both.

1. Show us the results of this test.

http://wiki.fluidnc.com/en/support/setup/limit_switches#test-switches

2. Are you using pull up or pull down resistors?

MitchBradley, 2025-01-27:
That setup, with an ESP32 apparently connected directly to switches without intermediate signal conditioning, will be susceptible to noise.  Spend a few bucks to get a proper controller board with noise filtering on the inputs.

qulet, 2025-01-27:
Proszę o podanie linku do takiej płytki.

Mam na  krańcówkach mam aktualnie separacje na transoptorach .

Please provide a link to such boards.

I have on limit switches I have access on optocouplers.

qulet, 2025-01-28:
TEST SWITCHES  
G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
[MSG:DBG: /macrocfg.json not found]
[MSG:DBG: 404.htm not found]
[MSG:DBG: /preferences2.json not found]
[MSG:DBG: 404.htm not found]
$SS
<Alarm|MPos:0.000,0.000,0.000,0.000|FS:0,0|Pn:XYZ>
[MSG:DBG: /preferences.json not found]
[MSG:DBG: 404.htm not found]
[MSG:INFO: FluidNC v3.8.3 https://github.com/bdring/FluidNC]
[MSG:INFO: Compiled with ESP32 SDK:v4.4.7-dirty]
[MSG:INFO: Local filesystem type is littlefs]
[MSG:INFO: Configuration file:4axis_v4_10V.yaml]
[MSG:DBG: Running after-parse tasks]
[MSG:DBG: Checking configuration]
[MSG:INFO: Machine ESP32 V4]
[MSG:INFO: Board ESP32]
[MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]
[MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]
[MSG:INFO: Stepping:RMT Pulse:2us Dsbl Delay:0us Dir Delay:1us Idle Delay:250ms]
[MSG:INFO: Axis count 4]
[MSG:INFO: Axis X (-990.000,10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.33 Dir:gpio.32 Disable:NO_PIN]
[MSG:INFO:  X Neg Limit gpio.34:low]
[MSG:INFO: Axis Y (-1010.000,-10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.26 Dir:gpio.27 Disable:NO_PIN]
[MSG:INFO:  Y Neg Limit gpio.39:low]
[MSG:INFO: Axis Z (-210.000,-10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.12 Dir:gpio.25 Disable:NO_PIN]
[MSG:INFO:  Z Neg Limit gpio.36:low]
[MSG:INFO: Axis A (-1000010.000,-10.000)]
[MSG:INFO:   Motor0]
[MSG:INFO:     stepstick Step:gpio.22 Dir:gpio.21 Disable:NO_PIN]
[MSG:INFO:  A Neg Limit gpio.35]
[MSG:INFO: Kinematic system: Cartesian]
[MSG:INFO: Connecting to STA SSID:Orange_Swiatlowod_D290]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.22]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.13 Dir:NO_PIN Fwd:gpio.2:low:pu Rev:NO_PIN Freq:5000Hz Period:8191]
[MSG:INFO: Flood coolant gpio.14:low:pu]
[MSG:INFO: Probe gpio.15:low:pu]
ok
<Alarm|MPos:0.000,0.000,0.000,0.000|FS:0,0|Pn:XYZ>
$X
ok
$X
ok
$Message/Level=Debug
ERROR:8
Grbl '$' command cannot be used unless Grbl is IDLE. Ensures smooth operation during a job.
[MSG:ERR: Command requires idle state]
$X
ok
$Message/Level=Debug
ERROR:8
Grbl '$' command cannot be used unless Grbl is IDLE. Ensures smooth operation during a job.
[MSG:ERR: Command requires idle state]
$X
ok
$HY
[MSG:DBG: Homing Cycle Y]
[MSG:DBG: Starting from 0.000,0.000,0.000]
[MSG:DBG: Planned move to 0.000,10.000,0.000 @ 500.000]
[MSG:DBG: CycleStop PrePulloff]
[MSG:INFO: ALARM: Homing Fail Pulloff]
ALARM:8
Homing fail. Pull off travel failed to clear limit switch. Try increasing pull-off setting or check wiring.
ok
<Alarm|MPos:0.000,10.000,0.000,0.000|FS:0,0|Pn:XYZ>
$Message/Level=Debug
ok
[MSG:DBG:  X Neg Limit 0]
[MSG:DBG:  X Neg Limit 1]
[MSG:DBG: Limit switch tripped for X motor 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Limit switch tripped for Y motor 0]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG: Limit switch tripped for Z motor 0]
Send Command...
  

Autoscroll

qulet, 2025-01-28:
I forgot to trigger the A axis limits

[MSG:DBG:  X Neg Limit 0]
[MSG:DBG:  X Neg Limit 1]
[MSG:DBG: Limit switch tripped for X motor 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Limit switch tripped for Y motor 0]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG: Limit switch tripped for Z motor 0]
[MSG:DBG:  A Neg Limit 1]
[MSG:DBG: Limit switch tripped for A motor 0]
[MSG:DBG:  A Neg Limit 0][MSG:DBG:  X Neg Limit 0]
[MSG:DBG:  X Neg Limit 1]
[MSG:DBG: Limit switch tripped for X motor 0]
[MSG:DBG:  Y Neg Limit 0]
[MSG:DBG:  Y Neg Limit 1]
[MSG:DBG: Limit switch tripped for Y motor 0]
[MSG:DBG:  Z Neg Limit 0]
[MSG:DBG:  Z Neg Limit 1]
[MSG:DBG: Limit switch tripped for Z motor 0]
[MSG:DBG:  A Neg Limit 1]
[MSG:DBG: Limit switch tripped for A motor 0]
[MSG:DBG:  A Neg Limit 0]

qulet, 2025-01-28:
Problem solved!!!!
all limit_neg_pin: had to change from "low" to "high"

Thank you all for your HELP
I am closing the topic.

closed 2025-01-28 by qulet

bdring, 2025-01-28:
Please consider donating to this project!!!