bdring/FluidNC #1434: 10 volts inoperative output (spindle )

DVA85, 2025-01-21:
### Wiki Search Terms

show file .yaml below

### Controller Board

nothing here

### Machine Description

nothing here

### Input Circuits

```yaml
nothing here
```

### Configuration file

```yaml
board: 6 Pack
name: 5 Pack External Router XYYZR 10V Spindle
meta: Pilou 27/12/2024 V1

arc_tolerance_mm: 0.002
junction_deviation_mm: 0.010
verbose_errors: false
report_inches: false
enable_parking_override_control: false
use_line_numbers: false

start:
  must_home: true
  deactivate_parking: false

stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 319.027
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 806.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.26:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1
        disable_pin: I2SO.0

  y:
    steps_per_mm: 798.005
    max_rate_mm_per_min: 1500.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 1080.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 0.000
      feed_mm_per_min: 50.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.33:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

    motor1:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8    
    
  z:
    steps_per_mm: 1926.453
    max_rate_mm_per_min: 1000.000
    acceleration_mm_per_sec2: 25.000
    max_travel_mm: 226.700
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 0.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 2.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

 a:
    steps_per_mm: 53.333
    max_rate_mm_per_min: 10000.0
    acceleration_mm_per_sec2: 500.000
    max_travel_mm: 400.000
    soft_limits: false
    homing:
      cycle: 0
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.18
        direction_pin: I2SO.17
        disable_pin: I2SO.16

probe:
  pin: gpio.35:low
  check_mode_start: true

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

RapidChange:
  alignment: Y_AXIS
  direction: POSITIVE
  number_of_pockets: 4
  pocket_offset: 38.000
  x_pocket_1: -439.720
  y_pocket_1: -150.170
  z_engage: -80.000
  z_traverse: -10.000
  z_safe_clearance: 0.000
  engage_feed_rate: 1600
  load_rpm: 1600
  unload_rpm: 1600
  dust_cover_enabled: False
  tool_recognition_enabled: False
  tool_setter_enabled: True
  x_tool_setter: -439.730
  y_tool_setter: -212.000
  z_seek_start: 0.000
  z_safe_tool_setter: 0
  set_tool_offset: 43
  seek_feed_rate: 300
  seek_retreat: 5
  set_feed_rate: 100
  set_tool_max_travel: 50.000
  dust_cover_axis: A_AXIS
  dust_cover_open_position: 0.000
  dust_cover_closed_position: 0.000
  dust_cover_use_output: False
  z_tool_recognition_zone_1: 0.000
  z_tool_recognition_zone_2: 0.000
  dust_cover_pin: NO_PIN
  tool_recognition_pin: NO_PIN

coolant:
  mist_pin: gpio.12
  delay_ms: 0

parking:
  enable: true
  axis: Z
  target_mpos_mm: -5.0
  rate_mm_per_min: 800.0
  pullout_distance_mm: 5.0
  pullout_rate_mm_per_min: 250.0

macros:
  startup_line0:
  startup_line1:
  macro0: G54G0Z0&G54G0X0Y0
  macro1: 
  macro2:
  macro3:
  after_reset: G20
  after_unlock: G91

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.2:low:pu
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

10V:
  forward_pin: gpio.15
  reverse_pin: gpio.14
  pwm_hz: 5000
  output_pin: gpio.13
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 6000=25%.000 12000=50.000% 24000=100.000%
  off_on_alarm: true
```

### Startup Messages

```text
no message when start to run
```

### User Interface Software

Fluidnc

### What happened?

nothing here

### GCode File

nothing here

### Other Information

nothing here

MitchBradley, 2025-01-21:
There is not enough information for us to help.  "no message when start to run" does not tell us which user interface program or sender you are using when looking for messages, so we cannot guess what might be wrong with your setup.  Please tell us exactly what you did to get this far.  Give us a lot of details.

bdring, 2025-01-21:
I saw your video. It is clear that something is not set up right. The io.13 LED and either the io.14 or io.15 LEDs should be lit.

[image]

How are you turning on the spindle?

You should be sending the `M3S24000` gcode line from the console.

You can get the startup messages by sending $SS at the console as described in the problem report template.

[image]

DVA85, 2025-01-21:
je vais voir pour la ligne de code m3S24000 et pour le message de démarrage, pour le moment voici mes constatations et mes connections ... je n'arrive pas à avoir 10 volts sur votre carte... la led io.13 qui correspond à ma config FluidNc.... output_pin: gpio.13 n'est jamais allumée...
J'ai avec m3 ou m4 les leds io.15 et io.14 qui s'allument (forward_pin: gpio.15 reverse_pin: gpio.14) et io.12 pour le liquide de refroidissement (mist_pin: gpio.12)....
J'ai un VDF YL620 branché comme suit : carte vers VDF.... io.14/Rev... io.15/Fwd... io.13/10v... Gnd/Gnd...
j'ai reglé également les parametres de mon VDF comme suit : 
P00.01 sur 1
P07.08 sur 3

MitchBradley, 2025-01-21:
Please tell us what action you are performing, using what user interface program, in order to try and make the 10V signal go active.  You said "je n'arrive pas à avoir 10 volts" but you did not say what you did in order to ask FluidNC to make that signal go active.

bdring, 2025-01-21:
I cannot help you until I see the startup messages. There could be an error in your configuration file. I don't know if the configuration file is loaded.

What interface software are you using. The FluidNC WebUI, a gcode sender, FluidTerm? Can you connect a basic serial terminal to the  controller?

http://wiki.fluidnc.com/en/features/serial_terminals

DVA85, 2025-01-22:
I installed FluidNc on the PCB following the video ( https://www.youtube.com/watch?v=NxfWvtqhs1M&t=268s )

 from 4min30 to 9minutes then 11.19min to 18.20 minutes...with Rapid Change ATC + WebUi...etc...

and in my config here is the part declared for 10 volts:10V:

  forward_pin: gpio.15
  reverse_pin: gpio.14
  pwm_hz: 5000
  output_pin: gpio.13
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 6000=25%.000 12000=50.000% 24000=100.000%
  off_on_alarm: true

DVA85, 2025-01-22:
but I don't understand why the 10 volts is not at the output for the spindle... especially since m3 lights up then m4... should this open the "relay" to send 10 v??
For the rest everything works correctly... homing, 3.3v, 5v, movements of the X, Y, Y1, Z axes according to the motor 1, motor 2, motor 3, motor 4 outputs... and so my config.yaml takes this into account....
Is the 10 v correctly declared in my config.yaml??

bdring, 2025-01-22:
Please send me the startup messages.

Measure the voltage with the wires disconnected from the vfd.

What happens to the voltage when you send m3s6000, the m3s12000, then m3s24000

bdring, 2025-01-22:
Can you show the connections on the VFD?

DVA85, 2025-01-22:
results of my tests...
m3s6000=3.5v
m3s12000=5.7v
m3s24000=10.27v

same voltages for m4s6000, s12000, s24000

same voltage without wires connected

m5 off io.13

$ss ( see video )

VDF connection ( see photo )

[image]

[image]
[image]

[image]

i get  a video for $ss but i don' t know how to send on github....

MitchBradley, 2025-01-22:
You have is connected wrong. 

[image]

V10 is an output from the VFD.  You need to connect the FluidNC 0-10V signal to the adjacent VI1 VFD *input*

MitchBradley, 2025-01-22:
You might also need a wire from FluidNC GND to VFD XGND .  The existing wire to VFD GND is for the analog 0-10V signal.  It might possibly work for the digital FWD and REV signals too, but it is better to have a separate one.

[image]

DVA85, 2025-01-22:
ok for this, i understand...i wire from Pcb GND/XGND ( VFD) and 0-10v signal from Pcb to VI 1....but do i need to change parameters of my VFD ?

Basic settings VFD......P00.01 = 0, P07.08= 0 but i read others settings on 620 YL...

do i need change with P00.01=1 and P07.08=3

**P00.01** 
Start / stop command source 

0:  Operator VFD Front panel button

1:  From external input (X1-X6); (VFD Stop
Button enabled)

2: From external input (X1-X6); (VFD Stop
Button disabled)

3:From Serial (Modbus Rs485)

4:By user application program control
Electric machinery

**P07.08** 
Frequency 1 select method 

0: Operator VFD Front Panel (parameter:
P03.06) 0 0 0

 1: Pre-set Freq, P00.01 Set frequency
value ,Operation panel keyboard,Can be
set directly

 2:No. X paragraph frequency

 3: Analog Input. :(P03.10-P03.13)

4: external simulation amount 2 (VI2)

5: (Modbus Rs485) Given frequency

6: User application program, given
frequency

7:( Pid)Output frequency 

thank's a lot...

MitchBradley, 2025-01-22:
>do i need change with P00.01=1 and P07.08=3

yes

DVA85, 2025-01-22:
than's a lot i'll try this and i'll come back to you...

bdring, 2025-01-22:
Let me make a better drawing for you. You may have damaged the controller by connecting to the 10V terminal.

Do all measurements I asked for with all wires to the VDF removed.

DVA85, 2025-01-22:
the VFD responds normally if I keep the control board, so I changed the settings...I don't think it was damaged...

latest tests...with connections (photo)

m3s6000....Forward lights up on the control panel (VFD)
m5 extinction io.13
m4s6000....Reverse lights up on the control panel (VFD)
m5 extinction io.13

but the spindle does not turn despite
P00.01 = 1
P07.08 = 3

but I don't know what parameter to change in my VFD settings....

DVA85, 2025-01-22:
[image]

bdring, 2025-01-22:
Before you connect to the VFD. What are the voltages with

M3S6000
M3S1200
M3S24000

DVA85, 2025-01-22:
i think i understand the reason why the spindle stay off...

I understand your connection...for mine it is missing the GREY cable which goes from the common (fluidNc card) to XGND (Vfd) and therefore removed my bridge between GND/Fwd/Rev Com (FluidNc card)...I think that will completely resolve my search...I will keep you informed on Sunday...thank you very much

bdring, 2025-01-22:
Hopefully connecting the 10V output on the controller to the 10V output on the VFD did not damage the opamp on the controller.

DVA85, 2025-01-26:
Results...no spindle start despite wiring with Com/XGND as in your photo...and the following parameters, P00.01 to 1 and 
P07.08 to 3.......also on "on" the 1st white pin of the red square module above the green strip or the 4 pins on "off" ...

I don't see which parameter to change on my Vfd or how to test the 10 volts of the Vfd...........

in manual the vfd dashboard responds and the spindle turns...........

the card responds well too...

I'm a bit short of solutions.....

[YL620-A Setup table 190707.pdf](https://github.com/user-attachments/files/18550366/YL620-A.Setup.table.190707.pdf)

bdring, 2025-01-26:
If the io.13 LED on the controller lights up with the M3S24000 command and you do not have any voltage on the 10v output pin of the controller, you probably broke the circuit on the controller by shorting it to the wrong pin on the VFD.

MitchBradley, 2025-01-26:
You can test the VFD without FluidNC by connecting a potentiometer on the GND, VI1 and V10 pins.  Most VFD manuals have a diagram showing that wiring.  If wired correctly and the VFD parameters are set correctly, turning that potentiometer knob will change the VFD speed, although you might also need to press the run button.

DVA85, 2025-01-26:
j'ai bien la Led io.13 qui s'allume, et j'ai également sur la Pcb les voltages en fonction de la demande S6000, S12000, S24000...
le voltage va de 5.5v à 10.5  volts en sortie Pcb 0-10 v...
mais par controleur tu veux dire du Vfd...? Dans ce cas je dois tester la sortie 10 volts du VFD ?? est-ce correct ?

DVA85, 2025-01-30:
Great everything works perfectly... if it can help here are my parameters for my VFD YL 620...
P00.00 = 400 ( fq 400 hz )
P00.01 = 1
P07.08 = 3
P03.12 = 100 ( fq mini )
P03.13 = 400 ( fq maxi )
for the white dip switches in the red module above the green strip you have to put on "on" the 2 and the 4 (activation 10 volts of the Vfd) and leave on "off" the 1 and the 3...

The wiring is correct with the configuration below...

A big thank you to all ....Phil

[image]

closed 2025-01-30 by bdring