bdring/FluidNC #1438: Problem:  Problem:

kotovasia5120, 2025-01-26:
### Wiki Search Terms

FluidNC Limit Switch and Homing Setup

### Controller Board

custom

### Machine Description

cnc3018

### Input Circuits

```yaml
100 ohm resistor in series with NC switch
```

### Configuration file

```yaml
board: custom
name: cnc/laser 3018
meta: 
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 4
  dir_delay_us: 0
  disable_delay_us: 0
  segments: 12

spi:
  miso_pin: NO_PIN
  mosi_pin: NO_PIN
  sck_pin: NO_PIN

sdcard:
  cs_pin: NO_PIN
  card_detect_pin: NO_PIN
  frequency_hz: 8000000

kinematics:
  Cartesian:

axes:
  shared_stepper_disable_pin: gpio.2
  shared_stepper_reset_pin: NO_PIN
  homing_runs: 2
  x:
    steps_per_mm: 800.000000
    max_rate_mm_per_min: 1000.000000
    acceleration_mm_per_sec2: 25.000000
    max_travel_mm: 2950.000000
    soft_limits: true
    motor0:
      limit_neg_pin: gpio.33
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000000
      stepstick:
        step_pin: gpio.18
        direction_pin: gpio.17
        disable_pin: NO_PIN
        ms1_pin: gpio.19:high
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN  
    homing:
      cycle: 2
      allow_single_axis: true
      positive_direction: false
      mpos_mm: 0.000
      seek_mm_per_min: 200
      feed_mm_per_min: 50
      seek_scaler: 1.1
      feed_scaler: 1.5 
      settle_ms: 250

  y:
    steps_per_mm: 800.000000
    max_rate_mm_per_min: 1000.000000
    acceleration_mm_per_sec2: 25.000000
    max_travel_mm: 1000.000000
    soft_limits: false
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000000
      stepstick:
        step_pin: gpio.0
        direction_pin: gpio.5
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN  

  z:
    steps_per_mm: 800.000000
    max_rate_mm_per_min: 1000.000000
    acceleration_mm_per_sec2: 25.000000
    max_travel_mm: 1000.000000
    soft_limits: false
    motor0:
      limit_neg_pin: NO_PIN
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000000
      stepstick:
        step_pin: gpio.16
        direction_pin: gpio.4
        disable_pin: NO_PIN
        ms1_pin: NO_PIN
        ms2_pin: NO_PIN
        ms3_pin: NO_PIN
        reset_pin: NO_PIN  

pwm:
  pwm_hz: 5000
  direction_pin: NO_PIN
  output_pin: gpio.15
  enable_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 300
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 10000=100.000%
  off_on_alarm: false  
  

control:
  safety_door_pin: NO_PIN
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: NO_PIN
  macro1_pin: NO_PIN
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN
  fault_pin: NO_PIN
  estop_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: NO_PIN
  delay_ms: 0

probe:
  pin: NO_PIN
  toolsetter_pin: NO_PIN
  check_mode_start: true
  hard_stop: false

macros:
  startup_line0:
  startup_line1:
  Macro0:
  Macro1:
  Macro2:
  Macro3:
  after_homing:
  after_reset:
  after_unlock:

start:
  must_home: false
  deactivate_parking: false
  check_limits: true

parking:
  enable: false
  axis: Z
  target_mpos_mm: -5.000000
  rate_mm_per_min: 800.000000
  pullout_distance_mm: 5.000000
  pullout_rate_mm_per_min: 250.000000

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN

user_inputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN
  digital4_pin: NO_PIN
  digital5_pin: NO_PIN
  digital6_pin: NO_PIN
  digital7_pin: NO_PIN

arc_tolerance_mm: 0.002000
junction_deviation_mm: 0.010000
verbose_errors: true
report_inches: false
enable_parking_override_control: false
use_line_numbers: true
planner_blocks: 16
```

### Startup Messages

```text
[MSG:INFO: Connecting to STA SSID:kotovasia_chup]
[MSG:INFO: Connecting.]
[MSG:INFO: Connecting..]
[MSG:INFO: Connected - IP is 192.168.1.113]
[MSG:INFO: WiFi on]
[MSG:INFO: Start mDNS with hostname:http://fluidnc.local/]
[MSG:INFO: HTTP started on port 80]
[MSG:INFO: Telnet started on port 23]
[MSG:INFO: PWM Spindle Ena:NO_PIN Out:gpio.15 Dir:NO_PIN Freq:5000Hz Period:8191]
```

### User Interface Software

WebUI

### What happened?

When returning to the x-axis, it returns home to the switch at high speed, stops, but does not roll back to measure at low speed. The cycle returns home, infinitely.
Connections:
100 ohm resistor in series with the trip switch.

kotovasia5120, 2025-01-26:
At the moment only one limit switch is installed

MitchBradley, 2025-01-26:
We are unable to offer support for custom controller boards.  It requires too much of our time, with no return, benefiting only one person.

bdring, 2025-01-26:
There is a flowchart for switch setup on the wiki. Follow it and report back with yourcresults.

kotovasia5120, 2025-01-26:
Thanks for the answer. Wiki is not always easy to navigate. I did not see this page.
As a result, I saw in the debug that when the limit switch was triggered, it was triggered not once, but many times per second.
It was necessary to enable :pu for the contact since the limit switch is NC.

> There is a flowchart for switch setup on the wiki. Follow it and report back with yourcresults.

closed 2025-01-26 by kotovasia5120

MitchBradley, 2025-01-26:
If you have ideas for how the wiki navigation could be better, please request wiki edit permission and improve it.

bdring, 2025-01-26:
It is in the navbar on the left side. It is the first page suggested if you search for homing and second page suggested if you search for limit switches.

I added a link in the FAQ.