bdring/FluidNC #1441: Problem: Settings for VFD/Spindle not working

kuito, 2025-01-27:
### Wiki Search Terms

1-10V spindle
spindle setup
spindle configuration

### Controller Board

FluidNC 6x CNC Controller v1.1

### Machine Description

X-Carve Pro:  2x Y Motors, 1x X Motor, 1x Z Motor, 1xVFD H100-1.5C2, 1x Spindle (see photo)

It's the 6x CNC Controller to 4 Stepperonline DM642T drivers

[image]
[image]

[image]
[image]
[image]

 

### Input Circuits

```yaml

```

### Configuration file

```yaml
board: 6x
name: GRFN PRJX XCP XYYZ 6x-fluidnc
stepping:
  engine: I2S_STREAM
  idle_ms: 255
  pulse_us: 10
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 42.353
    max_rate_mm_per_min: 7620.000
    acceleration_mm_per_sec2: 250.000
    max_travel_mm: 1219.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 25.000
      seek_mm_per_min: 750.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.2
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.2
        direction_pin: I2SO.1:low
        disable_pin: I2SO.0

  y:
    steps_per_mm: 42.353
    max_rate_mm_per_min: 7620.000
    acceleration_mm_per_sec2: 350.000
    max_travel_mm: 1219.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 25.000
      seek_mm_per_min: 750.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.26:pu
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.5
        direction_pin: I2SO.4
        disable_pin: I2SO.7

    motor1:
      limit_neg_pin: gpio.33:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      standard_stepper:
        step_pin: I2SO.10
        direction_pin: I2SO.9
        disable_pin: I2SO.8

  z:
    steps_per_mm: 200.000
    max_rate_mm_per_min: 300.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 50.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      settle_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.32:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 3.000
      standard_stepper:
        step_pin: I2SO.13
        direction_pin: I2SO.12
        disable_pin: I2SO.15

i2so:
  bck_pin: gpio.22
  data_pin: gpio.21
  ws_pin: gpio.17

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

probe:
  pin: gpio.39:low
  toolsetter_pin: gpio.36:low

# Using MOSFETs (Check Spindle Pin Usage
# coolant:
  # flood_pin: gpio.12
  # mist_pin: gpio.4
  # delay_ms: 0

start:
  must_home: false

# #begin PWM
# pwm:
  # pwm_hz: 5000
  # direction_pin: NO_PIN
  # output_pin: gpio.13
  # enable_pin: gpio.14
  # disable_with_s0: false
  # s0_with_disable: true
  # spinup_ms: 0
  # spindown_ms: 0
  # tool_num: 0
  # speed_map: 0=0.000% 10000=100.000%
  # off_on_alarm: false

# #begin 10V
10V:
  forward_pin: gpio.15
  reverse_pin: gpio.14
  pwm_hz: 400
  output_pin: gpio.13
  enable_pin: NO_PIN
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 1000=0.000% 6000=25% 24000=100.000%
  off_on_alarm: false
```

### Startup Messages

```text
[16:54:01] [ $ss ] [MSG:INFO: FluidNC v3.7.12 https://github.com/bdring/FluidNC]

[16:54:01] [ $ss ] [MSG:INFO: Compiled with ESP32 SDK:v4.4.4]

[16:54:01] [ $ss ] [MSG:INFO: Local filesystem type is littlefs]

[16:54:01] [ $ss ] [MSG:INFO: Configuration file:6x-grfn-prjx-xyyz.yaml]

[16:54:01] [ $ss ] [MSG:INFO: Machine GRFN PRJX XCP XYYZ 6x-fluidnc]

[16:54:01] [ $ss ] [MSG:INFO: Board 6x]

[16:54:01] [ $ss ] [MSG:INFO: I2SO BCK:gpio.22 WS:gpio.17 DATA:gpio.21]

[16:54:01] [ $ss ] [MSG:INFO: SPI SCK:gpio.18 MOSI:gpio.23 MISO:gpio.19]

[16:54:01] [ $ss ] [MSG:INFO: SD Card cs_pin:gpio.5 detect:NO_PIN freq:8000000]

[16:54:01] [ $ss ] [MSG:INFO: Stepping:I2S_stream Pulse:10us Dsbl Delay:0us Dir Delay:1us Idle Delay:255ms]

[16:54:01] [ $ss ] [MSG:INFO: Axis count 3]

[16:54:01] [ $ss ] [MSG:INFO: Axis X (150.000,1369.000)]

[16:54:01] [ $ss ] [MSG:INFO: Motor0]

[16:54:01] [ $ss ] [MSG:INFO: standard_stepper Step:I2SO.2 Dir:I2SO.1:low Disable:I2SO.0]

[16:54:01] [ $ss ] [MSG:INFO: X Neg Limit gpio.2]

[16:54:01] [ $ss ] [MSG:INFO: Axis Y (-1069.000,150.000)]

[16:54:01] [ $ss ] [MSG:INFO: Motor0]

[16:54:01] [ $ss ] [MSG:INFO: standard_stepper Step:I2SO.5 Dir:I2SO.4 Disable:I2SO.7]

[16:54:01] [ $ss ] [MSG:INFO: Y Neg Limit gpio.26:pu]

[16:54:01] [ $ss ] [MSG:INFO: Motor1]

[16:54:01] [ $ss ] [MSG:INFO: standard_stepper Step:I2SO.10 Dir:I2SO.9 Disable:I2SO.8]

[16:54:01] [ $ss ] [MSG:INFO: Y2 Neg Limit gpio.33:low]

[16:54:01] [ $ss ] [MSG:INFO: Axis Z (100.000,150.000)]

[16:54:01] [ $ss ] [MSG:INFO: Motor0]

[16:54:01] [ $ss ] [MSG:INFO: standard_stepper Step:I2SO.13 Dir:I2SO.12 Disable:I2SO.15]

[16:54:01] [ $ss ] [MSG:INFO: Z Neg Limit gpio.32:low]

[16:54:01] [ $ss ] [MSG:INFO: Kinematic system: Cartesian]

[16:54:01] [ $ss ] [MSG:INFO: 10V Spindle Ena:NO_PIN Out:gpio.13 Dir:NO_PIN Fwd:gpio.15 Rev:gpio.14 Freq:400Hz Period:131071]

[16:54:01] [ $ss ] [MSG:INFO: Using spindle 10V]

[16:54:01] [ $ss ] [MSG:INFO: Probe Pin: gpio.39:low]

[16:54:01] [ $ss ] [MSG:INFO: Connecting to STA SSID:icee]

[16:54:01] [ $ss ] [MSG:INFO: Connecting.]

[16:54:01] [ $ss ] [MSG:INFO: Connecting..]

[16:54:01] [ $ss ] [MSG:INFO: Connecting...]

[16:54:01] [ $ss ] [MSG:INFO: Connected - IP is 192.168.1.196]

[16:54:01] [ $ss ] [MSG:INFO: WiFi on]

[16:54:01] [ $ss ] [MSG:INFO: Start mDNS with hostname:http://slurpee.local/]

[16:54:01] [ $ss ] [MSG:INFO: SSDP Started]

[16:54:01] [ $ss ] [MSG:INFO: HTTP started on port 80]

[16:54:01] [ $ss ] [MSG:INFO: Telnet started on port 23]

[16:54:01] [ $ss ] ok
```

### User Interface Software

openbuilds

### What happened?

the spindle seems to be under powered doesn't go full speed.  When I take a multimeter to the V1-10 output and ground it only shows 1v .. and i've used a small screwdriver on the adjustment and it doesn't seem to do anything.. no change at all .. when i shut down the spindle it does go to zero though.

I've tested M3S18000, M3S24000 and it has no difference 

Separate q:
I also had to reverse the movement on the X to get it to go the right direction.. hoping that is ok.

### GCode File

NA

### Other Information

The video of the spindle is at:  https://www.dropbox.com/scl/fi/e0vealv1pyt2pi5ujf4nv/IMG_4729.MOV?rlkey=vnz71qerl6j4u0e1fmjyzveeb&dl=0

I have a couple laying around and both do the similar thing.. the settings in the VFD are the same as what it was before.. i haven't altered them at all.  

You will probably notice that I didn't deviate much from the sample config file for the 6x when I setup the machine.. other than some of the motor settings that the xcp had listed in their grbl config.

would love to finally get this machine going again.. it's so nice, but their custom board kept having a motor driver dying and they quit supporting it for me.. so i am looking to get completely off of inventables hardware.

bdring, 2025-01-27:
When you send M3S12000 do the io.13 and the io.15 LEDs glow? 
Is 13 dimmer?
Send S24000 does io.13 get brighter?

What voltage is present on the 10v output terminal at 24000 when the VFD is not connected?

Are you sure the pot is turning? It typically only turns with a small slotted screw driver. Watch carefully to see it turn. Be careful not to over-rotate the pot.

kuito, 2025-01-27:
Hard to tell if there is a discernible difference in the brightness of the LED.

When I connect the wire that was in the AI1 to the multimeter and ground it.. when I activate M3S12000 it goes to like 1.7v then stays at .7v
M3S24000 it jumps to like 3-4v then goes to sub 1v too.

I do see the pot turning but I do have a feeling I may have over twisted the pot and it is freely spinning — it is hard to see moving, but I can see it moving.  

If I love the pot when m3s24000 is running should the voltage change?  If so, it is not. 

I apologize for my ignorance with these things but will donate when I get home (didn’t bring wallet to my shop somehow).. 

I really appreciate your help //g

MitchBradley, 2025-01-27:
Is your voltmeter in AC mode?

kuito, 2025-01-28:
Yes it was .. 🤦‍♂️ 
Okie.. so I was able to get m3s24000 to register 9.9/10v and m3s12000 to register 5.0v
And there is now a difference in the LED.

Of course, I re-connect my VFD and spindle and it isn’t firing up.  The battle continues.

MitchBradley, 2025-01-28:
double check your vfd settings.  I have seen them reset to defaults for obscure reasons.

kuito, 2025-01-28:
https://www.dropbox.com/scl/fi/9hszjgmeoq4p7iq8etfld/IMG_4753.MOV?rlkey=bhzp3nc54a1qix3r7elef8wwx&st=3l2iclbh&dl=0

This is the spindle and VFD while I go from 18000 to 24000 — does the rVAL (output freq) look right for those two spindle speeds?

F001=1
F002=1
F003=400.0
F004=400.0
F005=400.0
F006=1.0
F007=.00.5
F008=220.0
F009=12.6
F010=3.0
F011=0.0
F012=1

This is what I’m getting from the first 12 settings on the H100.

MitchBradley, 2025-01-28:
According to this page https://www.aliexpress.us/item/3256807279354478.html , A12 is the 0-10V input.

[image]

A11 seems to be an output, according to the arrow directions.  The representation of a potentiometer is goofy and the output arrow could be confused with a pot wiper, but I think the arrows represent signal direction.

MitchBradley, 2025-01-28:
There is also this oh-so-helpful text

[image]

MitchBradley, 2025-01-28:
I think this might be the manual https://workerbeeit.com/H100SeriesManual.pdf

MitchBradley, 2025-01-28:
The manual says this 

[image]

[image]

Maybe A11 really is the right input to use.

kuito, 2025-01-28:
It was the input that the old inventables card used as the input .. AI1 and the manual I was looking at is similar to the last one you posted.

MitchBradley, 2025-01-28:
I cannot tell from your photos where the two black wires go.  

[image]

One of them goes to the VFD GND next to A11, but which one, and where does the other go?  I think it should be hooked up like this.

[image]

The 6x out-common pin is not internally connected to the Gnd for the 0-10V output

[image]

It is okay to connect out-common and Gnd, but unless you do so, or otherwise connect the 6x Gnd to the VFD GND, the VFD will not have a ground reference for the 0-10V output.

kuito, 2025-01-28:
I believe VFD GND goes to 6x GND.  Out common goes to a splitter that grounds with the power supply.. if I remember correctly.

Tomorrow, I can run the out common wire from 6x to the terminal you marked on the VFD.  I haven’t tried that yet.

Thanks for your help with this.. I will report back tomorrow.

kuito, 2025-01-29:
I followed your instructions of putting the 6x common out to the top VFD GND terminal and then the 6x GND to the bottom VFD GND and I left the 1-10V out to the AI1.  And I got:
https://www.dropbox.com/scl/fi/k6ii15b4d470z48yzeexn/IMG_4770.MOV?rlkey=44kkr1ea3u1l1dqzoi9y0o9jm&st=pj40pl5x&dl=0

I’m pretty stoked now and thank you so much.

[image]
I’m going to add proximity sensor limit switches to the XCP.. once I figure out which wires correspond to each other.

The last part is getting the XCP HMI to function too

[image]
[image]
[image]

And then re-route and clean up this mess..

[image]
Need to find a good way to anchor the 6x down to the controller box.

Another hopefully quick question.. do you know what this breakout card did for the original Inventables card?

[image]

Thanks again for all your help.  I am very appreciative of //g